zephyrproject-rtos/zephyr #44822: STM32F103 Custom Board Clock Config Error

lainslie, 2022-04-13:
Hi there,

I recently started working with a custom stm32f103c8t6 pcb. I seem to be having trouble with setting up the device tree correctly for the system clock. I was able to successfully flash the pcb, with blinking led sample, through stm32cubeide with the following clock configuration:

[image]

With this config, I attempted to replicate it in my {boardname}.dts file in the following manner:

[image]

I have been able to verify that in fact my program is being programmed onto the microcontroller, however, when the GPIO pin is written to, nothing occurs. Further to this, if k_msleep is used, the program crashes, which pointed in the direction of a clock issue? Any help would be greatly appreciated.

I've used both the zephyrsdk and gnuarm toolchain just to exhaust all options
Windows 10

additional details:
Defconfig -
[image]

lainslie, 2022-04-13:
Hi @FRASTM, 

Thanks for your prompt reply. Unfortunately that does not make a difference. I am very unsure what could possibly be wrong.

FRASTM, 2022-04-13:
Setting the systemClock from the HSI 8MHz is not usual. But I do not see any pb with the nucleo_f103rb running the samples/basic/blinky
(note that the stm32Cube calls LL_RCC_HSI_SetCalibTrimming(16) before enabling the HSI clock.)

erwango, 2022-05-18:
@lainslie Is your problem still present ?

If yes, can you try to replicate this configuration and see if solving the issue?
```
&clk_hsi {
	//hse-bypass;
	//clock-frequency = <DT_FREQ_M(8)>; /* STLink 8MHz clock */
	status = "okay";
};

&pll {
	mul = <9>;
	clocks = <&clk_hsi>;
	status = "okay";
};

&rcc {
	clocks = <&pll>;
	clock-frequency = <DT_FREQ_M(72)>;
	ahb-prescaler = <1>;
	apb1-prescaler = <2>;
	apb2-prescaler = <1>;
};
```

Also, regarding the code used, can you try to use `sample/basic/blinky` (if you have a led on your board) and report the status ?

lainslie, 2022-05-25:
Hi there, 

Thanks for all the replies. I ended up finding the issue, and it was not a clock configuration issue, meaning the original clock config that I posted did in fact work. The issue was due to the fact that the LED Pin that was being used was also used for JTAG programming whereas I only required SWD programming. 

It was fixed by adding this to my custom .dts file:

&pinctrl {
	swj-cfg = "jtag-disable";
};

erwango, 2022-05-25:
@lainslie Thanks for the update. I'm closing the point

closed 2022-05-25 by erwango