zephyrproject-rtos/zephyr #44830: Unable to set compiler warnings on app exclusively

riuriuriu, 2022-04-13:
### Discussed in https://github.com/zephyrproject-rtos/zephyr/discussions/42861

<div type='discussions-op-text'>

<sup>Originally posted by **riuriuriu** February 16, 2022</sup>
How can we apply stricter compiler warnings only on the CMake target `app`? 

Unfortunately, they will also be applied to all  Zephyr interface headers which get included by the app (gcc uses `-I` and not `-isystem` option).

For example, this simple change to the hello world sample will trigger build errors:
```diff
diff --git a/samples/hello_world/CMakeLists.txt b/samples/hello_world/CMakeLists.txt
index ecb7d24bb8..cd246b1c75 100644
--- a/samples/hello_world/CMakeLists.txt
+++ b/samples/hello_world/CMakeLists.txt
@@ -2,7 +2,10 @@
 
 cmake_minimum_required(VERSION 3.20.0)
 
+set(BOARD nucleo_f429zi)
+
 find_package(Zephyr REQUIRED HINTS $ENV{ZEPHYR_BASE})
 project(hello_world)
 
 target_sources(app PRIVATE src/main.c)
+target_compile_options(app PRIVATE -Werror=sign-conversion)
```

Which results in:

```bash
ccache /home/user/bin/zephyr-sdk-0.13.2/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc -DCORE_CM4 -DHSE_VALUE=8000000 -DKERNEL -DSTM32F429xx -DUSE_FULL_LL_DRIVER -DUSE_HAL_DRIVER -D_FORTIFY_SOURCE=2 -D__PROGRAM_START -D__ZEPHYR__=1 -I/home/user/projects/company/zephyrproject2/zephyr/include -I/home/user/projects/company/zephyrproject2/zephyr/build/zephyr/include/generated -I/home/user/projects/company/zephyrproject2/zephyr/soc/arm/st_stm32/stm32f4 -I/home/user/projects/company/zephyrproject2/zephyr/drivers -I/home/user/projects/company/zephyrproject2/zephyr/soc/arm/st_stm32/common -I/home/user/projects/company/zephyrproject2/modules/hal/cmsis/CMSIS/Core/Include -I/home/user/projects/company/zephyrproject2/modules/hal/stm32/stm32cube/stm32f4xx/soc -I/home/user/projects/company/zephyrproject2/modules/hal/stm32/stm32cube/stm32f4xx/drivers/include -I/home/user/projects/company/zephyrproject2/modules/hal/stm32/stm32cube/stm32f4xx/drivers/include/Legacy -I/home/user/projects/company/zephyrproject2/modules/hal/stm32/stm32cube/common_ll/include -isystem /home/user/projects/company/zephyrproject2/zephyr/lib/libc/minimal/include -isystem /home/user/bin/zephyr-sdk-0.13.2/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/10.3.0/include -isystem /home/user/bin/zephyr-sdk-0.13.2/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/10.3.0/include-fixed -Werror=sign-conversion -Os -imacros /home/user/projects/company/zephyrproject2/zephyr/build/zephyr/include/generated/autoconf.h -ffreestanding -fno-common -g -gdwarf-4 -fdiagnostics-color=always -mcpu=cortex-m4 -mthumb -mabi=aapcs -mfp16-format=ieee -imacros /home/user/projects/company/zephyrproject2/zephyr/include/toolchain/zephyr_stdint.h -Wall -Wformat -Wformat-security -Wno-format-zero-length -Wno-main -Wno-pointer-sign -Wpointer-arith -Wexpansion-to-defined -Wno-unused-but-set-variable -Werror=implicit-int -fno-asynchronous-unwind-tables -fno-pie -fno-pic -fno-reorder-functions -fno-defer-pop -fmacro-prefix-map=/home/user/projects/company/zephyrproject2/zephyr/samples/hello_world=CMAKE_SOURCE_DIR -fmacro-prefix-map=/home/user/projects/company/zephyrproject2/zephyr=ZEPHYR_BASE -fmacro-prefix-map=/home/user/projects/company/zephyrproject2=WEST_TOPDIR -ffunction-sections -fdata-sections -std=c99 -nostdinc -MD -MT CMakeFiles/app.dir/src/main.c.obj -MF CMakeFiles/app.dir/src/main.c.obj.d -o CMakeFiles/app.dir/src/main.c.obj -c /home/user/projects/company/zephyrproject2/zephyr/samples/hello_world/src/main.c
In file included from /home/user/projects/company/zephyrproject2/zephyr/include/kernel_includes.h:21,
                 from /home/user/projects/company/zephyrproject2/zephyr/include/kernel.h:17,
                 from /home/user/projects/company/zephyrproject2/zephyr/include/zephyr.h:18,
                 from /home/user/projects/company/zephyrproject2/zephyr/samples/hello_world/src/main.c:7:
/home/user/projects/company/zephyrproject2/zephyr/include/sys/atomic.h: In function 'atomic_test_bit':
/home/user/projects/company/zephyrproject2/zephyr/include/sys/atomic.h:81:49: error: conversion to 'unsigned int' from 'int' may change the sign of the result [-Werror=sign-conversion]
   81 | #define ATOMIC_ELEM(addr, bit) ((addr) + ((bit) / ATOMIC_BITS))
      |                                                 ^
/home/user/projects/company/zephyrproject2/zephyr/include/sys/atomic.h:130:32: note: in expansion of macro 'ATOMIC_ELEM'
  130 |  atomic_val_t val = atomic_get(ATOMIC_ELEM(target, bit));
      |                                ^~~~~~~~~~~
/home/user/projects/company/zephyrproject2/zephyr/include/sys/atomic.h:132:27: error: conversion to 'unsigned int' from 'int' may change the sign of the result [-Werror=sign-conversion]
  132 |  return (1 & (val >> (bit & (ATOMIC_BITS - 1)))) != 0;
      |                           ^
/home/user/projects/company/zephyrproject2/zephyr/include/sys/atomic.h: In function 'atomic_test_and_clear_bit':
/home/user/projects/company/zephyrproject2/zephyr/include/sys/atomic.h:81:49: error: conversion to 'unsigned int' from 'int' may change the sign of the result [-Werror=sign-conversion]
   81 | #define ATOMIC_ELEM(addr, bit) ((addr) + ((bit) / ATOMIC_BITS))
      |                                                 ^
/home/user/projects/company/zephyrproject2/zephyr/include/sys/atomic.h:151:19: note: in expansion of macro 'ATOMIC_ELEM'
  151 |  old = atomic_and(ATOMIC_ELEM(target, bit), ~mask);
```

And many more errors.

Is there a preferred way to set app specific flags and ignore zephyr headers?

Thanks
</div>

The expectation is to enable certain compiler checks the application (CMake target `app`) without being spammed with compiler warnings which origin from Zephyr headers. Is that already possible?

henrikbrixandersen, 2022-05-03:
You are including said headers in your application, so how would the compiler know not to check them? The headers are included by the preprocessor before handing the source off to the compiler, for which you have set the compiler flags.

mbolivar-nordic, 2022-05-03:
Closing the issue since there is an existing discussion

closed 2022-05-03 by mbolivar-nordic