zephyrproject-rtos/zephyr #44834: Add support for gpio expandeux NXP PCAL95xx

XavierChapron, 2022-04-13:
**Is your enhancement proposal related to a problem? Please describe.**

As of zephyr3.0 Zephyr support different version of GPIO expander from NXP, the following drivers are available:
- [gpio_pca953x.c](https://github.com/zephyrproject-rtos/zephyr/blob/main/drivers/gpio/gpio_pca953x.c)
- [gpio_pca95xx.c](https://github.com/zephyrproject-rtos/zephyr/blob/main/drivers/gpio/gpio_pca95xx.c)
- [gpio_pcal6408a.c](https://github.com/zephyrproject-rtos/zephyr/blob/main/drivers/gpio/gpio_pcal6408a.c)

I would like to add a driver for the PCAL9555 version which would be really close to the PCA95XX driver.
Basically, the only difference I have seen is that the L version as an additional register that must be configured to enable interruptions.
I am wondering if y should create a fork of gpio_pca95xx.c or create a DTS property to update the driver behavior regarding interruption configuration.

**Describe the solution you'd like**
I was thinking of adding a DTS property in nxp,pca95xx.yaml which would look like:
```
    has-interrupt-mask-reg:
      type: boolean
      required: false
      description: Has Interrupt mask register (PCAL95xx)
```
Then we could use this property to conditionally use the following diff that configures the described register:
```
diff --git a/drivers/gpio/gpio_pca95xx.c b/drivers/gpio/gpio_pca95xx.c
index 06094dcd3c..896606d57a 100644
--- a/drivers/gpio/gpio_pca95xx.c
+++ b/drivers/gpio/gpio_pca95xx.c
@@ -93,6 +93,7 @@ struct gpio_pca95xx_drv_data {
                uint16_t dir;
                uint16_t pud_en;
                uint16_t pud_sel;
+               uint16_t int_mask;
        } reg_cache;
 
        struct k_sem lock;
@@ -249,6 +250,17 @@ static inline int update_pul_en_regs(const struct device *dev, uint16_t value)
                               &drv_data->reg_cache.pud_en, value);
 }
 
+#ifdef CONFIG_GPIO_PCA95XX_INTERRUPT
+static inline int update_int_mask_regs(const struct device *dev, uint16_t value)
+{
+       struct gpio_pca95xx_drv_data * const drv_data =
+               (struct gpio_pca95xx_drv_data * const)dev->data;
+
+       return write_port_regs(dev, REG_INT_MASK_PORT0,
+                              &drv_data->reg_cache.int_mask, value);
+}
+#endif
+
 /**
  * @brief Setup the pin direction (input or output)
  *
@@ -592,6 +604,18 @@ static int gpio_pca95xx_pin_interrupt_configure(const struct device *dev,
 
        k_sem_take(&drv_data->lock, K_FOREVER);
 
+       uint16_t reg_out;
+       reg_out = drv_data->reg_cache.int_mask;
+       WRITE_BIT(reg_out, pin, (mode == GPIO_INT_MODE_DISABLED));
+
+       ret = update_int_mask_regs(dev, reg_out);
+       if (ret != 0) {
+               LOG_ERR("PCA95XX[0x%X]: failed to enable interrupt "
+                       "on pin %d (%d)", config->i2c_slave_addr,
+                       config->int_gpio_pin, ret);
+               goto err;
+       }
+
        /* Update interrupt masks */
        enabled = ((mode & GPIO_INT_MODE_DISABLED) == 0U);
        edge = (mode == GPIO_INT_MODE_EDGE);
```
(This diff applies on zephyr-2.5, but I would make one for zephyr main branche if needed.)

**Describe alternatives you've considered**
An alternative would be to create a fork of the driver and to insert the small changes described above in it.
A last option would have been to deduce the version at runtime from the WHOAMI register, but I can't see it in the datasheet.

dcpleung, 2022-04-13:
If you anticipate hardware with mixed PCA95xx chips, it would be better to guard the register update using the devicetree property for each device, so that the chip without the register will not have invalid reads/writes to non-existent register.

XavierChapron, 2022-04-19:
> If you anticipate hardware with mixed PCA95xx chips, it would be better to guard the register update using the devicetree property for each device, so that the chip without the register will not have invalid reads/writes to non-existent register.

Absolutely, the code in the issue description was there just to show the length of the diff, but if this solution is selected, I would add some logic with conditional code based on the DTS flag. Something like https://github.com/zephyrproject-rtos/zephyr/blob/main/drivers/gpio/gpio_pca95xx.c#L57

Would you agree with such a solution?

XavierChapron, 2022-04-26:
Closed by https://github.com/zephyrproject-rtos/zephyr/pull/44987

closed 2022-04-26 by XavierChapron