zephyrproject-rtos/zephyr #44847: battery: Initial dedicated API

aaronemassey, 2022-04-13:
---
name: RFC / Proposal
about: Submit an RFC / Proposal
title: 'Dedicated Battery Subsystem API'
labels: RFC
assignees: 'aaronemassey'

---

## Introduction

Add a dedicated battery subsystem API, for battery and battery fuel gauge
control.

### Problem description

A limited battery fuel gauge API exists in the increasingly large
`sensor_channel` `enum` defined by the [sensor API].
However, batteries are also controlled by drivers; this falls outside the
purview of the [sensor API].

### Proposed change

Add a dedicated battery subsystem driver API that combines sensor reading and
battery manipulation as a single device driver API.
This includes deprecating the [sensor API] battery fuel gauge APIs in favor of
the dedicated battery driver API.

This battery driver subsystem API is a modified combination of header files from
the [ChromeOS battery API] and [ChromeOS battery fuel gauge API].

## Detailed RFC

### Proposed change (Detailed)

See the accompanying initial [pull request] for detailed changes.

Followup changes will include driver unit tests leveraging the added API.
Upon request, these may be included in the aforementioned initial PR.

### Dependencies

The battery fuel gauge API implemented in the [sensor API] will be deprecated.
In future PRs, clients of the deprecated battery fuel gauge channels in the
[sensor API] will be refactored to leverage the new battery subsystem API.
Upon request, this refactor work can be included in the initial PR.

### Concerns and Unresolved Questions

#### Not an Expert

I am no expert on batteries nor Zephyr, so there may be cases I do not know to consider.

#### Nominal Increase in Binary Size

In code that does nothing functionally different than reading the same battery
fuel gauge sensor channels as before, some function pointers of the API will be
set to NULL.
This will result in a nominal increase in binary size with no
added functionality.

## Alternatives

### Status Quo

Pros

+ No risk of increase in binary size
+ No additional work on deprecating and migrating existing code

Cons

+ No unified mature API for battery manipulation

### Keep Both

Instead of replacing the battery fuel gauge API as a set of channels in the
[sensor API], add the battery device driver subsystem as an addition.

Pros

+ API clients may keep to the sensor API to avoid risk in the nominal binary
  size increase as well as familiarizing themselves with a new API.
+ No deprecation work is necessary.

Cons

+ Maintaining two similar APIs for the similar or same use-case could result in
  inconsistencies that could frustrate clients.

### Separate Battery and Battery Fuel Gauge APIs

Instead of a singular unified battery API, implement a battery controller
subsystem API in addition to a battery fuel gauage API.

Pros
+ We could feasibly keep the limited sensor battery fuel gauge API to avoid
  deprecation work.
+ We still get a device driver API for controlling the battery.

Cons
+ Two seperate APIs for reading and controlling batteries may be confusing given
  both use-cases are so coupled.

[ChromeOS battery API]: https://source.chromium.org/chromiumos/chromiumos/codesearch/+/main:src/platform/ec/include/battery.h
[ChromeOS battery fuel gauge API]: https://source.chromium.org/chromiumos/chromiumos/codesearch/+/main:src/platform/ec/include/battery_fuel_gauge.h
[sensor API]: ../sensor/sensor.h
[pull request]: https://github.com/zephyrproject-rtos/zephyr/pull/44846

tristan-google, 2022-04-13:
Happy to see this happening. Are there plans to add an implementation of this API for the Smart Battery standard?

aaronemassey, 2022-04-13:
> Happy to see this happening. Are there plans to add an implementation of this API for the Smart Battery standard?

That's a goal if the Zephyr community is interested!

aaronemassey, 2022-11-17:
See https://github.com/zephyrproject-rtos/zephyr/pull/46817 for MVP

aaronemassey, 2022-11-17:
https://github.com/zephyrproject-rtos/zephyr/issues/52315

matt-wood-ct, 2025-07-28:
It would be nice to see the fuel gauge driver API improved, Its current implementation is rather soft with minimal documentation of the fields functions, additionally it is unclear which fields are actually mandatory to implement.  

Without that nailed down it seems impossible to implement a subsystem which actually utilities these drivers generically.  

Many of the fields are vague and or duplicated, for example `FUEL_GAUGE_CONNECT_STATE` and `FUEL_GAUGE_PRESENT_STATE`. Other fields refer to flags which are completely undefined example `FUEL_GAUGE_CURRENT_DIRECTION`, `FUEL_GAUGE_FLAGS` and `FUEL_GAUGE_SBS_MODE`. The function `battery_cutoff` seems especially haphazard, given it has not clear means to undo it? or really much clarify on its actual function, I saw a TSC member talking about it in the discord today and even they were confused as to what it did.

To comment on the discussion of legacy sensor vs new fuel gauge drivers, if some people are still using the sensor API, we could just provide a stub driver to use any fuel gauge driver using the sensor API to allow users to continue their use case while we move everything over to the appropriate API. But IMO this API is not current fit for function, I think the bulk of the core actions should have actual API functions not just a cluster of semi coherent properties which are insistently implemented and supported.

Initial suggestion would be to force the following properties into being dedicated functions:

- State of charge
- Current direction
- Voltage
- Current

These should all be mandatory for any fuel gauge driver to allow fuel gauges to be usable generically, any other functions could be considered optional.

aaronemassey, 2025-08-01:
> These should all be mandatory for any fuel gauge driver to allow fuel gauges to be usable generically, any other functions could be considered optional.

In the much earlier proposed version of the API this was actually the case but we got a lot of push back for having these functions declared and required. I would also be interested in revisting this now that some time has passed and it seems like there's more interest in having the dedicated functions, time permitting. Unfortunately, I don't have a lot of time for this API at the moment and Tom owns numerous others as well. So any significant improvements sourced from him or I will be slow coming.