zephyrproject-rtos/zephyr #44856: Various kernel timing-related tests fail on hifive1 board

snematbakhsh, 2022-04-13:
As [fkokosinski](https://github.com/fkokosinski) mentioned in https://github.com/zephyrproject-rtos/zephyr/pull/44432#issuecomment-1092494107, some tests have been failing on hifive1 for a while.

> I've done some digging and bisecting, and from what I found the first bad commit is https://github.com/zephyrproject-rtos/zephyr   /commit/085125525895d17216e71512fd37e137dc1d35d4, but I have no idea why the CI didn't catch it back then. Also, the CI reports 12 failing tests, while in my local env only 3 of them are failing, regardless of whether I use Renode or QEMU to run the simulations, which are:

    tests/kernel/tickless/tickless_concept/kernel.tickless.concept
    tests/kernel/sleep/kernel.common.timing
    tests/kernel/workq/work_queue/kernel.workqueue

fkokosinski, 2022-04-26:
Thanks for reporting this issue. I'm looking into it.

fkokosinski, 2022-05-25:
Closing this one as it was addressed by #45344.

closed 2022-05-25 by fkokosinski