zephyrproject-rtos/zephyr #44861: WiFi support for STM32 boards

ademyankov, 2022-04-14:
Hello, 
   there is no WiFi support for STM32 boards w/ TrustZone b_u585i_iot02a, nucleo_l552ze_q, stm32l562e-dk, using TF-M.

Not having network for those boards in Zepher OS is a complete show-stopper!

Thus, obviously, sample, like _zephyr/samples/net/wifi_ does not work with these boards.

The CMake configuration phase states:
```
CMake Warning at /mnt/c/projects/zephyr/zephyr/CMakeLists.txt:810 (message):
  No SOURCES given to Zephyr library: drivers__wifi

  Excluding target from build.
```

I would like to ask a few questions:

Q1. Is there a plan on the road to add WiFi support for STM32?

Q2. If I would try to do it myself where should I look for? Is the doc or readme how to add/port this into Zepher OS? Is there an already ported WiFi for some other board that I can reverse engineer and at least try to do a similar approach?

Q3. A workaround approach. If I just bring the entire STM WiFi driver code into my app project and add all necessary board initialization into the main() will that approach work? At least I can do some primitive sockets on top of the STM WiFi driver.

Thank you.

erwango, 2022-04-19:
Hi, 

Thanks for your interest.

First, please note that out of the 3 boards you mention, only b_u585i_iot02a embeds a (MXCHIP) WIFI module.

> Q1. Is there a plan on the road to add WiFi support for STM32?

There is no plan currently to enable support for this module in short term.

>  Q2: If I would try to do it myself where should I look for? 

There are already some wifi driver implementations available there: `drivers/wifi`.
For instance, [disco_l475_iot01 board](https://docs.zephyrproject.org/2.6.0/boards/arm/disco_l475_iot1/doc/index.html supports WIFI using `drivers/wifi/eswifi` driver.

> Q3: If I just bring the entire STM WiFi driver code into my app project and add all necessary board initialization into the main() will that approach work? 

To some extent, yes, it should work. One link that could be useful in this process: https://github.com/zephyrproject-rtos/hal_stm32#use-stm32cube-in-your-application. Also when doing so, you should check that Cube BSP code does not interfere with Zephyr basic board settings (clocks, pin config, ...)

closed 2022-06-29 by erwango

jvermillard, 2023-02-08:
It would be nice to support the B_U585I_IOT02A wifi module