zephyrproject-rtos/zephyr #44867: CSIS: Add support for parallel operations and make implementation thread-safe

MariuszSkamra, 2022-04-14:
**Is your enhancement proposal related to a problem? Please describe.**
The current implementation disallows parallel operations on different set members. This might be problematic, as CSIS will be most likely used from different contexts.

Looking at the code, there is a static structure active_members that holds context. I guess the user might get into trouble when tries to use this API with multithreads and forbids concurrent operations on 2 sets.

asbjornsabo, 2022-08-25:
Zephyr-bt F2F: Applies for all profiles in LA Audio.

Thalley, 2024-02-09:
This is still valid, but needs to be split up into multiple issues, as fixing the entire LE Audio stack in a single go would be impossible

Thalley, 2024-05-09:
The parallel operation support was added by https://github.com/zephyrproject-rtos/zephyr/issues/44867. The thread-safe part require a new issue.

closed 2024-05-09 by Thalley