zephyrproject-rtos/zephyr #44872: k_timer callback timing incorrect with multiple lightly loaded cores

kv2019i, 2022-04-14:
**Describe the bug**
Regression introduced in commit 
https://github.com/zephyrproject-rtos/zephyr/pull/44845

Test rig to reproduce on Intel cAVS2.5 platform:
https://github.com/kv2019i/zephyr/commit/9d759bb7626198ec050174eb70c0bc91d28eb9b7

With PR44845 (main line Zephyr).
```
work0: worst 39137 avg 38399 expected 38400 (N = 8000)
work1: worst 40537 avg 38399 expected 38400 (N = 8000)
work2: worst 40418 avg 38398 expected 38400 (N = 8000)
work3: worst 40945 avg 38399 expected 38400 (N = 8000)
```

With PR44845 reverted:
```
work0: worst 42339 avg 39140 expected 38400 (N = 3000)
work3: worst 41466 avg 39137 expected 38400 (N = 3000)
work2: worst 41804 avg 39136 expected 38400 (N = 3000)
work1: worst 41593 avg 39140 expected 38400 (N = 3000)
```

**Additional context**
Related to prior bugs:
https://github.com/zephyrproject-rtos/zephyr/issues/44758
https://github.com/zephyrproject-rtos/zephyr/issues/43964

Add any other context that could be relevant to your issue, such as pin setting,
target configuration.

nashif, 2022-04-14:
when to make a test out of the above and put in the zephyr tree...

andyross, 2022-04-14:
I... hadn't realized that had merged so quickly.  Sorry, I was hoping to get this kind of feedback in the PR. :)

Will take a look with this rig.  The one I was using seemed solid.  Is this regressing the SOF performance issue too?

andyross, 2022-04-14:
I'm having trouble duplicating this.  Playing around with the rig I have, I do see very occasional excursions out to a few hundred cycles (and one weird run where there was an inexplicable single 5000 cycle event in the middle of the test -- maybe some kind of interference from the host?).  But I see them more or less symmetrically with or without the patch under discussion.

The big difference is the feature advertised: on current zephyr main, I see the delays only on core 0 (which is handling the interrupt) and much smaller numbers elsehwere, where on the previous fix core 0 and 1[1] would be delayed in lockstep because of the contended lock.

[1] Interestingly not core 2 and 3, which I guess are starting their ISRs earlier and get the lock themselves?  Not sure about this, would be good to investigate why they aren't behaving the same.

kv2019i, 2022-04-14:
@andyross Did you try my test rig (which is patching your test rig)? I found the problem does not happen with your original rig, but if I modified it to stop the spinning in the work thread, then I started getting bad averages.

andyross, 2022-04-14:
OK OK, I think I see what you're measuring here, and why it's only evident with an unloaded system.  And... it's not exactly a bug.

Here's what happens with the original (nested announce lock) fix: all the cores enter the timer interrupt simultaneously, and then spin waiting for the first one to finish the timeout iteration.  Then they return from the interrupt (actually two in sequence, there's an IPI handler that has to run too), which because the timeouts have all fired is the correct high priority thread that got woken up.

But with the current code, the cores no longer contend like that.  The second/third/fourth cores enter the timer interrupt and quickly exit, returning to the idle thread.  Now the resulting IPI from the timeout handler needs to wait for that context switch into the idle thread to complete, and then interrupt the idle core, and then switch into the thread we want.  Effectively we're doing twice the work!

But... that's actually because of the "optimization".  We really did get through the interrupt quickly and back into the app thread (which happens to have been the idle thread) faster.  It's just that this particular workload happens to be following that (faster!) timer interrupt with one triggered synchronously, which is a little pessimal.  But even so: we only pay this cost if we entered the idle thread in the first place, which is the right place (unloaded workloads) to put overhead.

So maybe we should be OK with this as long as it's not causing slipped ticks or other visible symptoms on real workloads?  But if we aren't... we don't want to fix it here, I don't think.  Take a look at the current version of the rig at https://github.com/andyross/zephyr/tree/cavs-latency-rig

This implements the same loaded/unloaded choice, and reports peak latencies for each reporting interval and not globally.

But it also has a patch that masks timer interrupts for secondary cores when they aren't needed (i.e. when timeslicing isn't used).  And when that is done, times get **a whole lot better**.  This hardware seems not to like simultaneous interrupts to the same ISR (not surprising I guess, all the memory those ISRs are hitting is uncached and they're taking locks that pretty much by definition are going to contend).

SOF doesn't use timeslicing currently (though I know there are some plans to do so in the future), so this might be a good choice for now.  Certainly the patch would be easy to verify.

andyross, 2022-04-14:
Short version: I'm tempted to close this as not-a-bug since it only affects latencies on unloaded systems.  But if we want to further optimize timer latencies, let's for sure look at that interrupt masking change (at the cost of timeslicing support), which is super effective.

andyross, 2022-04-14:
Also, if we want to preserve timeslicing AND the better latencies, it's probably time to start work on a more orthogonal timeslice architecture (make each CPU's slice interrupt a separate timeout, and implement a cpu-targetted IPI so that the right core can be poked).  More work.  Still very doable.

kv2019i, 2022-04-19:
@andyross I can verify masking timer-irq on secondary cores (patch in cavs-latency-rig branch) indeed fixes the avg problem.
In practise I think we should go ahead with this for SOF.

But, but, I'd still argue there is  a flaw in the logic. With the new rig and without the irq matching patch, I get result like this:

work0: last -1357 peak   2804 **avg   692** (N = 5894)
work1: last   424 peak   4067 **avg   699** (N = 5893) 
work2: last    80 peak   2945 **avg   699** (N = 5893)
work3: last  -797 peak   3327 **avg   698** (N = 5893)

I'd argue this is breaking the "k_timer interface contract". I mean it seems reasonable to have a compromise where enabling certain features (let's say time slicing on cAVS hw) will cause app to see more variation and higher peaks. But I would not expect the timer avg value to still change. In this case the timer is not really 1ms timer anymore, but a 1.02ms timer.

It seems there is some rounding error still when multiple cores are in compare_isr() at the same time. The avg of roughly 700 cycles of the 38.4Mhz platform clock is very close to one 50kHz tick (768*50kHz==38.4Mhz) of the Zephyr scheduler, so it would seem like in one of the cores ends up announcing one extra tick every millisecond, if all cores are allow to enter compare_isr() concurrently.

If I keep interrupts unmasked on all cores, but revert "kernel/timeout: Cleanup/speedup parallel announce logic", I get

work0: last    46 peak   -552 avg     0 (N = 14002)
work1: last  -329 peak  -1420 avg     0 (N = 14002)
work2: last  -808 peak  -2369 avg     0 (N = 14002)
work3: last   658 peak   1649 avg     0 (N = 14003)

... i.e. the avg stays 0 as expected.

andyross, 2022-04-28:
Been playing with this still.  The fix (#45174) that merged yesterday got ahead of the rig needed to measure it, which I have working reasonable well now here: https://github.com/andyross/zephyr/tree/xtensa-idle-overhead-rig

This basically measures the time (in Xtensa CCOUNT cycles) needed to:

1. Suspend the currently running thread
2. Context switch into the idle thread
3. Handle an interrupt (posted synchronously from within the idle thread, this is the needed whitebox hack)
4. Context switch back into the original thread

Which is to say, this is measuring the full path through an idle event.  In some sense this is the "minimal overhead required for Zephyr to do nothing".

And I've collected numbers for three configurations: "no SMP, uniprocessor kernel", "SMP, but without KERNEL_COHERENCE" (this isn't useful in practice as it only works with 1 cpu, but it's a good number for analysis), and "full KERNEL_COHERENCE support".

No invalidate fix:

```
SMP=n       1270
COHERENCE=n 2555
COHERENCE=y 6075
```

With fix (current Zephyr main):

```
SMP=n       1270
COHERENCE=n 2558
COHERENCE=y 5884
```

Some immediate lessons:

* None of these numbers are particularly slow, really.  At worst, a thread running on a 400 MHz TGL core will take 15us to "do nothing" (i.e. suspend itself via some mechanism and then get the CPU back).  That seems fairly tolerable to me.

* The numbers for single-cpu operation basically match my intuition from where Zephyr stands on other platforms, which is that it's **REALLY CRAZY FAST**.  Seriously, a little over a thousand cycles is the kind of thing other OSes like to cite for operations like "taking a lock", and we're getting two context switches and an interrupt in there.  Obviously SOF needs more complicated configuration, but it's nice to see our blazingly fast RTOS kernel still hiding in there.

* SMP by itself, without KERNEL_COHERENCE, is almost twice as slow.  This seems odd.  Certainly SMP kernels do have extra tracking and require more overhead for a lot of operations (e.g. spinlocks start having data, etc...).  But a 2x overhead is suspicious.  We weren't seeing this early on, so it's possible something has regressed.

* COHERENCE is good for another 2x factor.  This one seems more reasonable to me.  Context switching is where we take almost all the overhead here.  Again, it's not outrageously slow to my eyes.

* There's some code in the rig to dirty some stack space to measure flush overhead (the invalidation is now optimized out for typical cases, but we still need to flush the live stack of the outgoing thread).  As it happens this doesn't really do much.  Dirtying 1024 bytes takes ~100 cycles extra, which isn't very expensive (again, almost all that data would all come out of the cache at some point, the flush is just front loading it).

* Because the invalidate fix already merged, I don't have any measurement of what it was trying to fix.  But this would be interesting to measure too: basically read the stack_buf[] array back in after we resume to see how long that takes if it's uncached.

andyross, 2022-05-10:
Closing this per earlier discussion.  I think we have a good handle on the performance characteristics here and they don't seem to be a blocker for real applications on hardware.  We can always reopen or add new issues to track some of the features discussed (e.g. doing cache management at "migration time", adjusting interrupt targetting, etc...)