zephyrproject-rtos/zephyr #44874: error log for locking a mutex in an ISR

benediktibk, 2022-04-14:
**Is your enhancement proposal related to a problem? Please describe.**
I ran into the issue of locking a mutex within the call context of an ISR. An issue caused by myself, but it took quite a while to figure it out.

**Describe the solution you'd like**
Either an error log message or maybe even a hard fault when trying to call for instance k_mutex_lock within an invalid call context. I am not totally sure if it is possible to implement this, and it would possibly have a performance impact. So maybe just as an optional feature for mutexes.

jfischer-no, 2022-04-14:
This is already implemented, you have to enable Kconfig option CONFIG_ASSERT, see https://github.com/zephyrproject-rtos/zephyr/blob/main/kernel/mutex.c#L101

closed 2022-04-14 by jfischer-no

benediktibk, 2022-04-14:
Nice. Thanks!