zephyrproject-rtos/zephyr #44882: doc: Section/chapter "Supported Boards" missing from pdf documentation

amarkovicsa, 2022-04-14:
**Section/chapter "Supported Boards" missing from pdf documentation**

In online documentation, there is a page on "Supported Boards":

https://docs.zephyrproject.org/latest/boards/index.html#boards

However, in pdf documentation, that page is missing. This also causes all hyperlinks from the rest of the pdf to that page to be displayed as plain text, rather than hyperlink.

This is visible in 3.0.0 documentation - https://docs.zephyrproject.org/3.0.0/zephyr.pdf - and also on the most recent builds from main branch.

**To Reproduce**

Steps to reproduce the behavior:

1. Follow the procedure for building documentation from https://docs.zephyrproject.org/latest/contribute/documentation/generation.html?highlight=documentation%20build
2. After the step "ninja pdf", open and observe output file in <zephyr/doc/_build/latex/Zephyr.pdf>
3. In the opening chapter "Introduction", and just before the section "1.1. Licensing" will be displayed as plain text: "The full list of supported boards based on these architectures can be found here." - no hyperlink, and there should be one (word "here")
4. There are several similar cases of missing links in the Zephyr.pdf.
5. Section "Supported Boards" is also missing from the document.

**Expected behavior**

Section "Supported Boards" should be an integral part of pdf documentation, and all hyperlinks from the same pdf to it should work.

nashif, 2022-04-15:
This is the expected and desired behaviour, we do not want to have the board documentation in the PDF, those should only  be referenced and available online.

amarkovicsa, 2022-04-15:
> This is the expected and desired behaviour, we do not want to have the board documentation in the PDF, those should only be referenced and available online.

OK, thanks for explaining. In this case, I am going to close this issue (I am going to press the button "Close with comment" for this comment.). Please let me know if anything else should be perhaps done regarding this issue.

closed 2022-04-15 by amarkovicsa