zephyrproject-rtos/zephyr #44904: PR#42879 causes a hang in the shell history

keith-zephyr, 2022-04-15:
**Describe the bug**
PR#42879 causes a hang in the shell history code.   If this change is reverted, I'm unable to reproduce the failure.

Tested out on a board with these configs
```
CONFIG_SHELL_PRINTF_BUFF_SIZE=180
CONFIG_SHELL_BACKEND_SERIAL_TX_RING_BUFFER_SIZE=1024
CONFIG_SHELL_BACKEND_SERIAL_INIT_PRIORITY=0
CONFIG_SHELL_BACKEND_SERIAL_INTERRUPT_DRIVEN=y
CONFIG_SHELL_BACKEND_SERIAL_RX_RING_BUFFER_SIZE=64
CONFIG_SHELL_MINIMAL=y
CONFIG_SHELL_CMD_BUFF_SIZE=128
CONFIG_SHELL_HISTORY=y
```

**To Reproduce**
1. Input a command to the console that is 9 bytes in length
2. Input a command to the console that is 5 bytes in length
3. Repeat back to step 1

After looping approximately 17 times, `z_shell_history_put()` hangs, causing a watchdog timeout.

**Expected behavior**
The shell thread should not hang.

carlescufi, 2022-04-19:
@npitre will you be able to look at fixing this soon or should we just revert your change?

npitre, 2022-04-19:
> @npitre will you be able to look at fixing this soon or should we just revert your change?

I intend to have it fixed it today.

PR #44960 (2022-04-20): shell: fix history optimization
Fixes #44904