zephyrproject-rtos/zephyr #44908: Support ESP32 ADC

wolterhv, 2022-04-15:
This is a request to support the ADC peripheral on the ESP32 board. This would benefit Zephyr since the ESP32 is a popular board; supporting it better may increase Zephyr's user base.

I will be working to implement support for this peripheral. This issue is meant to track progress and signal the community that there are efforts being made in this area. This is a volunteering effort from a community member, not involved in any way with Espressif (OEM).

Some documents of interest:

* [ESP32 Technical Reference Manual](https://www.espressif.com/sites/default/files/documentation/esp32_technical_reference_manual_en.pdf). This has detailed information about the subsystems in the board, ADC and clocks included.
* [ESP32 Datasheet](https://www.espressif.com/sites/default/files/documentation/esp32_datasheet_en.pdf). This has some information about the ESP32 internals, though it is less in-depth than the TRM.

sylvioalves, 2022-04-15:
@wolterhv, great to see that.  
Please, consider using `hal` calls so we can keep this a common approach.  
https://github.com/zephyrproject-rtos/hal_espressif/blob/zephyr/components/hal/adc_hal.c

sylvioalves, 2022-05-25:
@wolterhv Do you have updates?

wolterhv, 2022-06-04:
@sylvioalves No progress yet, actually just getting to it now. Let me see if I understand the suggested approach correctly: the idea is to implement a HAL-based ADC driver so that it works for all versions of ESP32, and implement the chip-specific code in the HAL?

prayassamriya, 2022-08-05:
@wolterhv can you please provide rough estimate by when it could be ready. I would like to plan my project accordingly

prayassamriya, 2022-08-25:
@sylvioalves, is there any updates on adc driver development for ESP?

closed 2022-11-09 by github-actions[bot]

wolterhv, 2023-02-27:
For reference, at least some of the features in the task list have been implemented by @marekmatej in PR https://github.com/zephyrproject-rtos/zephyr/pull/52751