zephyrproject-rtos/zephyr #44921: Can't run helloworld using mps_an521_remote

arpitkalla, 2022-04-17:
**Describe the bug**
Error when helloworld is tride to run on the qemu using board mps2_an521_remote.

**To Reproduce**
Steps to reproduce the behavior:
`west build -p -b mps2_an521_remote samples/hello_world -t run`

**Expected behavior**
Expectation is like the output of `west build -p -b mps2_an521 samples/hello_world -t run`

**Logs and console output**

```
# west build -p -b mps2_an521_remote samples/hello_world -t run

-- west build: running target run
[1/153] Preparing syscall dependency handling

[3/153] Generating include/generated/version.h
-- Zephyr version: 3.0.99 (/Users/arpitk/Desktop/my-workspace/zephyr), build: zephyr-v3.0.0-2733-g13f382287ad0
[142/153] Linking C executable zephyr/zephyr_pre0.elf

[146/153] Linking C executable zephyr/zephyr_pre1.elf

[152/153] Linking C executable zephyr/zephyr.elf
Memory region         Used Size  Region Size  %age Used
           FLASH:       13772 B       512 KB      2.63%
            SRAM:        4192 B         1 MB      0.40%
           SRAM1:          0 GB         4 MB      0.00%
         SRAM2_3:          0 GB         4 MB      0.00%
        IDT_LIST:          0 GB         2 KB      0.00%
[152/153] To exit from QEMU enter: 'CTRL+a, x'[QEMU] CPU: cortex-m33
char device redirected to /dev/ttys003 (label hostS0)
/Users/arpitk/Desktop/my-workspace/zephyr/build/zephyr/boards/arm/mps2_an521/empty_cpu0-prefix/src/empty_cpu0-build/zephyr/zephyr.elf: No such file or directory
qemu-system-arm: -device loader,file=/Users/arpitk/Desktop/my-workspace/zephyr/build/zephyr/boards/arm/mps2_an521/empty_cpu0-prefix/src/empty_cpu0-build/zephyr/zephyr.elf: Cannot load specified image /Users/arpitk/Desktop/my-workspace/zephyr/build/zephyr/boards/arm/mps2_an521/empty_cpu0-prefix/src/empty_cpu0-build/zephyr/zephyr.elf
FAILED: zephyr/CMakeFiles/run_qemu /Users/arpitk/Desktop/my-workspace/zephyr/build/zephyr/CMakeFiles/run_qemu
cd /Users/arpitk/Desktop/my-workspace/zephyr/build && /usr/local/bin/qemu-system-arm -cpu cortex-m33 -machine mps2-an521 -nographic -m 16 -vga none -net none -pidfile qemu.pid -chardev stdio,id=con,mux=on -serial chardev:con -mon chardev=con,mode=readline -icount shift=7,align=off,sleep=off -rtc clock=vm -chardev pty,id=hostS0 -serial chardev:hostS0 -device loader,file=/Users/arpitk/Desktop/my-workspace/zephyr/build/zephyr/zephyr.elf -device loader,file=/Users/arpitk/Desktop/my-workspace/zephyr/build/zephyr/boards/arm/mps2_an521/empty_cpu0-prefix/src/empty_cpu0-build/zephyr/zephyr.elf
ninja: build stopped: subcommand failed.
```

**Environment (please complete the following information):**
 - OS: MacOS
 - Toolchain: gcc-arm-none-eabi

carlescufi, 2022-04-19:
@arpitkalla could you please try with the Zephyr SDK on Linux, using the bundled toolchain and qemu?

stephanosio, 2022-04-19:
I just tested this on macOS using the Zephyr SDK 0.14.1 and the QEMU installed from Hombrew, and am unable to reproduce this.