zephyrproject-rtos/zephyr #44925: intel_adsp_cavs25: multiple tests failed after running tests/boards/intel_adsp

LixinGuoX, 2022-04-18:
**Describe the bug**
When running tests/boards/intel_adsp on cavs25, it may stuck at some test functions and get failed due to timeout, then other tests which should get passed (for example: tests/crypto/tinycrypt) failed without any log.
This issue can be avoided by reverting this   PR: #43267 or power cycle the board.

**To Reproduce**
Steps to reproduce the behavior:
1. twister -W -p intel_adsp_cavs25 --device-testing --device-serial-pty=$HOME/zephyrproject/zephyr/soc/xtensa/intel_adsp/tools/cavstwist.sh --west-flash=$HOME/zephyrproject/zephyr/soc/xtensa/intel_adsp/tools/cavstwist.sh -T tests/crypto/tinycrypt -T tests/boards/intel_adsp/ -j 1 -vv

**Expected behavior**
Single failing case should not affect other cases.

**Logs and console output**
for tests/boards/intel_adsp/smoke:
```
DEBUG   - DEVICE: INFO:cavs-fw:Mapped PCI bar 0 of length 16384 bytes.
DEBUG   - DEVICE: INFO:cavs-fw:Selected output stream 7 (GCAP = 0x1009701)
DEBUG   - DEVICE: INFO:cavs-fw:Mapped PCI bar 4 of length 1048576 bytes.
DEBUG   - DEVICE: INFO:cavs-fw:Detected cAVS 1.8+ hardware
DEBUG   - DEVICE: INFO:cavs-fw:Trimming 16 bytes of extended manifest
DEBUG   - DEVICE: INFO:cavs-fw:Resetting HDA device
DEBUG   - DEVICE: INFO:cavs-fw:Powering down DSP cores
DEBUG   - DEVICE: INFO:cavs-fw:Configuring HDA stream 7 to transfer firmware image
DEBUG   - DEVICE: INFO:cavs-fw:type of mem is <class 'mmap.mmap'>
DEBUG   - DEVICE: INFO:cavs-fw:Mapped 2M huge page at 0x152400000 to contain 106496 bytes of firmware
DEBUG   - DEVICE: INFO:cavs-fw:Filled the buffer descriptor list (BDL) for DMA.
DEBUG   - DEVICE: INFO:cavs-fw:Starting DSP, ADSPCS = 0xf0f
DEBUG   - DEVICE: INFO:cavs-fw:Wait for ROM startup
DEBUG   - DEVICE: INFO:cavs-fw:Sending IPC command, HIPIDR = 0x81004000
DEBUG   - DEVICE: INFO:cavs-fw:Starting DMA, FW_STATUS = 0x5000001
DEBUG   - DEVICE: INFO:cavs-fw:Waiting for firmware handoff, FW_STATUS = 0x5000001
DEBUG   - DEVICE: INFO:cavs-fw:cAVS firmware load complete
DEBUG   - DEVICE: --
DEBUG   - DEVICE: Running test suite intel_adsp
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_smp_boot_delay
DEBUG   - DEVICE: Launch cpu1
DEBUG   - DEVICE: Launch cpu2
DEBUG   - DEVICE: Launch cpu3
DEBUG   - DEVICE: PASS - test_smp_boot_delay in 0.61 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_cpu_halt
DEBUG   - DEVICE: halt/restart core 1...
DEBUG   - DEVICE: halt/restart core 2...
DEBUG   - DEVICE: 
DEBUG   - DEVICE: Assertion failed at WEST_TOPDIR/zephyr/tests/boards/intel_adsp/smoke/src/cpus.c:167: halt_and_restart: (alive_flag is true)
DEBUG   - DEVICE: cpu didn't halt
DEBUG   - DEVICE: FAIL - test_cpu_halt in 0.251 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_post_boot_ipi
DEBUG   - DEVICE: PASS - test_post_boot_ipi in 0.1 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_cpu_behavior
DEBUG   - DEVICE: Per-CPU smoke test 0...
DEBUG   - DEVICE: Cache behavior check
DEBUG   - DEVICE: CCOUNT/WALCLK ratio 10.409
DEBUG   - DEVICE: CPI = 2.499
DEBUG   - DEVICE: Per-CPU smoke test 1...
DEBUG   - DEVICE: Cache behavior check
DEBUG   - DEVICE: CCOUNT/WALCLK ratio 10.405
DEBUG   - DEVICE: CPI = 2.499
DEBUG   - DEVICE: Per-CPU smoke test 2...
DEBUG   - DEVICE: Cache behavior check
DEBUG   - DEVICE: CCOUNT/WALCLK ratio 10.411
DEBUG   - DEVICE: CPI = 2.499
DEBUG   - DEVICE: Per-CPU smoke test 3...
DEBUG   - DEVICE: Cache behavior check
DEBUG   - DEVICE: CCOUNT/WALCLK ratio 10.413
DEBUG   - DEVICE: CPI = 2.499
DEBUG   - DEVICE: PASS - test_cpu_behavior in 0.2 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_host_ipc
DEBUG   - DEVICE: Simple message send...
DEBUG   - DEVICE: 
DEBUG   - DEVICE: Assertion failed at WEST_TOPDIR/zephyr/tests/boards/intel_adsp/smoke/src/hostipc.c:44: test_host_ipc: WAIT_FOR(cavs_ipc_is_complete(CAVS_HOST_DEV), 10000, k_msleep(1)) is false
DEBUG   - DEVICE: 
DEBUG   - DEVICE: FAIL - test_host_ipc in 0.12 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_clock_calibrate
DEBUG   - DEVICE: 
DEBUG   - DEVICE: Assertion failed at WEST_TOPDIR/zephyr/tests/boards/intel_adsp/smoke/src/hostipc.c:28: ipc_done: (done_flag is true)
DEBUG   - DEVICE: 
DEBUG   - DEVICE: Assertion failed at WEST_TOPDIR/zephyr/tests/boards/intel_adsp/smoke/src/main.c:34: test_clock_calibrate: WAIT_FOR(host_dt != 0, 10000, k_msleep(1)) is false
DEBUG   - DEVICE: done called unexpectedly
DEBUG   - DEVICE: 
DEBUG   - DEVICE: FAIL - test_clock_calibrate in 1.12 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_ipm_cavs_host
DEBUG   - Timed out while monitoring serial output on intel_adsp_cavs25
```
for tests/crypto/tinycrypt:
```
DEBUG   - DEVICE: INFO:cavs-fw:Mapped PCI bar 0 of length 16384 bytes.
DEBUG   - DEVICE: INFO:cavs-fw:Selected output stream 7 (GCAP = 0x1009701)
DEBUG   - DEVICE: INFO:cavs-fw:Mapped PCI bar 4 of length 1048576 bytes.
DEBUG   - DEVICE: INFO:cavs-fw:Detected cAVS 1.8+ hardware
DEBUG   - DEVICE: INFO:cavs-fw:Trimming 16 bytes of extended manifest
DEBUG   - DEVICE: INFO:cavs-fw:Resetting HDA device
DEBUG   - DEVICE: INFO:cavs-fw:Powering down DSP cores
DEBUG   - DEVICE: INFO:cavs-fw:Configuring HDA stream 7 to transfer firmware image
DEBUG   - DEVICE: INFO:cavs-fw:type of mem is <class 'mmap.mmap'>
DEBUG   - DEVICE: INFO:cavs-fw:Mapped 2M huge page at 0x152400000 to contain 94208 bytes of firmware
DEBUG   - DEVICE: INFO:cavs-fw:Filled the buffer descriptor list (BDL) for DMA.
DEBUG   - DEVICE: INFO:cavs-fw:Starting DSP, ADSPCS = 0xf0f
DEBUG   - DEVICE: INFO:cavs-fw:Wait for ROM startup
DEBUG   - DEVICE: INFO:cavs-fw:Sending IPC command, HIPIDR = 0x81004000
DEBUG   - DEVICE: INFO:cavs-fw:Starting DMA, FW_STATUS = 0x5000001
DEBUG   - DEVICE: INFO:cavs-fw:Waiting for firmware handoff, FW_STATUS = 0x5000001
DEBUG   - DEVICE: INFO:cavs-fw:cAVS firmware load complete
DEBUG   - DEVICE: --
DEBUG   - Timed out while monitoring serial output on intel_adsp_cavs25
```
**Environment (please complete the following information):**
 - OS: Linux
 - Commit b433813f5fb5655ee18133dfac176c9e8dfe9a45

edersondisouza, 2022-04-22:
Hi @LixinGuoX ,

You mention that this issue is sporadic ("it may stuck"). How frequent is that? I tried running your command a few times using latest main, and while I got some issues in the "intel_adsp/smoke" ones, none of them was a hang that prevented any further tests from running.

Do you have other examples of this issue?

LixinGuoX, 2022-04-22:
Yes, that may be difficult to reproduce when running light-weight tests, for most of the time intel_adsp/smoke execute fail due to assertion failure. But if we run the whole tests directory, the reproducibility of hang is nearly 100%.
#43828 for reference, thanks.

edersondisouza, 2022-04-27:
No luck either, still couldn't reproduce it =/

Do you have another system to test this issue? Just to exclude some faulty hw, environment, etc.

LixinGuoX, 2022-04-28:
I am sorry to tell you that I have tried on another cavs25 board, this issue happens again. I ran the `tests/boards/intel_adsp/smoke/` for several times and it hung as expected, then tests like `tests/crypto/tinycrypt/` which should pass in normal state failed due to timeout.
I am trying to get a list of tests that would be affected by the hung state of `tests/boards/intel_adsp/smoke/`

edersondisouza, 2022-04-28:
That's sad =/
Just to check, are you testing with latest main?

LixinGuoX, 2022-04-29:
Yes, checked with latest main commit, this issue happens again.
And when get the hung state, rerun the smoke test, it will stuck at starting the second cpu `z_smp_start_cpu(2)`
```
DEBUG   - DEVICE: Running test suite intel_adsp
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_smp_boot_delay
DEBUG   - DEVICE: Launch cpu1
DEBUG   - DEVICE: Launch cpu2

```

LixinGuoX, 2022-05-11:
Here is the list of tests that would be affected by the hung state of `tests/boards/intel_adsp/smoke/`
```
tests/boards/intel_adsp/hda_log/boards.intel_adsp
tests/boards/intel_adsp/mm/boards.intel_adsp.mm
tests/application_development/gen_inc_file/buildsystem.include_file
tests/misc/print_format/printk.format
tests/misc/print_format/printk.format_newlib
tests/drivers/hwinfo/api/drivers.hwinfo.api
tests/drivers/sensor/generic/drivers.sensor.generic
tests/drivers/sensor/generic/drivers.sensor.generic.fpu
tests/drivers/dma/loop_transfer/drivers.dma.loop_transfer
tests/drivers/dma/chan_blen_transfer/drivers.dma.chan_blen_transfer
tests/drivers/dma/scatter_gather/drivers.dma.scatter_gather
tests/drivers/can/utilities/drivers.can.utilities
tests/lib/devicetree/api_ext/libraries.devicetree.api_ext
tests/lib/devicetree/api/libraries.devicetree.api
tests/lib/devicetree/devices/libraries.devicetree.devices
tests/lib/onoff/libraries.onoff
tests/lib/notify/libraries.data_structures
tests/lib/mem_alloc/libraries.libc.minimal.mem_alloc
tests/lib/mem_alloc/libraries.libc.newlib.mem_alloc
tests/lib/mem_alloc/libraries.libc.minimal.mem_alloc_negative_testing
tests/lib/sys_util/lib.sys_util
tests/lib/heap_align/lib.heap_align
tests/lib/json/libraries.encoding.json
tests/lib/sprintf/libraries.libc.sprintf
tests/lib/sprintf/libraries.libc.sprintf_new
tests/lib/fdtable/libraries.os.fdtable
tests/lib/heap/lib.heap
tests/lib/mem_blocks/lib.mem_blocks
tests/lib/time/libraries.libc.time
tests/lib/c_lib/libraries.libc
tests/lib/p4workq/lib.p4wq
tests/lib/newlib/thread_safety/libraries.libc.newlib.thread_safety
tests/lib/newlib/heap_listener/libraries.libc.newlib.heap_listener
tests/subsys/ipc/icmsg_buf/subsys.ipc.icmsg_buf
tests/subsys/logging/log_core_additional/logging.add.async
tests/subsys/logging/log_core_additional/logging.add.sync
tests/subsys/logging/log_core_additional/logging.add.log1
tests/subsys/logging/log_immediate/logging.log_immediate
tests/subsys/logging/log_immediate/logging.log_immediate.clean_output
tests/subsys/logging/log_output/logging.log_output
tests/subsys/logging/log_core/logging.log_core
tests/subsys/logging/log_list/logging.log_list
tests/subsys/logging/log_benchmark/logging.log_benchmark_v1
tests/subsys/logging/log_benchmark/logging.log_benchmark_v2
tests/subsys/logging/log_benchmark/logging.log_benchmark_v2_speed
tests/subsys/logging/log_msg/logging.log_msg
tests/subsys/fs/fs_api/filesystem.api
tests/subsys/cpp/libcxx/cpp.libcxx.newlib
tests/subsys/cpp/libcxx/cpp.libcxx.newlib_nano
tests/subsys/cpp/cxx/cpp.main
tests/subsys/shell/shell_history/shell.history
tests/subsys/shell/shell/shell.core
tests/posix/eventfd/portability.posix.eventfd
tests/posix/eventfd_basic/posix.eventfd_basic
tests/posix/eventfd_basic/posix.eventfd_basic.posix_api
tests/posix/fs/portability.posix.fs
tests/posix/fs/portability.posix.fs.newlib
tests/posix/getopt/shell.getopt
tests/ztest/automain/testing.ztest.automain
tests/ztest/base/testing.ztest.base.verbose_0
tests/ztest/base/testing.ztest.base.verbose_1
tests/ztest/base/testing.ztest.base.verbose_2
tests/ztest/custom_output/testing.ztest.regular_output
tests/ztest/custom_output/testing.ztest.customized_output
tests/ztest/register/testing.ztest.register
tests/ztest/mock/testing.ztest.mock
tests/ztest/error_hook/testing.ztest.error_hook.no_userspace
tests/kernel/obj_tracking/kernel.objects.tracking
tests/kernel/mem_slab/mslab_concept/kernel.memory_slabs.concept
tests/kernel/device/kernel.device
tests/kernel/device/kernel.device.pm
tests/kernel/smp/kernel.multiprocessing.smp
tests/kernel/queue/kernel.queue
tests/kernel/threads/thread_stack/kernel.threads.thread_stack
tests/kernel/events/event_api/kernel.events
tests/kernel/msgq/msgq_api/kernel.message_queue
tests/kernel/msgq/msgq_usage/kernel.message_queue_usage
tests/kernel/fatal/message_capture/kernel.logging.message_capture
tests/kernel/mbox/mbox_usage/kernel.mailbox.usage
tests/kernel/mem_heap/k_heap_api/kernel.k_heap_api
tests/kernel/mem_heap/mheap_api_concept/kernel.memory_heap
tests/kernel/mutex/sys_mutex/system.mutex.nouser
tests/kernel/lifo/lifo_api/kernel.lifo
tests/kernel/lifo/lifo_usage/kernel.lifo.usage
tests/kernel/smp_boot_delay/kernel.multiprocessing.smp_boot_delay
tests/kernel/fifo/fifo_api/kernel.fifo
tests/kernel/workq/work/kernel.work.api
tests/kernel/common/kernel.common
tests/kernel/common/kernel.common.misra
tests/kernel/mp/kernel.multiprocessing
tests/kernel/semaphore/semaphore/kernel.semaphore
tests/kernel/mem_protect/sys_sem/kernel.memory_protection.sys_sem
tests/kernel/mem_protect/sys_sem/kernel.memory_protection.sys_sem.nouser
tests/kernel/profiling/profiling_api/kernel.common.profiling
tests/kernel/spinlock/kernel.multiprocessing.spinlock
tests/kernel/timer/timer_monotonic/kernel.timer.monotonic
tests/crypto/tinycrypt/crypto.tinycrypt
tests/crypto/rand32/crypto.rand32
tests/crypto/rand32/crypto.rand32.random_sw_systimer
tests/crypto/tinycrypt_hmac_prng/crypto.tinycrypt.hmac_prng
tests/crypto/mbedtls/crypto.mbedtls
tests/benchmarks/data_structure_perf/rbtree_perf/benchmark.data_structure_perf.rbtree
tests/benchmarks/data_structure_perf/dlist_perf/benchmark.data_structure_perf.dlist
```

edersondisouza, 2022-05-18:
Hi @LixinGuoX ,

Can you check if my draft PR  #45777 helps to mitigate this issue? If so, I'll remove the draft status there.

LixinGuoX, 2022-05-19:
Hi @edersondisouza,
Yes, this PR may help to mitigate this issue, cavs25 hangs at `test_ipm_cavs_host `will not affect other testcases like tinycrypt.
But if hangs at `test_smp_boot_delay` like below will still affect other testcases.
```
DEBUG   - DEVICE: Running TESTSUITE intel_adsp
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_smp_boot_delay
DEBUG   - DEVICE: Launch cpu1
DEBUG   - Timed out while monitoring serial output on intel_adsp_cavs25

```

edersondisouza, 2022-05-19:
In this case, does it hang after the "load complete" message, as before?

LixinGuoX, 2022-05-20:
Yes, that's what we have in this case.
```
DEBUG   - DEVICE: Killed
DEBUG   - DEVICE: INFO:cavs-fw:Mapped PCI bar 0 of length 16384 bytes.
DEBUG   - DEVICE: INFO:cavs-fw:Selected output stream 7 (GCAP = 0x1009701)
DEBUG   - DEVICE: INFO:cavs-fw:Mapped PCI bar 4 of length 1048576 bytes.
DEBUG   - DEVICE: INFO:cavs-fw:Detected cAVS 1.8+ hardware
DEBUG   - DEVICE: INFO:cavs-fw:Trimming 16 bytes of extended manifest
DEBUG   - DEVICE: INFO:cavs-fw:Resetting HDA device
DEBUG   - DEVICE: INFO:cavs-fw:Powering down DSP cores
DEBUG   - DEVICE: INFO:cavs-fw:Configuring HDA stream 7 to transfer firmware image
DEBUG   - DEVICE: INFO:cavs-fw:type of mem is <class 'mmap.mmap'>
DEBUG   - DEVICE: INFO:cavs-fw:Mapped 2M huge page at 0x15b000000 to contain 106496 bytes of firmware
DEBUG   - DEVICE: INFO:cavs-fw:Filled the buffer descriptor list (BDL) for DMA.
DEBUG   - DEVICE: INFO:cavs-fw:Starting DSP, ADSPCS = 0xf0f
DEBUG   - DEVICE: INFO:cavs-fw:Wait for ROM startup
DEBUG   - DEVICE: INFO:cavs-fw:Sending IPC command, HIPIDR = 0x81004000
DEBUG   - DEVICE: INFO:cavs-fw:Starting DMA, FW_STATUS = 0x5000001
DEBUG   - DEVICE: INFO:cavs-fw:Waiting for firmware handoff, FW_STATUS = 0x5000001
DEBUG   - DEVICE: INFO:cavs-fw:cAVS firmware load complete
DEBUG   - DEVICE: --
DEBUG   - DEVICE: Running TESTSUITE intel_adsp
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_smp_boot_delay
DEBUG   - DEVICE: Launch cpu1
DEBUG   - DEVICE: Launch cpu2
DEBUG   - DEVICE: Launch cpu3
DEBUG   - Timed out while monitoring serial output on intel_adsp_cavs25
```

edersondisouza, 2022-05-20:
So, when next test starts, you still see:

> DEBUG   - DEVICE: INFO:cavs-fw:Waiting for firmware handoff, FW_STATUS = 0x5000001
DEBUG   - DEVICE: INFO:cavs-fw:cAVS firmware load complete
DEBUG   - DEVICE: --
DEBUG   - Timed out while monitoring serial output on intel_adsp_cavs25

?

Or are you saying that after first hang, more hangs will follow, but without any pattern? Some tests even start while others not?

LixinGuoX, 2022-05-20:
Yes, can't see any log
```
DEBUG   - DEVICE: Killed
DEBUG   - DEVICE: INFO:cavs-fw:Mapped PCI bar 0 of length 16384 bytes.
DEBUG   - DEVICE: INFO:cavs-fw:Selected output stream 7 (GCAP = 0x1009701)
DEBUG   - DEVICE: INFO:cavs-fw:Mapped PCI bar 4 of length 1048576 bytes.
DEBUG   - DEVICE: INFO:cavs-fw:Detected cAVS 1.8+ hardware
DEBUG   - DEVICE: INFO:cavs-fw:Trimming 16 bytes of extended manifest
DEBUG   - DEVICE: INFO:cavs-fw:Resetting HDA device
DEBUG   - DEVICE: INFO:cavs-fw:Powering down DSP cores
DEBUG   - DEVICE: INFO:cavs-fw:Configuring HDA stream 7 to transfer firmware image
DEBUG   - DEVICE: INFO:cavs-fw:type of mem is <class 'mmap.mmap'>
DEBUG   - DEVICE: INFO:cavs-fw:Mapped 2M huge page at 0x15b000000 to contain 172032 bytes of firmware
DEBUG   - DEVICE: INFO:cavs-fw:Filled the buffer descriptor list (BDL) for DMA.
DEBUG   - DEVICE: INFO:cavs-fw:Starting DSP, ADSPCS = 0xf0f
DEBUG   - DEVICE: INFO:cavs-fw:Wait for ROM startup
DEBUG   - DEVICE: INFO:cavs-fw:Sending IPC command, HIPIDR = 0x81004000
DEBUG   - DEVICE: INFO:cavs-fw:Starting DMA, FW_STATUS = 0x5000001
DEBUG   - DEVICE: INFO:cavs-fw:Waiting for firmware handoff, FW_STATUS = 0x5000001
DEBUG   - DEVICE: INFO:cavs-fw:cAVS firmware load complete
DEBUG   - Timed out while monitoring serial output on intel_adsp_cavs25
```

edersondisouza, 2022-05-28:
Hi @LixinGuoX - I updated my draft PR #45777 - can you check if it makes things better? I'm afraid we're entering trial and error territory here, and things appeared more stable with that patch - even if issue is not exactly fixed by it.

LixinGuoX, 2022-05-30:
Hi @edersondisouza, seems like this PR helps to reproduce this issue, now the reproducibility is 100%, with some additional error log as below:
```
DEBUG   - DEVICE: Killed
DEBUG   - DEVICE: INFO:cavs-fw:Mapped PCI bar 0 of length 16384 bytes.
DEBUG   - DEVICE: INFO:cavs-fw:Selected output stream 7 (GCAP = 0x1009701)
DEBUG   - DEVICE: INFO:cavs-fw:Mapped PCI bar 4 of length 1048576 bytes.
DEBUG   - DEVICE: INFO:cavs-fw:Detected cAVS 1.8+ hardware
DEBUG   - DEVICE: INFO:cavs-fw:Trimming 16 bytes of extended manifest
DEBUG   - DEVICE: INFO:cavs-fw:Resetting HDA device
DEBUG   - DEVICE: INFO:cavs-fw:Stalling and Resetting DSP cores
DEBUG   - DEVICE: INFO:cavs-fw:Powering down DSP cores
DEBUG   - DEVICE: INFO:cavs-fw:Configuring HDA stream 7 to transfer firmware image
DEBUG   - DEVICE: INFO:cavs-fw:type of mem is <class 'mmap.mmap'>
DEBUG   - DEVICE: INFO:cavs-fw:Mapped 2M huge page at 0x170400000 to contain 106496 bytes of firmware
DEBUG   - DEVICE: INFO:cavs-fw:Filled the buffer descriptor list (BDL) for DMA.
DEBUG   - DEVICE: INFO:cavs-fw:Starting DSP, ADSPCS = 0x0
DEBUG   - DEVICE: INFO:cavs-fw:Wait for ROM startup
DEBUG   - DEVICE: INFO:cavs-fw:Sending IPC command, HIPIDR = 0x81004000
DEBUG   - DEVICE: INFO:cavs-fw:Starting DMA, FW_STATUS = 0x5000001
DEBUG   - DEVICE: INFO:cavs-fw:Waiting for firmware handoff, FW_STATUS = 0x5000001
DEBUG   - DEVICE: INFO:cavs-fw:cAVS firmware load complete
DEBUG   - DEVICE: --
DEBUG   - DEVICE: Running TESTSUITE intel_adsp
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_smp_boot_delay
DEBUG   - DEVICE: Launch cpu1
DEBUG   - DEVICE: Launch cpu2
DEBUG   - DEVICE: Launch cpu3
DEBUG   - DEVICE: PASS - test_smp_boot_delay in 0.61 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_cpu_halt
DEBUG   - DEVICE: halt/restart core 1...
DEBUG   - DEVICE: halt/restart core 2...
DEBUG   - DEVICE: halt/restart core 3...
DEBUG   - DEVICE: PASS - test_cpu_halt in 0.452 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_post_boot_ipi
DEBUG   - DEVICE: PASS - test_post_boot_ipi in 0.1 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_cpu_behavior
DEBUG   - DEVICE: Per-CPU smoke test 0...
DEBUG   - DEVICE: Cache behavior check
DEBUG   - DEVICE: CCOUNT/WALCLK ratio 10.417
DEBUG   - DEVICE: CPI = 2.499
DEBUG   - DEVICE: Per-CPU smoke test 1...
DEBUG   - DEVICE: Cache behavior check
DEBUG   - DEVICE: CCOUNT/WALCLK ratio 10.413
DEBUG   - DEVICE: CPI = 2.499
DEBUG   - DEVICE: Per-CPU smoke test 2...
DEBUG   - DEVICE: Cache behavior check
DEBUG   - DEVICE: CCOUNT/WALCLK ratio 10.410
DEBUG   - DEVICE: CPI = 2.499
DEBUG   - DEVICE: Per-CPU smoke test 3...
DEBUG   - DEVICE: Cache behavior check
DEBUG   - DEVICE: CCOUNT/WALCLK ratio 10.400
DEBUG   - DEVICE: CPI = 2.499
DEBUG   - DEVICE: PASS - test_cpu_behavior in 0.2 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_host_ipc
DEBUG   - DEVICE: Simple message send...
DEBUG   - DEVICE:
DEBUG   - DEVICE: Assertion failed at WEST_TOPDIR/zephyr/tests/boards/intel_adsp/smoke/src/hostipc.c:44: test_host_ipc: WAIT_FOR(cavs_ipc_is_complete(CAVS_HOST_DEV), 10000, k_msleep(1)) is false
DEBUG   - DEVICE:
DEBUG   - DEVICE: FAIL - test_host_ipc in 0.12 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_clock_calibrate
DEBUG   - DEVICE:
DEBUG   - DEVICE: Assertion failed at WEST_TOPDIR/zephyr/tests/boards/intel_adsp/smoke/src/hostipc.c:28: ipc_done: (done_flag is true)
DEBUG   - DEVICE: done called unexpectedly
DEBUG   - DEVICE: ASSERTION FAIL [0] @ WEST_TOPDIR/zephyr/kernel/sched.c:1717
DEBUG   - DEVICE: aborting essential thread 0x9e021338
DEBUG   - DEVICE: @ WEST_TOPDIR/zephyr/lib/os/assert.c:43
DEBUG   - DEVICE: E: >>> ZEPHYR FATAL ERROR 4: Kernel panic on CPU 0
DEBUG   - DEVICE: E: Current thread: 0x9e021338 (idle 00)
DEBUG   - DEVICE: E: Halting system
DEBUG   - DEVICE: FAIL - test_clock_calibrate in 1.7 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_ipm_cavs_host
```

edersondisouza, 2022-06-01:
Err... that wasn't the kind of stability I was expecting =D

Nevertheless, I spotted some mistakes with my previous patch, can you check again with my updated PR, please? From 100 runs, I've seen it hang only twice, and none of them propagated to the next test. Let's hope things get better on your side too... =D

LixinGuoX, 2022-06-01:
Yes, it works on my side, the hang state will not affect other tests anymore.

edersondisouza, 2022-06-11:
@LixinGuoX as the #45777 was merged, can you check if this bug is still happening, and if not, close this issue? If it still happening, please, provide more information.

LixinGuoX, 2022-06-13:
Yes, we are not able to find this issue in our daily testing since the PR merged, I think it's time to close this issue.
Thank you @edersondisouza !

closed 2022-06-13 by LixinGuoX