zephyrproject-rtos/zephyr #44931: Bluetooth: Samples: broadcast_audio_source

carlst, 2022-04-18:
**Describe the bug**
`samples/bluetooth/broadcast_audio_source` crashes when FPU enabled in `prj.conf` with:

`CONFIG_NEWLIB_LIBC=y`
`CONFIG_FPU=y`
`CONFIG_FP16=y`
`CONFIG_FP_HARDABI=y`

FPU is used e.g. by `modules/liblc3codec`

Please also mention any information which could help others to understand
the problem you're facing:

 - What have you tried to diagnose or workaround this issue?
Stack size in `subsys/bluetooth/host/hci_core.c` seems carefully tuned (to 640B), though increasing `tx_thread_stack`  _similar_ to that mentioned in link below (e.g. 640B+256B) seems to work:
https://docs.zephyrproject.org/latest/kernel/services/other/float.html?highlight=fpu

**To Reproduce**
Steps to reproduce the behavior:
1. Use nRF52840DK
2. build and run `samples/bluetooth/broadcast_audio_source`
3. sample crashes immediately

**Expected behavior**
Sample should not crash.

**Impact**
Low priority

**Environment (please complete the following information):**
 - Commit SHA: 13f382287ad050d9600362bffca8bc03b93bbbe4

Thalley, 2022-04-19:
@carlst You mentioned stack sizes. Did you see a stack overflow? 

Can you provide a crash log of some sort?

carlst, 2022-04-19:
Crash is almost immediate and there is little info. The instruction address (r15/pc) looks suspicious.
[broadcast_audio_source_20220419.txt](https://github.com/zephyrproject-rtos/zephyr/files/8516660/broadcast_audio_source_20220419.txt)

Removing  `CONFIG_BT_DEBUG_LOG=y` did not affect the crash.

I am using Segger Embedded Studio on Windows 10.  Have not yet found a way to copy Call Stack out of SES, though only info shown is `z_arm_usage_fault()`

Previously tried several compiler switches to attempt a stack guard but without visible success:

`set(CMAKE_C_FLAGS "${CMAKE_C_FLAGS} \`
 `-fstack-protector-all -fstack-protector-strong -fstack-check \`
 `-g -O0 \`
 `-mfloat-abi=hard \`
 `-mfp16-format=none \`
 `-fsingle-precision-constant \`
`")`

Thalley, 2022-04-20:
> Crash is almost immediate and there is little info. The instruction address (r15/pc) looks suspicious. [broadcast_audio_source_20220419.txt](https://github.com/zephyrproject-rtos/zephyr/files/8516660/broadcast_audio_source_20220419.txt)
> 

The log does contain the following line
```
[1;32mrtt:~$ [m[7D[J[7D[J[00:01:11.934,326] [1;31m<err> os: >>> ZEPHYR FATAL ERROR 2: Stack overflow on CPU 0[0m
```
So it likely just a stack overflow :)

Thalley, 2022-04-20:
@carlst Can you please double check your commit SHA? The build number from the SHA you provided does not match the log you've sent.

carlst, 2022-04-20:
SES does not display the log when `z_arm_usage_fault()` is hit. There was no obvious way to copy the registers from SES at that point, but I have since found a copy op and a register dump at  `z_arm_usage_fault()` is included in the attached file below. The stack pointer appears to refer to the stack of a system work queue when that fault occurs. Not sure if that is meaningful; if corruption is caused by save/restore of FPU registers, that may be one place to check.

The log is displayed after continuing execution of the program, so although that may be due to background thread displaying the log, I did not trust the log 100%. However, it was all the debug info available at the time. Zephyr did not support Windows natively when I began using SES. Perhaps command-line tools (which may be more transparent than SES) may work now.

Please see registers captured at time of fault and new log below.
These should match SHA: 1135796089c7ac1191a27c5ed9e5792105a83184
Sorry if Windows editor is mangling text encoding in the log.

[broadcast_audio_source_registers_20220420.txt](https://github.com/zephyrproject-rtos/zephyr/files/8523521/broadcast_audio_source_registers_20220420.txt)

[broadcast_audio_source_20220420.txt](https://github.com/zephyrproject-rtos/zephyr/files/8523524/broadcast_audio_source_20220420.txt)

Thalley, 2022-04-20:
Thanks. I'm able to reproduce it (even on the current `main` branch), but having some trouble figuring out the issue, so I'll need to dig deeper :)

carlst, 2022-04-20:
In a system with FPU, a stack adjustment is made (to make space for saving FP registers).
With CONFIG_FPU_SHARING (default mode if CONFIG_FPU used), this is automatic on context switch.

To see this, in `arch\arm\core\aarch32\thread.c`, set a breakpoint in `z_arm_thread_stack_info_adjust()` for line:
`thread->stack_info.start +=`
					`FP_GUARD_EXTRA_SIZE;`

The system here saves 96 bytes. If `tx_thread_stack` in `subsys\bluetooth\host\hci_core.c` is increased by 96 bytes, the crash does not occur.

Additionally, if the default `tx_thread_stack` (640B here) is used, but `prj.conf` is changed as follows:
`...`
`CONFIG_NEWLIB_LIBC`
`#CONFIG_FPU=y`
`#CONFIG_FP16=y`
`#CONFIG_FP_HARDABI=y`

the crash does not occur. 

It is unclear to me how the original stack size was computed, but adding space for FP register storage (see `MPU_GUARD_ALIGN_AND_SIZE_FLOAT` in `thread.c`) eliminates the crash.

Thalley, 2022-04-21:
I can reproduce it here with just setting `CONFIG_FPU=y`. Some other people are seeing related issues with ISO, so may be a memory corruption somewhere, and not actually a stack overflow. 

We'll keep investigating :) 

Ping @cvinayak

Thalley, 2022-04-21:
@carlst Added a PR to increase the default TX stack size

carlst, 2022-04-21:
Yes, that could be.

One other item: in addition to space for FP register storage (96B in this case), there appears to be a stack guard region defined in:
`arch\arm\core\aarch32\mpu\arm_core_mpu.c`, `z_arm_configure_dynamic_mpu_regions()`.

If `CONFIG_FPU=n`, it is 32B.
If `CONFIG_FPU=y`, it is 128B.

closed 2022-04-22 by carlescufi