zephyrproject-rtos/zephyr #44942: Dictionary logging to automatically remove strings from the installed image

UnicycleBloke, 2022-04-19:
I have been trying to use the dictionary-based logging feature because my target device has only 64KB ROM. My client creates a log in the flash which can be examined if a device is returned from the field. They currently fill half the space with strings used at run time, and with those same strings stored in the log. A dictionary solution which removes the strings entirely is highly desirable. This is what they assumed the Zephyr feature offers. It doesn't. It should. 

When I first looked at this, the strings could not be removed at all, so the feature only reduced the amount of data in the log. This is only half a solution. A recent merge means that the strings are now stored in a section called "log_strings_sections" which *can* be removed, presumably with objcopy. Sadly, that section is placed my the linker in the middle of the ROM. Removing it has no effect on the footprint of the installed image. 

A secondary issue was that removing the section with objcopy led to a warning about an "empty loadable section - was this your intention?", or words to that effect. That's disconcerting.

A tertiary issue is that there was little useful documentation about any of this potentially wonderful feature. There are just some comments that the feature exists, and an example program.

The simplest solution might be to reorder the standard linker script so that "log_strings_sections" is last in the ROM image, or not "loadable" (not sure of the right term), or outside of the physical ROM. The ELF contains a bunch of DWARF debug sections which are not installed on the device. Can we achieve something like that?

I wrote my own dictionary-based logger without much difficultly, but my client wants a Zephyr-only solution. I used a hash for the string IDs rather than an address/reference/offset (whatever Zephyr does). This meant that my dictionary section was simply optimised out of the image. Granted this is not super helpful if you want to read the dictionary, but I figured I could grab it from an unoptimised ELF or something - we'll see.

nordic-krch, 2022-04-19:
@nashif, @carlescufi do you know who can help from linker side?

carlescufi, 2022-04-19:
> @nashif, @carlescufi do you know who can help from linker side?

@tejlmand or maybe @dcpleung

optical-o, 2023-03-29:
Any progress on this ? This would be really helpful for the footprint reduction.

clamattia, 2023-07-10:
We also need this feature.
Are there any plans to implement this in the future?
What is currently the biggest blocker/challenge? I would expect, that if the strings are already in their own section, it should be possible to configure it in a way, such that they do not end up in the final binary?

dcpleung, 2023-07-14:
A manual way is to create a linker script where the log format string section is outside of usable memory ranges. The log strings still have addresses associated with them for parsing, but the section does not (and actually cannot) get copied onto the ROM since it's outside of usable memory ranges. Depending on the tool to generate the final flash-able binary, you can do a objcopy to remove the section, or the tool can generate a .bin file without the section. Of course, one needs to make sure none of the format strings are being used in other part of the app.

The problem with doing this automatically is where to put the log format string section. There is no generic/universal way of doing this in generic linker script in the tree because there is no way of universally declare a hole in the address space as each boards/platforms has their own different address space.

clamattia, 2023-07-17:
> A manual way is to create a linker script where the log format string section is outside of usable memory ranges. The log strings still have addresses associated with them for parsing, but the [...]

Thanks for the explanation.

clamattia, 2023-07-17:
Might it be possible to solve it in a generic/universal way with the processor instead of the linker?

dcpleung, 2023-07-17:
Currently, the memory addresses of the strings act as the identifiers of those strings. So the strings need to be present during linking (and thus assigned addresses). Using these addresses as the identifiers is (possibly) the least intrusive way of doing dictionary logging, but does require some post processing for the whole end-to-end solution.

The other possibility is to pre-hash the strings before compiling the source code. This would require us to write a source code parser (even after code is preprocessed by the compiler) to identify those strings and transform (i.e. modify) the source code file during build (by replacing call to logging subsystem with hashed version). This gets complicated and will require constant maintenance if logging subsystem are modified.

nordic-krch, 2023-10-26:
PR is opened. @clamattia @UnicycleBloke can you take a look at #64424 if that works for you?

optical-o, 2023-10-26:
@nordic-krch Could you please expand on what exactly is meant by formatting the string?  Or present me an example of it. 

I presume that is done only in rare cases? Does this include adding a timestamp in front of the log?

nordic-krch, 2023-10-26:
@optical-o Formatting a string is a process of analyzing string content in search for format specifiers and replacing format specifiers with arguments provided to the string, e.g. `LOG_INF("string %d %s", 100, str)` is a string with 2 arguments (and 2 format specifier in `fmt` string).
In dictionary logging string is not formatted in the target but rather pointer to that string and arguments are sent to the host and host is parsing this data. In that case, target is never looking into the string but only needs a pointer to that string.

optical-o, 2023-10-26:
Just one last question. 
The LINKER_DEVNULL_MEMORY will be excluded during the size calculation? So we can have firmware size + devnull to be larger than the target flash memory. Just to be sure that the firmware is buildable and will not fail on some build system checks.

Otherwise, it sounds perfect. Great work.

nordic-krch, 2023-10-26:
DEVNULL_MEMORY create additional "virtual" memory region of configurable size (default 256k) next to ROM. So if code+devnull content exceed "real" flash size it won't fail to build.

clamattia, 2023-10-27:
> PR is opened. @clamattia @UnicycleBloke can you take a look at #64424 if that works for you?

Very nice 💪 I am not so familiar with linker scripts. As far as I could tell, it is exactly, what we would have needed. Great work.

closed 2023-10-27 by carlescufi

optical-o, 2024-01-11:
How do you configure this feature when the dictionary output is dependent on the LOG_OUTPUT ?

nordic-krch, 2024-01-11:
It works with UART dictionary frontend, this configuration: https://github.com/zephyrproject-rtos/zephyr/blob/c1f037fc5b38e9de76cd4f6be6c51f7e4e2c4a9c/samples/subsys/logging/dictionary/sample.yaml#L30

optical-o, 2024-01-12:
Is there any reason why would the stripping not work when the user uses only the log_output_dict part? 

Most of the log_backends do have the option to use the dictionary output. Or that is a different functionality that the UART Dictionary frontend ?

nordic-krch, 2024-01-12:
@optical-o you are right. I opened #67510 to allow string stripping for dictionary backends.

PR #64424 (2023-10-27): Log strings removal from the binary
PR introduces option to remove logging strings from binary. It can be used if logging backend/frontend does not format the string but only uses the address of the string (dictionary based logging).
It is done by creating a memory region (devnull region) adjacent to the ROM region. Then section with logging strings is put in that region with `(COPY)` property so that it ends in elf file but not in binary.

Support is initially added to Cortex-M and RISCV but it should be fairly easy to extend other linker scripts.

`samples/subsys/logging/dictionary/sample.lo [... cut]