zephyrproject-rtos/zephyr #44947: cmsis_dsp: matrix: error during building libraries.cmsis_dsp.matrix.unary_f64 for qemu_cortex_m3

gopiotr, 2022-04-19:
**Describe the bug**
Test `libraries.cmsis_dsp.matrix.unary_f64` from `tests/lib/cmsis_dsp/matrix` cannot be built for `qemu_cortex_m3` platform.

**To Reproduce**
Steps to reproduce the behavior:
1. Run Twister by command:
```
./scripts/twister --inline-logs -v -v --overflow-as-error -p qemu_cortex_m3 -T tests/lib/cmsis_dsp/matrix -s tests/lib/cmsis_dsp/matrix/libraries.cmsis_dsp.matrix.unary_f64
```
2. Observe output logs - especially those from building process

**Expected behavior**
Test should be able to build on `qemu_cortex_m3` platform, or limits in test scenario definition (in `testcase.yaml` file) should be changed to exclude this platform.

**Impact**
Test introduce misleading information, that `libraries.cmsis_dsp.matrix.unary_f64` can be run on `qemu_cortex_m3`, but in fact it can't be.

**Logs and console output**
```
[ 94%] Linking C executable zephyr_pre0.elf
/home/redbeard/zephyr-sdk-0.13.1/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/10.3.0/../../../../arm-zephyr-eabi/bin/ld.bfd: zephyr_pre0.elf section `rodata' will not fit in region `FLASH'
/home/redbeard/zephyr-sdk-0.13.1/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/10.3.0/../../../../arm-zephyr-eabi/bin/ld.bfd: region `FLASH' overflowed by 33660 bytes
collect2: error: ld returned 1 exit status
make[2]: *** [zephyr/CMakeFiles/zephyr_pre0.dir/build.make:116: zephyr/zephyr_pre0.elf] Error 1
make[1]: *** [CMakeFiles/Makefile2:2968: zephyr/CMakeFiles/zephyr_pre0.dir/all] Error 2
make: *** [Makefile:91: all] Error 2
```

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: zephyr-sdk-0.13.1
 - Commit SHA: b029bfd65c2c4cf0dfb1ffaee381c9236da0b7e2

stephanosio, 2022-04-19:
You are explicitly running twister with `--overflow-as-error`.  Note that the default behaviour is to automatically skip the platforms that do not have enough resources to run this platform.

This is not a bug.

stephanosio, 2022-04-19:
Closing this once because, unlike #44948, `qemu_cortex_m3` is not one of the listed integration platforms for this test.

closed 2022-04-19 by stephanosio

stephanosio, 2022-04-20:
> also by set requirements like `min_flash` or `min_ram` (and those requirements are used for all platforms)

`min_flash` and `min_ram` are already set for these tests. The problem is that `min_flash` and `min_ram` are by no means a fundamental solution.

Every platform has different resource requirements; for instance, `min_ram` of 64KB may be enough to allow a test to run on platform A, but it may not be enough for platform B because, for instance, its HAL takes up a lot more RAM than that of the platform A.

Similarly, increasing `min_ram` to a higher value may hinder the test from running on some platforms that are actually capable of running the test, hence reducing test coverage.

gopiotr, 2022-04-21:
I understood. Thank you for your answer :)