zephyrproject-rtos/zephyr #44948: cmsis_dsp: transofrm: error during building cf64.fpu and rf64.fpu for mps2_an521_remote

gopiotr, 2022-04-19:
**Describe the bug**
Tests `libraries.cmsis_dsp.transform.cf64.fpu` and `libraries.cmsis_dsp.transform.rf64.fpu` from `tests/lib/cmsis_dsp/transform` cannot be built for `mps2_an521_remote` platform.

**To Reproduce**
Steps to reproduce the behavior:
1. Run Twister by command:
```
./scripts/twister --inline-logs -v -v --overflow-as-error -p mps2_an521_remote -T tests/lib/cmsis_dsp/transform -s tests/lib/cmsis_dsp/transform/libraries.cmsis_dsp.transform.cf64.fpu -s tests/lib/cmsis_dsp/transform/libraries.cmsis_dsp.transform.rf64.fpu
```
2. Observe output logs - especially those from building process

**Expected behavior**
Test should be able to build on `mps2_an521_remote` platform, or limits in test scenario definition (in `testcase.yaml` file) should be changed to exclude this platform (then also `integration_platform` entry should be changed).

**Impact**
Test introduce misleading information, that `cf64.fpu` and `rf64.fpu` tests can be run on `mps2_an521_remote`, but in fact they can't be.

**Logs and console output**
From `libraries.cmsis_dsp.transform.cf64.fpu`:
```
[ 90%] Linking C executable zephyr_pre0.elf
/home/redbeard/zephyr-sdk-0.13.1/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/10.3.0/../../../../arm-zephyr-eabi/bin/ld.bfd: zephyr_pre0.elf section `rodata' will not fit in region `FLASH'
/home/redbeard/zephyr-sdk-0.13.1/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/10.3.0/../../../../arm-zephyr-eabi/bin/ld.bfd: region `FLASH' overflowed by 444604 bytes
collect2: error: ld returned 1 exit status
make[2]: *** [zephyr/CMakeFiles/zephyr_pre0.dir/build.make:123: zephyr/zephyr_pre0.elf] Error 1
make[1]: *** [CMakeFiles/Makefile2:3147: zephyr/CMakeFiles/zephyr_pre0.dir/all] Error 2
make: *** [Makefile:91: all] Error 2
```

From `libraries.cmsis_dsp.transform.rf64.fpu`:
```
[ 90%] Linking C executable zephyr_pre0.elf
/home/redbeard/zephyr-sdk-0.13.1/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/10.3.0/../../../../arm-zephyr-eabi/bin/ld.bfd: zephyr_pre0.elf section `rodata' will not fit in region `FLASH'
/home/redbeard/zephyr-sdk-0.13.1/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/10.3.0/../../../../arm-zephyr-eabi/bin/ld.bfd: region `FLASH' overflowed by 44116 bytes
collect2: error: ld returned 1 exit status
make[2]: *** [zephyr/CMakeFiles/zephyr_pre0.dir/build.make:123: zephyr/zephyr_pre0.elf] Error 1
make[1]: *** [CMakeFiles/Makefile2:3147: zephyr/CMakeFiles/zephyr_pre0.dir/all] Error 2
make: *** [Makefile:91: all] Error 2
```

**Environment (please complete the following information):**
- OS: Linux
- Toolchain: zephyr-sdk-0.13.1
- Commit SHA: b029bfd65c2c4cf0dfb1ffaee381c9236da0b7e2

stephanosio, 2022-04-19:
Same as https://github.com/zephyrproject-rtos/zephyr/issues/44947#issuecomment-1102671532

gopiotr, 2022-04-19:
Okay, so is there any sense to keep in `integration_platform` entry with `mps2_an521_remote` platform?

stephanosio, 2022-05-26:
will fix this after https://github.com/zephyrproject-rtos/zephyr/issues/45319 by replacing `mps2_an521_remote` with `mps2_an386`.