zephyrproject-rtos/zephyr #44956: Deprecate the old spi_cs_control fields

bbilas, 2022-04-19:
## Introduction

Deprecate the old spi_cs_control fields such as `gpio_dev, gpio_pin and gpio_dt_flags` in favor of using `gpio_dt_spec` gpio member.

### Proposed change

Deprecate the `gpio_dev, gpio_pin and gpio_dt_flags` members from `spi_cs_control` struct.

### Proposed change (Detailed)

https://github.com/zephyrproject-rtos/zephyr/pull/39338

### Dependencies

All in-tree drivers.

### Concerns and Unresolved Questions

N/a

bbilas, 2022-04-26:
@henrikbrixandersen Should this issue be still opened since https://github.com/zephyrproject-rtos/zephyr/pull/39338 is already merged?

henrikbrixandersen, 2022-04-26:
> @henrikbrixandersen Should this issue be still opened since #39338 is already merged?

No, it should be closed. Thanks.

closed 2022-04-26 by henrikbrixandersen