zephyrproject-rtos/zephyr #44992: doc: Update Google Analytics tracking to use Google Analytics 4

dbkinder, 2022-04-20:
Zephyr docs use Google Analytics "Universal Analytics" to collect tracking information.
 
Google Analytics 4 is the next-generation measurement solution, and is replacing Universal Analytics. On July 1, 2023, Universal Analytics properties will stop processing new hits. They recommend that you complete your move to Google Analytics 4 before then. 

See https://support.google.com/analytics/answer/11583528 for more information.

stephanosio, 2022-04-20:
References:
* https://github.com/zephyrproject-rtos/zephyr/pull/16666
* https://github.com/zephyrproject-rtos/zephyr/commit/aa9b602e004176d3c995491e1736c6318f6cbc71

stephanosio, 2022-04-20:
NOTE: LF will handle this.