zephyrproject-rtos/zephyr #44997: zcbor build error when ZCBOR_VERBOSE is set

lairdjm, 2022-04-20:
**Describe the bug**
Upon trying to build smp_svr, where ZCBOR_VERBOSE is set by default, the build fails with errors like:
```
In file included from /tmp/aa/modules/lib/zcbor/include/zcbor_decode.h:12,
                 from /tmp/aa/modules/lib/zcbor/src/zcbor_decode.c:11:
/tmp/aa/modules/lib/zcbor/src/zcbor_decode.c:889:24: error: expected ')' before 'PRIuFAST32'
  889 |  zcbor_print("Found %" PRIuFAST32 " elements.\r\n", max_decode);
      |                        ^~~~~~~~~~
/tmp/aa/modules/lib/zcbor/include/zcbor_common.h:50:33: note: in definition of macro 'zcbor_print'
   50 | #define zcbor_print(...) printk(__VA_ARGS__)
      |                                 ^~~~~~~~~~~
/tmp/aa/modules/lib/zcbor/src/zcbor_decode.c:889:14: warning: spurious trailing '%' in format [-Wformat=]
  889 |  zcbor_print("Found %" PRIuFAST32 " elements.\r\n", max_decode);
      |              ^~~~~~~~~
/tmp/aa/modules/lib/zcbor/include/zcbor_common.h:50:33: note: in definition of macro 'zcbor_print'
   50 | #define zcbor_print(...) printk(__VA_ARGS__)
      |                                 ^~~~~~~~~~~
/tmp/aa/modules/lib/zcbor/src/zcbor_decode.c:889:21: note: format string is defined here
  889 |  zcbor_print("Found %" PRIuFAST32 " elements.\r\n", max_decode);
      |                     ^
[191/213] Building C object zephyr/kernel/CMakeFiles/kernel.dir/system_work_q.c.obj
ninja: build stopped: subcommand failed.
```

**Expected behavior**
No build errors

**Impact**
Annoyance

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: Zephyr SDK 0.14.0
 - Commit SHA or Version used: fc88ddf59cc2814ee88ed39a60e4673d3a13a2a6

lairdjm, 2022-04-20:
@oyvindronningstad FYI

oyvindronningstad, 2022-04-21:
cc @de-nordic

oyvindronningstad, 2022-04-21:
@lairdjm I cannot reproduce. Which board are you using? Can you confirm that you are trying to build `samples/subsys/mgmt/mcumgr/smp_svr/`?

lairdjm, 2022-04-21:
I can't remember the exact command or board I was building for yesterday but I'm able to reproduce it with `cmake -GNinja -DBOARD=nrf52840dk_nrf52840 -DOVERLAY_CONFIG="overlay-shell.conf;overlay-shell-mgmt.conf" -DCONFIG_ZCBOR_VERBOSE=y ..` whatever command I was using yesterday, it was being enabled by default and I had to disable it using menuconfig.

oyvindronningstad, 2022-04-21:
Thanks, I can reproduce it if I enable ZCBOR_VERBOSE explicitly.

It seems `zephyr/lib/libc/minimal/include/inttypes.h` is missing `PRIuFAST32` and `PRIxFAST32`.

PR #45017 (2022-04-28): modules: zcbor: kconfig: Disallow ZCBOR_VERBOSE with MINIMAL_LIBC
Because Minimal LIBC doesn't have PRIuFAST32 and PRIxFAST32 which is
needed when printing traces in zcbor.

Fixes #44997
closed 2022-04-28 by carlescufi