zephyrproject-rtos/zephyr #44998: SMP shell exec command causes BLE stack breakdown if buffer size is too small to hold response

lairdjm, 2022-04-20:
**Describe the bug**
If a `shell exec` SMP command is sent over bluetooth with the default 4.0 values i.e. no DLE, 23 byte MTU, small buffers etc. for something like the `help` command, the bluetooth stack emits a few errors about being unable to allocate buffers, but seems to have broken the stack and there is a phantom bluetooth connection and this then prevents any future connections.

Output:
```
uart:~$ [00:00:02.865,478] <inf> smp_bt_sample: Connected
[00:00:03.091,674] <err> bt_conn: Unable to allocate TX context
[00:00:03.092,193] <err> bt_conn: Unable to allocate TX context
[00:00:03.092,681] <err> bt_conn: Unable to allocate TX context
[00:00:02.865,478] <inf> smp_bt_sample: Connected
[00:00:03.091,674] <err> bt_conn: Unable to allocate TX context
[00:00:03.092,193] <err> bt_conn: Unable to allocate TX context
[00:00:03.092,681] <err> bt_conn: Unable to allocate TX context
uart:~$ [00:00:13.178,161] <err> bt_conn: not connected!
[00:00:13.178,192] <err> bt_conn: not connected!
[00:00:13.178,222] <err> bt_conn: not connected!
[00:00:13.178,253] <err> bt_conn: not connected!
[00:00:13.178,283] <err> bt_conn: not connected!
[00:00:13.178,314] <err> bt_conn: not connected!
[00:00:13.178,344] <err> bt_conn: not connected!
[00:00:13.178,375] <err> bt_conn: not connected!
[00:00:13.178,405] <err> bt_conn: not connected!
[00:00:13.178,405] <err> bt_conn: not connected!
[00:00:13.178,161] <err> bt_conn: not connected!
[00:00:13.178,192] <err> bt_conn: not connected!
[00:00:13.178,222] <err> bt_conn: not connected!
[00:00:13.178,253] <err> bt_conn: not connected!
[00:00:13.178,283] <err> bt_conn: not connected!
[00:00:13.178,314] <err> bt_conn: not connected!
[00:00:13.178,344] <err> bt_conn: not connected!
[00:00:13.178,375] <err> bt_conn: not connected!
[00:00:13.178,405] <err> bt_conn: not connected!
[00:00:13.178,405] <err> bt_conn: not connected!
```

**Expected behaviour**
Bluetooth to still be functional

**Impact**
Showstopper as it presents a DoS risk

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: Zephyr SDK v0.14.0
 - Commit SHA or Version used: fc88ddf59cc2814ee88ed39a60e4673d3a13a2a6

lairdjm, 2022-04-20:
@Thalley FYI, not looked into this issue as of yet, not sure if it's a bug in mcumgr or bluetooth but seems similar to a very old issue I saw in the zephyr bluetooth controller whereby it thought there was an active connection forever when the target device had actually disconnected

Thalley, 2022-04-21:
@lairdjm Can you please provide instructions on how to reproduce?

lairdjm, 2022-04-21:
Build `cmake -GNinja -DBOARD=nrf52840dk_nrf52840 -DOVERLAY_CONFIG="overlay-shell.conf;overlay-shell-mgmt.conf" -DCONFIG_MCUMGR_CMD_IMG_MGMT=n -DCONFIG_IMG_MANAGER=n -DCONFIG_MCUBOOT_BOOTUTIL_LIB=n -DCONFIG_BOOTLOADER_MCUBOOT=n -DCONFIG_USE_DT_CODE_PARTITION=n -DCONFIG_ROM_START_OFFSET=0x0 -DCONFIG_FLASH_LOAD_OFFSET=0x0 -DCONFIG_MCUMGR_CMD_FS_MGMT=n -DCONFIG_MCUMGR_SMP_BT=y -DCONFIG_MCUMGR_SMP_BT_AUTHEN=n -DCONFIG_MCUMGR_SMP_BT_CONN_PARAM_CONTROL=y -DCONFIG_MCUMGR_SMP_SHELL=y -DCONFIG_USE_SEGGER_RTT=n -DCONFIG_LOG_BACKEND_UART=y ..` then flash and use mcumgr: `mcumgr --conntype ble --connstring ctlr_name=hci0,peer_name='Zephyr' shell exec help`.
Output:
```
uart:~$ [00:00:02.800,231] <inf> smp_bt_sample: Connected
[00:00:03.016,021] <err> bt_conn: Unable to allocate TX context
[00:00:03.016,082] <err> bt_conn: Unable to allocate TX context
[00:00:03.016,174] <err> bt_conn: Unable to allocate TX context
[00:00:06.710,754] <err> bt_conn: not connected!
[00:00:06.710,815] <err> bt_conn: not connected!
[00:00:06.710,845] <err> bt_conn: not connected!
```
Disable bluetooth on PC and wait 20 seconds, re-enable and send command again. Connection timeout, can also not find device when scanning with phone.

davidschillerJKU, 2022-05-02:
Has anyone figured out a workaround (manually changing buffer sizes, ...)?
Or is there at least a way to recover the BT stack?

Thalley, 2022-07-04:
> Has anyone figured out a workaround (manually changing buffer sizes, ...)? Or is there at least a way to recover the BT stack?
> 
> UPDATE: Setting CONFIG_BT_L2CAP_TX_BUF_COUNT=128, as suggested [here](https://github.com/zephyrproject-rtos/zephyr/issues/25917#issuecomment-925370422), solved the problem for me.

If that solves it, then isn't it just a matter of updating the `prj.conf` file?

davidschillerJKU, 2022-07-04:
> If that solves it, then isn't it just a matter of updating the `prj.conf` file?
Yes, that's what I ended up doing.
It is however not obvious that it's necessary to increase the TX buffer count in order to get certain (verbose) commands to work.

lairdjm, 2022-07-04:
> If that solves it, then isn't it just a matter of updating the `prj.conf` file?

It does fix it, but in my opinion, if a buffer size is too small it should show an error but not cause the bluetooth stack to essentially break and leave it in an undefined state.

Thalley, 2022-07-04:
> It does fix it, but in my opinion, if a buffer size is too small it should show an error but not cause the bluetooth stack to essentially break and leave it in an undefined state.

In this case it's not a buffer size, but a buffer count. The buffer count will heavily depend on what else the device is doing as well, and thus not something we can easily enforce in the stack (of course, the default value should definitely work with the minimal number of buffers needed). 

But I agree that it should not put the stack into an undefined state, and if we can prevent that, we certainly should.

nordicjm, 2022-09-06:
Gave this a quick test and it seems to be resolved now for me with https://github.com/zephyrproject-rtos/zephyr/pull/49691 - closing issue.

closed 2022-09-06 by nordicjm