zephyrproject-rtos/zephyr #45012: sam_e70b_xplained: failed to run test case tests/drivers/can/timing/drivers.can.timing

shaoanxli, 2022-04-21:
**Describe the bug**
The test case failed after commit id: 5961ad2af8f3<Assertion failed at WEST_TOPDIR/zephyr/tests/drivers/can/timing/src/main.c:194: test_timing_values: (sp_err >= 0 is false)>.The root cause is not clear yet.

**To Reproduce**
Steps to reproduce the behavior:
1. source zephyr-env.sh
2. west build -p always -b sam_e70b_xplained tests/drivers/can/timing
3. Open another terminal and execute: picocom -b 115200 /dev/ttyACM0
4. west flash
5. See error
 or
2. twister --device-testing --hardware-map $HOME/sam_e70b_xplained.file -T tests/drivers/can/timing -vv
3. See error

**Expected behavior**
Test case should pass as expected.

**Impact**
The test case drivers.can.timing failed.

**Logs and console output**
```
DEBUG   - DEVICE: testing on device CAN_0
DEBUG   - DEVICE: Running test suite can_timing_tests
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_set_timing_min
DEBUG   - DEVICE: PASS - test_set_timing_min in 0.1 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_set_timing_max
DEBUG   - DEVICE: PASS - test_set_timing_max in 0.1 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_timing
DEBUG   - DEVICE: testing bitrate 20000, sample point 87.5% (valid): prop_seg = 0, phase_seg1 = 130, phase_seg2 = 19, prescaler = 50 OK, sample point error 0.2%
DEBUG   - DEVICE: testing bitrate 50000, sample point 87.5% (valid): prop_seg = 0, phase_seg1 = 174, phase_seg2 = 25, prescaler = 15 OK, sample point error 0.0%
DEBUG   - DEVICE: testing bitrate 125000, sample point 87.5% (valid): prop_seg = 0, phase_seg1 = 209, phase_seg2 = 30, prescaler = 5 OK, sample point error 0.0%
DEBUG   - DEVICE: testing bitrate 250000, sample point 87.5% (valid): prop_seg = 0, phase_seg1 = 174, phase_seg2 = 25, prescaler = 3 OK, sample point error 0.0%
DEBUG   - DEVICE: testing bitrate 500000, sample point 87.5% (valid): prop_seg = 0, phase_seg1 = 130, phase_seg2 = 19, prescaler = 2 OK, sample point error 0.2%
DEBUG   - DEVICE: testing bitrate 800000, sample point 80.0% (valid):
DEBUG   - DEVICE: Assertion failed at WEST_TOPDIR/zephyr/tests/drivers/can/timing/src/main.c:194: test_timing_values: (sp_err >= 0 is false)
DEBUG   - DEVICE: unknown error -22
DEBUG   - DEVICE: FAIL - test_timing in 0.80 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_timing_data
DEBUG   - DEVICE: testing bitrate 500000, sample point 87.5% (valid): prop_seg = 0, phase_seg1 = 25, phase_seg2 = 4, prescaler = 10 OK, sample point error 0.9%
DEBUG   - DEVICE: testing bitrate 1000000, sample point 87.5% (valid): prop_seg = 0, phase_seg1 = 25, phase_seg2 = 4, prescaler = 5 OK, sample point error 0.9%
DEBUG   - DEVICE: testing bitrate 500000, sample point 90.0% (valid): prop_seg = 0, phase_seg1 = 26, phase_seg2 = 3, prescaler = 10 OK, sample point error 0.0%
DEBUG   - DEVICE: testing bitrate 500000, sample point 80.0% (valid): prop_seg = 0, phase_seg1 = 23, phase_seg2 = 6, prescaler = 10 OK, sample point error 0.0%
DEBUG   - DEVICE: testing bitrate 500000, sample point 100.0% (invalid): OK
DEBUG   - DEVICE: testing bitrate 8000001, sample point 75.0% (invalid): OK
DEBUG   - DEVICE: PASS - test_timing_data in 0.60 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: Test suite can_timing_tests failed.
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: RunID: 3388c175cce6eae39d7aabc0b45b2cfc
DEBUG   - DEVICE: PROJECT EXECUTION FAILED
DEBUG   - run status: sam_e70_xplained/tests/drivers/can/timing/drivers.can.timing failed
INFO    - 1/1 sam_e70_xplained          tests/drivers/can/timing/drivers.can.timing        FAILED Failed (device 4.982s)
```

**Environment :**
OS: Linux,
Toolchain Zephyr SDK 0.13.2
Commit SHA or Version: 5961ad2af8f3

henrikbrixandersen, 2022-04-22:
Thank you for reporting this. Please rerun the test with an updated main branch (needs https://github.com/zephyrproject-rtos/zephyr/commit/7a5116acb25a6ed2d729a0d9a051532c0f8d857e) and update the console output. I need to see the CAN core clock frequency.

henrikbrixandersen, 2022-04-22:
I solved a similar issue on another platform recently: https://github.com/zephyrproject-rtos/zephyr/issues/44985

shaoanxli, 2022-04-24:
> Thank you for reporting this. Please rerun the test with an updated main branch (needs [7a5116a](https://github.com/zephyrproject-rtos/zephyr/commit/7a5116acb25a6ed2d729a0d9a051532c0f8d857e)) and update the console output. I need to see the CAN core clock frequency.

I ran the can.timing on main branch(https://github.com/zephyrproject-rtos/zephyr/commit/7a5116acb25a6ed2d729a0d9a051532c0f8d857e) ,and the console output was as follow :

```
*** Booting Zephyr OS build zephyr-v3.0.0-2888-g7a5116acb25a  ***
testing on device CAN_0 @ 150000000 Hz
Running test suite can_timing_tests
===================================================================
START - test_set_timing_min
PASS - test_set_timing_min in 0.1 seconds
===================================================================
START - test_set_timing_max
PASS - test_set_timing_max in 0.1 seconds
===================================================================
START - test_timing
testing bitrate 20000, sample point 87.5% (valid): prop_seg = 0, phase_seg1 = 130, phase_seg2 = 19, prescaler = 50 OK, sample point error 0.2%
testing bitrate 50000, sample point 87.5% (valid): prop_seg = 0, phase_seg1 = 174, phase_seg2 = 25, prescaler = 15 OK, sample point error 0.0%
testing bitrate 125000, sample point 87.5% (valid): prop_seg = 0, phase_seg1 = 209, phase_seg2 = 30, prescaler = 5 OK, sample point error 0.0%
testing bitrate 250000, sample point 87.5% (valid): prop_seg = 0, phase_seg1 = 174, phase_seg2 = 25, prescaler = 3 OK, sample point error 0.0%
testing bitrate 500000, sample point 87.5% (valid): prop_seg = 0, phase_seg1 = 130, phase_seg2 = 19, prescaler = 2 OK, sample point error 0.2%
testing bitrate 800000, sample point 80.0% (valid):
Assertion failed at WEST_TOPDIR/zephyr/tests/drivers/can/timing/src/main.c:194: test_timing_values: (sp_err >= 0 is false)
unknown error -22
FAIL - test_timing in 0.80 seconds
===================================================================
START - test_timing_data
testing bitrate 500000, sample point 87.5% (valid): prop_seg = 0, phase_seg1 = 25, phase_seg2 = 4, prescaler = 10 OK, sample point error 0.9%
testing bitrate 1000000, sample point 87.5% (valid): prop_seg = 0, phase_seg1 = 25, phase_seg2 = 4, prescaler = 5 OK, sample point error 0.9%
testing bitrate 500000, sample point 90.0% (valid): prop_seg = 0, phase_seg1 = 26, phase_seg2 = 3, prescaler = 10 OK, sample point error 0.0%
testing bitrate 500000, sample point 80.0% (valid): prop_seg = 0, phase_seg1 = 23, phase_seg2 = 6, prescaler = 10 OK, sample point error 0.0%
testing bitrate 500000, sample point 100.0% (invalid): OK
testing bitrate 8000001, sample point 75.0% (invalid): OK
PASS - test_timing_data in 0.60 seconds
===================================================================
Test suite can_timing_tests failed.
===================================================================
RunID: 46030aa4049112676a30e8b6e761a9c1
PROJECT EXECUTION FAILED
```

shaoanxli, 2022-04-25:
I have a question about can timing.I see that the datasheet of sam_e70b_xplained recommends that the can controller use pck5. Why the driver here only has three clock source options?

henrikbrixandersen, 2022-04-28:
From the SAM E70 data sheet, page 1362:

[image]

henrikbrixandersen, 2022-04-28:
@shaoanxli I have created a partial, draft pull request for changing the CAN core clock source of the SAM CAN driver, but I do not have the hardware to test and finish it. See https://github.com/zephyrproject-rtos/zephyr/pull/45223

shaoanxli, 2022-04-29:
I modified the code the same as https://github.com/zephyrproject-rtos/zephyr/pull/45223 ，but failed. The console output showed :
```
DEBUG   - DEVICE: E: Failed to leave init mode
DEBUG   - DEVICE: *** Booting Zephyr OS build zephyr-v3.0.0-3206-gf7171e20b33e  ***
DEBUG   - DEVICE:
DEBUG   - DEVICE: Assertion failed at WEST_TOPDIR/zephyr/tests/drivers/can/timing/src/main.c:291: test_main: (device_is_ready(dev) is false)
DEBUG   - DEVICE: CAN device not ready
DEBUG   - Timed out while monitoring serial output on sam_e70_xplained
DEBUG   - run status: sam_e70_xplained/tests/drivers/can/timing/drivers.can.timing timeout
INFO    - 1/1 sam_e70_xplained          tests/drivers/can/timing/drivers.can.timing        FAILED Timeout (device 64.775s)
ERROR   - see: /home/shaoanli/zephyrproject/zephyr/twister-out/sam_e70_xplained/tests/drivers/can/timing/drivers.can.timing/handler.log
```

henrikbrixandersen, 2022-04-29:
> I modified the code the same as #45223 ，but failed. The console output showed :

My guess is the UPLL clock is not enabled. Without having access to the hardware, it is difficult for me to debug this further.

henrikbrixandersen, 2022-04-29:
> I enable the upll by PMC->CKGR_UCKR |= CKGR_UCKR_UPLLEN . But still fails. Once I deleted the test { 800000, 800, false } , the whole test passed. So what is the purpose of adding the test { 800000, 800, false } ,is that necessary？

The purpose of that test is exactly what you have come across here - a clock configuration issue, which does not allow the CAN controller to operate in the most commonly used CAN bitrates. This test was not there when the SAM CAN driver was added, which is why the bug wasn't discovered during development.

shaoanxli, 2022-05-03:
I check the code and it seems that the test failed because -1 returns here zephyr/drivers/can/can_common.c:60 when the bitrate is 800000.

closed 2022-05-09 by carlescufi

CEmir10, 2022-07-05:
Hi guys, I have one application which uses can_mcan driver, i coud transfer data when i used 3.1.0rc1 version where the funcition use PLLA clock. Now when i switched to the newest version of zephyr where driver uses UPPL clock i can not transfer data. I am using custom board which has atsame70j19 microcontroller.

nandojve, 2022-07-05:
UPPL is 480MHz fixed clock. PCK-5 clock was selected using 6 as default divider at devicetree, which results in 80MHz clock to be used for the can peripheral. What you must have is a crystal at XIN/XOUT for UPPL to work. The changes were made at #45223. My initial assumption is that issue is not related with clock if CAN tests passes in your HW.

CEmir10, 2022-07-05:
so without crystal it can not work?

CEmir10, 2022-07-05:
I am getting erorr message can_mcan failed to leave init mode

nandojve, 2022-07-05:
[image]
[image]

CEmir10, 2022-08-16:
> I modified the code the same as #45223 ，but failed. The console output showed :
> 
> ```
> DEBUG   - DEVICE: E: Failed to leave init mode
> DEBUG   - DEVICE: *** Booting Zephyr OS build zephyr-v3.0.0-3206-gf7171e20b33e  ***
> DEBUG   - DEVICE:
> DEBUG   - DEVICE: Assertion failed at WEST_TOPDIR/zephyr/tests/drivers/can/timing/src/main.c:291: test_main: (device_is_ready(dev) is false)
> DEBUG   - DEVICE: CAN device not ready
> DEBUG   - Timed out while monitoring serial output on sam_e70_xplained
> DEBUG   - run status: sam_e70_xplained/tests/drivers/can/timing/drivers.can.timing timeout
> INFO    - 1/1 sam_e70_xplained          tests/drivers/can/timing/drivers.can.timing        FAILED Timeout (device 64.775s)
> ERROR   - see: /home/shaoanli/zephyrproject/zephyr/twister-out/sam_e70_xplained/tests/drivers/can/timing/drivers.can.timing/handler.log
> ```
> 
> And the log "Failed to leave init mode" is on drivers/can/can_mcan.c:483 .

How did you solve this problem "Failed to leave init mode", i am getting same error, even upll is enabled.

nandojve, 2022-08-17:
Hi @shaoanxli , could you confirm that you can RX/TX with current master for us? I only have one board and don`t have a CAN setup. Just to make sure we didn`t missed anything.

shaoanxli, 2022-08-17:
> Hi @shaoanxli , could you confirm that you can RX/TX with current master for us? I only have one board and don`t have a CAN setup. Just to make sure we didn`t missed anything.

Do you mean put one device as host and another as slave then transfer data to each other?

nandojve, 2022-08-17:
Yes, if you don`t mind and have availability/resources.

shaoanxli, 2022-08-17:
> Yes, if you don`t mind and have availability/resources.

ok, I will try.

CEmir10, 2022-08-22:
@shaoanxli does it work for you?

giovanniticozzi, 2022-09-20:
Hi, I am trying to build a MCAN driver for a SamE70 Xplained Ultra Board.

In SamE70 Xplained Ultra the external oscillator is connected throught XIN and XOUT is unconnected.
In read in this case I have to enable the Bypass mode to Bypass the Main crystall oscillator.

[image]

But even in this mode I have clock issue.

Do you know if even without a crystal at XIN/XOUT is possible to have CAN working?

henrikbrixandersen, 2022-09-20:
> Do you know if even without a crystal at XIN/XOUT is possible to have CAN working?

Please contact the chip vendor for questions like that. This issue is closed.

PR #45223 (2022-05-09): drivers: can: sam: use UPLL clock instead of PLLA clock
Use the UPLLCK clock for the CAN controller as recommended by the Atmel SAM E70 data sheet.

Fixes: #45012
commit ebbb4b67 (2022-05-09): drivers: can: sam: use UPLL clock instead of PLLA clock

Use the UPLLCK clock for the CAN controller as recommended by the Atmel SAM
E70 data sheet.

Move the configuration of the clock prescaler from Kconfig to devicetree
and limit it to the values recommended by the SAM E70 datasheet.

Fixes: #45012