zephyrproject-rtos/zephyr #45031: Make NVS support sector sizes larger than 64k

jachatzi, 2022-04-21:
**Is your enhancement proposal related to a problem? Please describe.**
I'm working with an STM32H7 with 128k sectors and I'd like to use the settings module with the NVS as backend (as proposed in the documentation of the flash circular buffer). However this does not work, as NVS does not support sector sizes that large.

**Describe the solution you'd like**
I'd like to have NVS support larger sectors. @Laczen: Did you use an uint16_t for the sector size because of design limitations?

**Describe alternatives you've considered**
I'll try the flash circular buffer for the time being. I guess the note in its documentation regarding the use as a backend for the settings module is due to the nvs allowing for less rewrites of the flash content? An explanation in this note would be nice to have.

Laczen, 2022-04-21:
@jachatzi, the reason a uint16_t is used is to keep the ate size small, the ate contains the starting address of the data. Increasing the sector size will therefore break compatibility with existing nvs systems.

For the STM32H7 with 128kB sectors there are other reasons why this is not a good fit with nvs:
a. The minimum write block size is 256 byte (266 with ecc) and as each write will require at least one block for the ate and one block for the data there would be a lot of wasted space,
b. The flash erase of one block can take up to 4.2 seconds so each time a garbage collection is performed the system might be blocked for 4.2 seconds.

Using fcb will probably reduce problem a, problem b remains. The main advantage of nvs over fcb is that a interrupted write on fcb might lead to a broken fcb system (although I have no knowledge of someone who has had this problem).

So even if nvs would support larger sectors I don't think it would suit your needs.

jachatzi, 2022-04-22:
@Laczen thanks for the prompt reply!

Couldn't we configure the max section size and thus ate size, such that it is backwards compatible, but still can be configured to be used with larger targets?
The minimum write block size of the H7 is, if I'm not mistaken, 256bits, not bytes. 

Can you point me to some issue/discussion about that problem with the interrupted write on fcb which might lead to a broken fcb system? I searched through the issues but couldn't find details. Would like to read a bit further into the topic.

Laczen, 2022-04-22:
> Couldn't we configure the max section size and thus ate size, such that it is backwards compatible, but still can be configured to be used with larger targets? 

This might be possible. Maybe a completely different approach is also possible where nvs would think that it has a smaller sector size and calls the gc over a multiple of sectors (need to think about that...). 
> The minimum write block size of the H7 is, if I'm not mistaken, 256bits, not bytes.

Correct it is bits, not byte. An ate uses 8 byte, so 24 byte are lost for each ate. 

> Can you point me to some issue/discussion about that problem with the interrupted write on fcb which might lead to a broken fcb system? I searched through the issues but couldn't find details. Would like to read a bit further into the topic.

As said I have no knowledge of someone ever running into this. The problem is that fcb writes the length of an "entry" before writing the data. If there is garbage in this length how will the system behave. It will find out there is no valid data (as there is no corresponding closing part), but where will it write/read afterwards. AFAIK it has never been evaluated what happens in this case (fcb was just copied from mynewt), You could try and create a test for this on native_posix (at the time I wrote nvs native_posix or a flash simulator was not available). In the mean time mynewt migrated to fcb2 that works in a similar way as nvs.

nordicjm, 2025-01-17:
@rghaddab can this be closed as completed with zms?

jachatzi, 2025-03-25:
@nordicjm & @rghaddab Sorry for the late reply. Just looked at ZMS and also tested it briefly on an STM32H743. Works like a charm! Thus I'll close the issue.

closed 2025-03-25 by jachatzi

Laczen, 2025-03-25:
@jachatzi take the time to test this thoroughly. The sector erase time on the stm32h743 for a 128kB sector is long (order of seconds IIRC), the system will become unresponsive when it is needed to do a gc operation. This could also trigger any watchdogs that are enabled.

This is one of the reasons why little effort was done to support nvs on devices with large sectors. Another reason is that the amount of available sectors is limited and storage would just be taking up to much space compared to the size available for the image.

jachatzi, 2025-03-26:
@Laczen thanks for the hint! I'm not using the new ZMS in production, just let the H7 run a little firmware which wrote increasing numbers in the settings module with ZMS as backend and read out the flash after a long time to check that I reached the second sector and everything works fine.

The product which came out of the project in which I initially stumbled upon the problem uses FCB as backend. There of course the problem of the long erase times still exists. We used sectors in the second bank in order not to interrupt the program. But I wouldn't normally store "customer parameters" in the internal flash of a µC anyways. With that project we only did so as we had quite tight space constraints and thus couldn't fit an EEPROM on the PCB.

zhang-wenchao, 2025-09-24:
I also encountered this problem, maybe there should be a hint when compiling.