zephyrproject-rtos/zephyr #45039: Bluetooth: Controller: Broadcast multiple BIS (broadcast ISO streams)

carlst, 2022-04-21:
**Is your enhancement proposal related to a problem? Please describe.**
The core architecture supports broadcasting multiple BIS (e.g. transmitting stereo audio or multiple language audio).

**Describe the solution you'd like**
Would like to transmit (audio over) multiple BIS using one BIG.

**Additional context**
The host and controller appear to contain the logic required for multiple BIS, however a few changes are required.

The following changes were made for a test system:

1. Add samples/bluetooth/broadcast_audio_source_multiple

2. Change prj.conf to support 2 streams:

`CONFIG_BT_AUDIO_BROADCAST_SRC_STREAM_COUNT=2`
`CONFIG_BT_CTLR_ADV_ISO_STREAM_MAX=2`
`CONFIG_BT_ISO_MAX_CHAN=2`
`CONFIG_BT_ISO_TX_BUF_COUNT=4`
`CONFIG_BT_CTLR_ISO_TX_BUFFER_SIZE=250`

3. Change nse calculation for multiple streams:

In `subsys/bluetooth/controller/ll_sw/ull_adv_iso.c`, `ll_big_create()`:
	`lll_adv_iso->nse = lll_adv_iso->bn * lll_adv_iso->irc * lll_adv_iso->num_bis;`

4. Add Tx ack for multiple streams:

In `subsys/bluetooth/controller/ll_sw/nordic/lll/lll_adv_iso.c`, change
`prepare_cb_common()` and `isr_tx_common()` to enqueue Tx ack of packets for all streams

The Tx acks were sent in a loop. For `prepare_cb_common()`, the loop was iterated from the highest-stream to the lowest stream, so that a stream 0 `link` could be defined when the loop completed.

The changes were limited as follows:
- testing has so far used only nRF52840DK
- there is currently no full ISOAL integration
- there is currently no support for interleaved BIS (only sequential BIS are supported)

cvinayak, 2022-05-03:
@carlst have you already sent a PR with your suggestions?

I had something in my PR here #45069, do check if this meets your requirement...

carlst, 2022-05-04:
@cvinayak I have not sent a PR. 

#45069 appears to provide a more comprehensive solution (for Broadcast multiple BIS).

PR #45069 (2022-09-05): Bluetooth: Controller: Support for multiple BIS sync selection
Added implementation to use the BIS array parameter and
selectively synchronize to subset of multiple Broadcast
Isochronous Streams in a Broadcast Isochronous Group.

Added implementation to calculate BIG event timing
reservations.

Fixes #41954 
Fixes #45039
closed 2022-06-16 by carlst