zephyrproject-rtos/zephyr #45044: cavstool: add a new option to dump contents to the specified file

keqiaozhang, 2022-04-22:
**Is your enhancement proposal related to a problem? Please describe.**
We are switching to use the cavstool.py to dump the mailbox or DMA trace contents, we want to save the trace to a specified file for debugging propose.  It would be great if there was an option to dump the contents to a specified file.

**Describe the solution you'd like**
$ cavstool.py -l -o outfile

**Describe alternatives you've considered**
$ cavstool.py -l &> outfile

marc-hb, 2022-04-27:
@keqiaozhang why is file redirection not enough?

nashif, 2022-04-27:
please submit a PR, this is a small change to the script.

marc-hb, 2022-04-28:
> @keqiaozhang why is stdout redirection > cavstool.log not enough?

There are various tricks to combine `sudo` + redirection, here's my favorite:

```shell
sudo cavstool.py ... | sudo tee [ -a ] log.txt > /dev/null
```

keqiaozhang, 2022-04-28:
> ```shell
> | sudo tee [ -a ]
> ```

Yes, we can use stdout redirection. Now I'm using '&>' for debugging.