zephyrproject-rtos/zephyr #45089: stm32: usart: rx pin inversion missing

helo9, 2022-04-23:
**Is your enhancement proposal related to a problem? Please describe.**
SBUS is a common serial protocol in radio control that uses inverted logic (1=L, 0=H). 

**Describe the solution you'd like**
STM32 can natively support this, by setting pin `RXINV` in `USART_CR2` for many chips. Therefore I would add this option
to the devicetree bindings and the API.

**Describe alternatives you've considered**
Leave the functionality out, as i

FRASTM, 2022-04-25:
That could be an USART  _option flag_ like it is done withthe PWM_STM32_COMPLEMENTARY by the https://github.com/zephyrproject-rtos/zephyr)/include/zephyr/dt-bindings/pwm/stm32_pwm.h

erwango, 2022-04-25:
@helo9, Can you check if what you request is matching following feature which was merged recently ?https://github.com/zephyrproject-rtos/zephyr/pull/44487/

helo9, 2022-04-25:
@erwango have seen those changes, but they do not exactly help me. Have already implemented a prototype of what I need: https://github.com/helo9/zephyr/commit/bd5c9e00fb2b1d8f17b693c4168bb4fa668cc88d

@FRASTM I thought about adding this to the `uart_stm32_init` and maybe the `uart_stm32_configure` functions.

I'll open a PR with my branch, so we can discuss it there.

helo9, 2022-05-15:
was closed by #45124

closed 2022-05-15 by helo9