zephyrproject-rtos/zephyr #45094: stm32: Add USB HS support to STM32H747

romainreignier, 2022-04-23:
**Is this request related to a missing driver support for a particular hardware platform, SoC or board? Please describe.**
I would like to add support for the Arduino Portenta H7 board in Zephyr, but I am new to Zephyr.
Adding a new dts seemed doable but some more changes are needed like the support of USB High Speed for the stm32h474.

**Describe why you are asking for this support?**
I am new to Zephyr and while I could look in the reference manual to find the addresses to fill the device tree, adding USB HS support seems a task a bit too hard for me. I have no experience working with the USB stack.

USB support is needed to be able to detect a baudrate change to 1200 bauds and reset the board to flash it.
Arduino did not add a SWD or JTAG header on the Portenta, even in the "Pro" range of products, so the USB bootloader is the only way to upload a program.
For now I can double tap the reset button to force the bootloader but having USB support would improve the workflow.

erwango, 2022-04-25:
@romainreignier Adding USB HS to stm32h747 is doable but this will allow its usage in Zephyr application, it won't have any effect on bootloading.

romainreignier, 2022-04-25:
Yes, I know but to allow to automatically reboot the the bootloader to upload a new sketch, the detection of the switch to 1200 bauds is needed like on the sam0 platform.
A callback must be called when the baudrate is changed while usind cdc-acm.

romainreignier, 2022-07-04:
This issue is still relevant.

erwango, 2023-01-19:
Done.
`usbotg_hs` is enabled on stm32h747i_disco

closed 2023-01-19 by erwango