zephyrproject-rtos/zephyr #45105: ACRN: failed to run testcase tests/kernel/fifo/fifo_timeout/

Zhaoningx, 2022-04-25:
Describe the bug

Failed to run the test case tests/kernel/fifo/fifo_timeout/.

To Reproduce
twister -p acrn_ehl_crb --device-testing -T tests/kernel/fifo/fifo_timeout/  --device-serial-pty="/opt/remotehw/acrn-test-pty.exp,ehlsku11-2" --west-flash="/opt/remotehw/remotehw-x86-acrn.sh,ehlsku11-2" -vv

Logs and console output

```
DEBUG   - DEVICE: ----- Entering VM 0 Shell -----
DEBUG   - DEVICE: *** Booting Zephyr OS build zephyr-v3.0.0-3037-gd92e3b88bacc  ***
DEBUG   - DEVICE: Running test suite test_fifo_timeout
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_timeout_empty_fifo
DEBUG   - DEVICE: PASS - test_timeout_empty_fifo in 0.12 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_timeout_non_empty_fifo
DEBUG   - DEVICE: PASS - test_timeout_non_empty_fifo in 0.1 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_timeout_fifo_thread
DEBUG   - DEVICE:
DEBUG   - DEVICE: Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/fifo/fifo_timeout/src/main.c:371: test_timeout_fifo_thread: packet != NULL is false
DEBUG   - DEVICE:
DEBUG   - DEVICE: FAIL - test_timeout_fifo_thread in 0.24 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_timeout_threads_pend_on_fifo
DEBUG   - DEVICE:
DEBUG   - DEVICE: Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/fifo/fifo_timeout/src/main.c:130: test_thread_pend_and_timeout: packet == NULL is false
DEBUG   - DEVICE:
DEBUG   - DEVICE: FAIL - test_timeout_threads_pend_on_fifo in 0.3 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_timeout_threads_pend_on_dual_fifos
DEBUG   - DEVICE: thread (q order: 2, t/o: 0, fifo 0x10bea0)
DEBUG   - DEVICE: E: Page fault at address (nil) (error code 0x2)
DEBUG   - DEVICE: E: Linear address not present in page tables
DEBUG   - DEVICE: E: Access violation: supervisor thread not allowed to write
DEBUG   - DEVICE: E: PTE: not present
DEBUG   - DEVICE: E: RAX: 0x000000000010aa50 RBX: 0x000000000010a030 RCX: 0x0000000000000044 RDX: 0x0000000000000000
DEBUG   - DEVICE: E: RSI: 0x000000000010a390 RDI: 0x0000000000112390 RBP: 0x0000000000115170 RSP: 0x0000000000115150
DEBUG   - DEVICE: E:  R8: 0x0000000000000001  R9: 0x0000000000000000 R10: 0x0000000000000000 R11: 0x0000000000000000
DEBUG   - DEVICE: E: R12: 0x0000000000000046 R13: 0x000000000000000a R14: 0x0000000000000001 R15: 0x0000000000000000
DEBUG   - DEVICE: E: RSP: 0x0000000000115150 RFLAGS: 0x0000000000010016 CS: 0x0018 CR3: 0x0000000000146000
DEBUG   - DEVICE: E: RIP: 0x000000000010651d
DEBUG   - DEVICE: E: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
DEBUG   - DEVICE: E: Current thread: 0x10a000 (unknown)
DEBUG   - DEVICE: E: Halting system
```

Environment (please complete the following information):

OS: Linux
Toolchain: zephyr SDK 13.2
Commit: https://github.com/zephyrproject-rtos/zephyr/commit/d92e3b88bacc

Zhaoningx, 2022-04-25:
I did some changes to debug this issue,
```
@@ -104,6 +104,7 @@ static bool is_timeout_in_range(uint32_t start_time, uint32_t timeout)
        diff = (uint32_t)k_cyc_to_ns_floor64(stop_time -
                        start_time) / NSEC_PER_USEC;
        diff = diff / USEC_PER_MSEC;
+       printf("spent time: %u\n", diff);
        return timeout <= diff;
 }

@@ -367,7 +368,7 @@ static void test_timeout_fifo_thread(void)
                                &timeout, NULL,
                                FIFO_THREAD_PRIO, K_INHERIT_PERMS, K_NO_WAIT);

-       packet = k_fifo_get(&fifo_timeout[0], K_MSEC(timeout + 10));
+       packet = k_fifo_get(&fifo_timeout[0], K_MSEC(timeout + 1000));
        zassert_true(packet != NULL, NULL);
        zassert_true(is_timeout_in_range(start_time, timeout), NULL);
        put_scratch_packet(packet);
```
the log:
```
DEBUG   - DEVICE: START - test_timeout_fifo_thread
DEBUG   - DEVICE: spent time: 20
DEBUG   - DEVICE: PASS - test_timeout_fifo_thread in 0.23 seconds
```
I find that the spent time is more than or equal to the timeout value of 20ms, So I guess this is due to it's not having enough time to wait for the subthread to write data to the FIFO.

Zhaoningx, 2022-05-23:
Hi @enjiamai, I need to debug when the ACRN is released from the daily testing, so I need more time to debug, I'll update tomorrow. Thanks.

Zhaoningx, 2022-05-24:
Hi, I looked into the issue, I find that the action of the function k_msleep() is not expected. In the test case` test_timeout_fifo_thread`,  it tested the data can pass by thread and sub-thread in 20ms timeout range, I added some `printk` to check every step spent time value, I find that after waiting for 10ms by calling `k_msleep(10)`, the real spent time is close to 20ms, so I'll look into it.

enjiamai, 2022-06-17:
Got the root cause. This happens only when CONFIG_SYS_CLOCK_HW_CYCLES_PER_SEC > the actually HW clock. That's why it only happens on one board which has lower clock frequency. It is not a zephyr bug. So close it.

closed 2022-06-17 by enjiamai