zephyrproject-rtos/zephyr #45112: Cannot install watchdog timeout on STM32WB

r2r0, 2022-04-25:
**Describe the bug**
iwdg_stm32_install_timeout does not return because it is stuck inside `while (LL_IWDG_IsReady(iwdg) == 0)` loop.
It looks like k_uptime_get_32 returns the same value for every call.

Please also mention any information which could help others to understand
the problem you're facing:
 - STM32WB55 clocked by PLL (based on HSE), with enabled tickless kernel (LPTIMER with LSE).
 - CONFIG_WDT_DISABLE_AT_BOOT is enabled.
 - After adding 3 NOP instruction to (mentioned above) `while` loop timeout is properly detected.
 - For some reasons busy timeout polling with `k_uptime_get_32` does not work well.

**To Reproduce**
Steps to reproduce the behavior:
Build with IWDG enabled and with CONFIG_WDT_DISABLE_AT_BOOT configuration option set.
Try to run application.

**Expected behavior**
`iwdg_stm32_install_timeout` should return with `-EINVAL` error.

**Impact**
showstopper

**Logs and console output**

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain Zephyr SDK, 0.14.1
 - Commit SHA d92e3b88bacc6e6fc1b6222421ef4accdf2d574e

**Additional context**
During call to `iwdg_stm32_install_timeout` the IWDG is disabled because `LL_IWDG_Enable` was never called before and then `LL_IWDG_IsReady` will always return 0.
`LL_IWDG_Enable` was not called before because of CONFIG_WDT_DISABLE_AT_BOOT is set.
It means that present implementation may be incorrect because `LL_IWDG_Enable` will be called during `wdt_setup` which (according to documentation) will happen after call to `wdt_install_timeout`.

FRASTM, 2022-04-25:
@r2r0  could you please check if PR #44989  helps

r2r0, 2022-04-25:
Thanks @FRASTM - in fact this PR removes the trigger of described problem.
After applying #44972 CPU leaves the while loop and the timeout hangout is avoided but it is still present.

FRASTM, 2022-04-27:
When you build the samples/subsys/task_wdt with `CONFIG_WDT_DISABLE_AT_BOOT=n` 
(by default in the prj.conf  the CONFIG_WDT_DISABLE_AT_BOOT is 'y') 
the timeout is installed during the initialisation of the device, with the same sequence as in _install_timeout_

r2r0, 2022-04-29:
Unfortunately blocking of while loop is visible only very rarely when console and debugger (even without breakpoints installed) are connected. When it happens then it looks that also LPTIM ISR is not working.
After power cycling problem disappears (usually) for long time.
Because I am not able to consistently reproduce problem then I will close this issue for now.
BTW> @FRASTM  would it be possible to advance merging of the #44972, please.

closed 2022-04-29 by r2r0