zephyrproject-rtos/zephyr #45114: Sample net/sockets/echo not working with disco_l475_iot1

manuja2000, 2022-04-25:
**Describe the bug**
net/sockets/echo not working with disco_l475_iot1,[ accept() function](https://github.com/zephyrproject-rtos/zephyr/blob/main/samples/net/sockets/echo/src/socket_echo.c#L61) not returning anything and no line is executing after it, but client which is running on linux gets connected. 

**To Reproduce**
Steps to reproduce the behavior:

Access point/wifi connected with the IoT board. [Reference](https://github.com/jptcarreira/zephyr-demo-wifi)

Terminal 1
In net/sockets/echo
1.  west build -b disco_l475_iot1 -d build/server 
2.  cd build/server
3. west flash

Terminal 2
In client.c directory
1.  gcc client.c -o client
2. ./client <"ip address of server"> [client code](https://www.thegeekstuff.com/2011/12/c-socket-programming/)

**Expected behavior**
Code execution after accept() call and communication to happen between client and server.

**Impact**
Showstopper

**Logs and console output**

[image]

Environment:

    OS: Linux(Oracle VM VirtualBox)
    Toolchain :Zephyr SDK

**Additional context**

Added in prj.conf

CONFIG_NET_SHELL=y
CONFIG_SHELL=y
CONFIG_NET_L2_WIFI_SHELL=y
CONFIG_WIFI=y

Removed from prj.conf 

CONFIG_NET_CONFIG_MY_IPV4_ADDR="192.0.2.1"
CONFIG_NET_CONFIG_PEER_IPV4_ADDR="192.0.2.2"

erwango, 2022-05-02:
@manuja2000 Can you check if your point is similar as this one: https://github.com/zephyrproject-rtos/zephyr/issues/38544?

manuja2000, 2022-05-04:
Yes, seems to be similar, when using the B-L475E-IOT01A, attempts to use a TCP socket result in a
hang.

manuja2000, 2022-05-10:
Hi @erwango, 
Is there anything that can help/resolve this issue?

erwango, 2022-05-10:
@manuja2000 I'm having a look. Though, I'm not able to reproduce your set up completely as I'm not able to establish connection between client and server.

> In net/sockets/echo

Isn't it `echo_server` that you're using?

Also, did you enable specific option to automatically connect to SSID? Did you add code from https://github.com/jptcarreira/zephyr-demo-wifi/blob/master/zephyr-demo-wifi/ to zephyr echo sample?

erwango, 2022-05-17:
@manuja2000 Not yet, but on my todo list

erwango, 2022-05-18:
@manuja2000 I've had a try to your sample, but I can't connect to WIFI network, while this is working ok using `sample/net/wifi`
:
```
Sucessfull connected to SSID:[wan_one]
Assigned IP address [192.168.187.170] 
error: listen: 9
exit
```
Any idea what could be going on vs this sample ?

erwango, 2022-05-18:
So I'm reproducing your issue using `CONFIG_NET_SOCKETS_OFFLOAD=n`

That said, I think the correct configuration is to use ``CONFIG_NET_SOCKETS_OFFLOAD=y`, which ends up with the following result:
```
Sucessfull connected to SSID:[NETGEAR_MBED]
Assigned IP address [10.0.0.11] 
error: listen: 9
exit
```
But then I wonder if this is really an error and if this configuration should actually working with `samples/net/sockets/echo`.
Maybe @rlubos you would be able to help us to understand what should be the correct configuration to get this sample working ?

rlubos, 2022-05-18:
What is the network driver that you're using? Whether to enable `CONFIG_NET_SOCKETS_OFFLOAD` or not really depends ont he net driver in use (there are only a few that make use of this option).

erwango, 2022-05-19:
> What is the network driver that you're using? Whether to enable `CONFIG_NET_SOCKETS_OFFLOAD` or not really depends ont he net driver in use (there are only a few that make use of this option).

This is `drivers/wifi/eswifi`. Should this sample be compatible with both `CONFIG_NET_SOCKETS_OFFLOAD=y` and `CONFIG_NET_SOCKETS_OFFLOAD=n` ?

rlubos, 2022-05-19:
I can already see `samples/net/sockets/echo` will not work with this driver if socket offloading is enabled, as the driver does not implment `listen()` call at all... Surprisingly, the sample does implement `accept()` function, so I'm not sure what did the driver author had in mind implementing it w/o `listen()`.

Now, the driver seems to implement both, socket offloading and net offloading, so in theory one disable socket offloading and try again with net offload. Given that the default for the driver is socket offloading, I'm not sure in what state the other configuration would be. As there seem to be a problem in that configuration as well, I'm assuming not best.

rlubos, 2022-05-19:
> This is `drivers/wifi/eswifi`. Should this sample be compatible with both `CONFIG_NET_SOCKETS_OFFLOAD=y` and `CONFIG_NET_SOCKETS_OFFLOAD=n` ?

And as for the sample, it just uses some basic socket API calls, so as long as the offloaded driver implements the API properly, w/o taking shortcuts like `eswifi`, it should make no difference whether net device is offloaded or not.

erwango, 2022-05-19:
Thanks @rlubos for the analysis.

> I'm not sure what did the driver author had in mind implementing it w/o `listen()`.

This was implemented in `https://github.com/zephyrproject-rtos/zephyr/pull/18641`. Maybe @[ssekar15](https://github.com/ssekar15), as the author, you would be able to comment on that  ?

Anyway, this doesn't seem to be a bug, but rather a limitation which would requirement an enhancement to the existing codebase. So I'm changing this issue status.
Also assigning the point to @loicpoulain as the assigned collaborator for this component (and I can see you're still around ;-) )

loicpoulain, 2022-05-22:
Please try the PR/fix: #45867

manuja2000, 2022-05-23:
Hi @loicpoulain,

I've tried this PR/fix, but still facing the same issue.

loicpoulain, 2022-05-23:
@manuja2000 I test with the same config you shared and connect manually to my AP, then connect from remote computer via telnet.

`*** Booting Zephyr OS build v3.1.0-rc2  ***
Single-threaded TCP echo server waits for a connection on port 4242...

uart:~$ wifi connect CISCO Windows@@$
Connection requested
Connection request failed (-5)
uart:~$ wifi connect CISCO passphrase
Connection requested
uart:~$
Connected
Connection #0 from 192.168.1.28
Connection from 192.168.1.28 closed
Connection #1 from 192.168.1.28`

manuja2000, 2022-05-23:
Thanks @loicpoulain.
This is working now.

PR #45867 (2022-06-05): Fix eswifi socket offload tcp server
Fix #45114
closed 2022-06-05 by carlescufi