zephyrproject-rtos/zephyr #45139: Extent sensor API by new sensor_read function

LeeNuss, 2022-04-26:
## Problem description

The current sensor API does not allow for the distinction between a sensor that is explicitly sampled by Zephyr (e.g. a simple temperature sensor that allows Zephyr to simple read back the current temperature) and a sensor that manages sampling itself (e.g. an IMU that has an internal sample clock and writes samples to an internal FIFO). In the latter case, it is not possible to retrieve the values from the FIFO using the existing API without losing information. Specifically the order of samples, which may be of differing types, is lost and the user cannot necessarily know how many samples are in the FIFO at any point in time.

This proposal is a consequence of the discussions in [PR43429](https://github.com/zephyrproject-rtos/zephyr/pull/43429).

## Solution implementation

This proposal adds one new function and one new struct to the sensor API. This function is optional (although can probably be trivially implemented for all existing drivers if desired). This function follows the same semantics as the C standard library read() function and allows for more complex sensor behaviours to be exposed without adding significant complexity to the API.

### New `sensor_read` API function
- semantics based on standard C function `read()`
- Prototype: `int sensor_read(struct sensor_reading *buf, int size)`
- Parameters: Pointer to buffer in user space, number of elements in the buffer
- Returns non-negative integer of number of read data points (can be 0), negative indicates an error
### User buffer:
- Struct array/any buffer datatype:
  ```
  struct sensor_reading {
      enum sensor_channel channel;
      struct sensor_value value;
  };
  ```
### How to read values from user space:
- Init sensor buffer: `struct sensor_reading sensor_buffer[SENSOR_BUFF_SIZE];`
- Loop overs sensor_read API until all elements processed (return value = 0):
  - Call sensor_read API: `int read_elements = sensor_read(sensor_buffer, SENSOR_BUFF_SIZE);`
  - Process elements in buffer

An example implementation can be seen in our [downstream project PR](https://github.com/gaitQ/sdk-zephyr/pull/7)

## Alternatives
- The buffer can be managed within the driver, however this won't allow much flexibility.
- Using the current API:
   - `sensor_sample_fetch_chan(dev, SENSOR_CHAN_ALL)` reads all data into a FIFO buffer
   - The call to `sensor_sample_fetch_chan` returns the size of the buffer - this would misuse the return value, which is usually 0 or a negative error code.
   - The app can then loop over all elements calling `sensor_channel_get(dev, SENSOR_CHAN_ALL, &sensor_value)` to retrieve the value together with the sensor channel type of the current entry, which would misuse the fractional part of the sensor_value struct as channel type
   - This approach requires many unintended API usage and therefore not really reasonable

yperess, 2023-10-23:
I think this is resolved by the new async API + https://github.com/zephyrproject-rtos/zephyr/pull/60063

yperess, 2024-01-04:
I'm going to close this issue since I believe the new API meets all the requirements. If I'm wrong please feel free to reopen and add details of what you weren't able to do.

closed 2024-01-04 by yperess