zephyrproject-rtos/zephyr #45182: MCUBoot Usage Fault on RT1060 EVK

danieldegrasse, 2022-04-27:
**Describe the bug**
When flashing the MCUBoot bootloader (`mcuboot/boot/zephyr`), the bootloader encounters a usage fault and does not load a bootable image. Reproduced on zephyr-v3.0.0-3118-g82634eda93, with a RT1060 EVK

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=mimxrt1060_evk mcuboot/boot/zephyr
3. make
4. make flash
5. see usage fault

**Expected behavior**
The bootloader should load an image if a valid one is found, or print an error about not finding a valid image.

**Impact**
MCUboot support for the RT1060 EVK is not functional.

**Logs and console output**
```
I: Starting bootloader
E: ***** USAGE FAULT *****
E:   Illegal use of the EPSR
E: r0/a1:  0x6000a9e8  r1/a2:  0x8000a804  r2/a3:  0x8000a7f4
E: r3/a4:  0x00000000 r12/ip:  0x00000000 r14/lr:  0x60008ae1
E:  xpsr:  0x60000000
E: Faulting instruction address (r15/pc): 0x00000000
E: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
E: Current thread: 0x800000f0 (unknown)
E: Halting system
```

**Environment (please complete the following information):**
 - OS: Ubuntu 20.04 (WSLv1)
 - Toolchain: Zephyr SDK
 - Commit SHA or Version used: zephyr-v3.0.0-3118-g82634eda93

kunoh, 2022-04-28:
Hi, I'm getting a similar error on RT1050 EVKB QSPI

I: Starting bootloader
E: ***** USAGE FAULT *****
E:   Illegal use of the EPSR
E: r0/a1:  0x6000aa44  r1/a2:  0x800087e4  r2/a3:  0x800087d4
E: r3/a4:  0x00000040 r12/ip:  0x00000000 r14/lr:  0x60008b73
E:  xpsr:  0x60000000
E: Faulting instruction address (r15/pc): 0x00000040
E: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
E: Current thread: 0x800000f8 (unknown)
E: Halting system

kunoh, 2022-04-29:
Thank you @danieldegrasse, it works perfectly!

dleach02, 2022-05-03:
@kunoh, can you review #45183

PR #45183 (2022-05-03): Fix MCUBoot Usage Fault
MCUBoot was encountering a usage fault due to read-while-write hazards in the mcux flash driver. Move all device data for the MCUX flash drivers out of flash and into RAM to prevent this fault.

Fixes #45182
commit 7aa883b5 (2022-05-03): drivers: flash_mcux_flexspi_nor: Move all device data to RAM

Move all device data to RAM. Since the flexspi driver accesses the
flash device that is being used for XIP, various RWW hazards can occur
if the flexspi driver is interacting with the flash device, while
running in XIP mode.

Fixes #45182
closed 2022-05-03 by dleach02