zephyrproject-rtos/zephyr #45186: Building Zephyr on Ubuntu fails when ZEPHYR_TOOLCHAIN_VARIANT is set to llvm

AmPaschal, 2022-04-28:
**Describe the bug**
When trying to build zephyr on Ubuntu 20.04.3 LTS, with ZEPHYR_TOOLCHAIN_VARIANT set to llvm, build fails with error message

CMake Error at /home/xxx/zephyrproject/zephyr/cmake/modules/dts.cmake:174 (message):
  command failed with return code: No such file or directory
Call Stack (most recent call first):
  /home/xxx/zephyrproject/zephyr/cmake/modules/zephyr_default.cmake:121 (include)
  /home/xxx/zephyrproject/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:51 (include)
  /home/xxx/zephyrproject/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:76 (include_boilerplate)
  CMakeLists.txt:4 (find_package)

But Zephyr builds successfully with ZEPHYR_TOOLCHAIN_VARIANT set to zephyr

Please also mention any information which could help others to understand
the problem you're facing:
 - What target platform are you using? Ubuntu 20.04.3 LTS
 - What have you tried to diagnose or workaround this issue? No
 - ...

**To Reproduce**
Steps to reproduce the behavior:
1. Follow all the steps on the Zephyr Getting started page for Ubuntu https://docs.zephyrproject.org/latest/develop/getting_started/index.html
2. Run command to build 'west build -p auto -b native_posix samples/basic/blinky'

**Expected behavior**
Zephyr builds successfully and an elf executable is produced

**Impact**
What impact does this issue have on your progress An error is thrown and zephyr doesn't get built, impeding my progress

**Logs and console output**
If applicable, add console logs or other types of debug information
e.g Wireshark capture or Logic analyzer capture (upload in zip archive).
copy-and-paste text and put a code fence (\`\`\`) before and after, to help
explain the issue. (if unable to obtain text log, add a screenshot)

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain Zephyr SDK
 - Commit SHA or Version used: c33ce95277cc0f51b61084d26b6bb3ae9f5a6b9c

**Additional context**
ZEPHYR_TOOLCHAIN_VARIANT=llvm

stephanosio, 2022-04-28:
Can you post a full build log so we can better understand the problem?

AmPaschal, 2022-04-28:
(.venv) pamusuo@berlin:~/zephyrproject/zephyr$ west build -p auto -b native_posix samples/basic/blinky
-- west build: generating a build system
Loading Zephyr default modules (Zephyr base).
-- Application: /home/pamusuo/zephyrproject/zephyr/samples/basic/blinky
-- Found Python3: /home/pamusuo/zephyrproject/.venv/bin/python3 (found suitable exact version "3.8.10") found components: Interpreter 
-- Cache files will be written to: /home/pamusuo/.cache/zephyr
-- Zephyr version: 3.0.99 (/home/pamusuo/zephyrproject/zephyr)
-- Found west (found suitable version "0.13.1", minimum required is "0.7.1")
-- Board: native_posix
-- Found dtc: /usr/bin/dtc (found suitable version "1.5.0", minimum required is "1.4.6")
-- Found toolchain: host (clang/ld)
-- Found BOARD.dts: /home/pamusuo/zephyrproject/zephyr/boards/posix/native_posix/native_posix.dts
CMake Error at /home/pamusuo/zephyrproject/zephyr/cmake/modules/dts.cmake:174 (message):
  command failed with return code: No such file or directory
Call Stack (most recent call first):
  /home/pamusuo/zephyrproject/zephyr/cmake/modules/zephyr_default.cmake:121 (include)
  /home/pamusuo/zephyrproject/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:51 (include)
  /home/pamusuo/zephyrproject/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:76 (include_boilerplate)
  CMakeLists.txt:4 (find_package)

-- Configuring incomplete, errors occurred!
FATAL ERROR: command exited with status 1: /usr/bin/cmake -DWEST_PYTHON=/home/pamusuo/zephyrproject/.venv/bin/python3 -B/home/pamusuo/zephyrproject/zephyr/build -S/home/pamusuo/zephyrproject/zephyr/samples/basic/blinky -GNinja -DBOARD=native_posix

AmPaschal, 2022-04-28:
@stephanosio Would this be helpful?
Please, let me know if there's any other data that's needed

AmPaschal, 2022-04-29:
I discovered I didn't have llvm and clang installed in the ubuntu machine I was using 😩😢
I installed it and have built the executable.

I'll be closing this issue

closed 2022-04-29 by AmPaschal