zephyrproject-rtos/zephyr #45189: sam_e70b_xplained: failed to run test case tests/benchmarks/cmsis_dsp/basicmath

shaoanxli, 2022-04-28:
**Describe the bug**
The test case failed after commit id: 900b6723e76897841888a18d20d710d7dfdbee7c <WEST_TOPDIR/zephyr/tests/benchmarks/cmsis_dsp/basicmath/... FAILED Timeout>.The same bug on frdm_k64f platform. The root cause is not clear yet.

**To Reproduce**
Steps to reproduce the behavior:
1. source zephyr-env.sh
2. west build -p always -b sam_e70b_xplained  tests/benchmarks/cmsis_dsp/basicmath
3. Open another terminal and execute: picocom -b 115200 /dev/ttyACM0
4. west flash
5. See error
  or
 2.twister --device-testing --hardware-map $HOME/sam_e70b_xplained.file -T tests/benchmarks/cmsis_dsp/basicmath -vv
 3.See error

**Expected behavior**
Test case should pass as expected.

**Impact**
The test case tests/benchmarks/cmsis_dsp/basicmath failed.
Take tests/benchmarks/cmsis_dsp/basicmath/benchmark.cmsis_dsp.basicmath.fpu as an example.

**Logs and console output**
```
DEBUG   - DEVICE: *** Booting Zephyr OS build zephyr-v3.0.0-3078-g900b6723e768  ***
DEBUG   - DEVICE: Running test suite basicmath_f32_benchmark
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_benchmark_vec_add_f32
DEBUG   - DEVICE: Processor Cycles = 3930
DEBUG   - DEVICE: PASS - test_benchmark_vec_add_f32 in 0.3 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_benchmark_vec_sub_f32
DEBUG   - DEVICE: Processor Cycles = 1928
DEBUG   - DEVICE: PASS - test_benchmark_vec_sub_f32 in 0.3 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_benchmark_vec_mult_f32
DEBUG   - DEVICE: Processor Cycles = 1970
DEBUG   - DEVICE: PASS - test_benchmark_vec_mult_f32 in 0.3 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_benchmark_vec_abs_f32
DEBUG   - DEVICE: Processor Cycles = 2127
DEBUG   - DEVICE: PASS - test_benchmark_vec_abs_f32 in 0.3 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_benchmark_vec_negate_f32
DEBUG   - DEVICE: Processor Cycles = 1425
DEBUG   - DEVICE: PASS - test_benchmark_vec_negate_f32 in 0.3 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_benchmark_vec_offset_f32
DEBUG   - DEVICE: Processor Cycles = 1405
DEBUG   - DEVICE: PASS - test_benchmark_vec_offset_f32 in 0.3 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_benchmark_vec_scale_f32
DEBUG   - DEVICE: Processor Cycles = 1433
DEBUG   - DEVICE: PASS - test_benchmark_vec_scale_f32 in 0.3 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_benchmark_vec_dot_prod_f32
DEBUG   - DEVICE: Processor Cycles = 2146
DEBUG   - DEVICE: PASS - test_benchmark_vec_dot_prod_f32 in 0.3 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: Test suite basicmath_f32_benchmark succeeded
DEBUG   - DEVICE: E: ***** MPU FAULT *****
DEBUG   - DEVICE: E:   Data Access Violation
DEBUG   - DEVICE: E:   MMFAR Address: 0x409f2c
DEBUG   - DEVICE: E: r0/a1:  0x00000000  r1/a2:  0x0000000a  r2/a3:  0x80000000
DEBUG   - DEVICE: E: r3/a4:  0x00000001 r12/ip:  0x00000000 r14/lr:  0x0040116b
DEBUG   - DEVICE: E:  xpsr:  0x01000000
DEBUG   - DEVICE: E: s[ 0]:  0x00000000  s[ 1]:  0x00000000  s[ 2]:  0x00000000  s[ 3]:  0x00000000
DEBUG   - DEVICE: E: s[ 4]:  0x00000000  s[ 5]:  0x00000000  s[ 6]:  0x00000000  s[ 7]:  0x00000000
DEBUG   - DEVICE: E: s[ 8]:  0x00000000  s[ 9]:  0x00000000  s[10]:  0x00000000  s[11]:  0x00000000
DEBUG   - DEVICE: E: s[12]:  0x00000000  s[13]:  0x00000000  s[14]:  0x00000000  s[15]:  0x00000000
DEBUG   - DEVICE: E: fpscr:  0x00000000
DEBUG   - DEVICE: E: Faulting instruction address (r15/pc): 0x004037ce
DEBUG   - DEVICE: E: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
DEBUG   - DEVICE: E: Current thread: 0x20400540 (main)
DEBUG   - DEVICE: E: Halting system
DEBUG   - Timed out while monitoring serial output on sam_e70_xplained
DEBUG   - run status: sam_e70_xplained/tests/benchmarks/cmsis_dsp/basicmath/benchmark.cmsis_dsp.basicmath.fpu timeout
INFO    - 2/2 sam_e70_xplained          tests/benchmarks/cmsis_dsp/basicmath/benchmark.cmsis_dsp.basicmath.fpu FAILED Timeout (device 63.067s)
```

**Environment (please complete the following information):**
OS: Linux,
Toolchain Zephyr SDK 0.13.2
Commit SHA or Version: 900b6723e76897841888a18d20d710d7dfdbee7c

stephanosio, 2022-04-28:
Can you try on the latest main? The commit https://github.com/zephyrproject-rtos/zephyr/commit/900b6723e76897841888a18d20d710d7dfdbee7c has been reverted: https://github.com/zephyrproject-rtos/zephyr/commit/462b13e16e013a7c8e9c7187ca6a01b99c19494f

cc @yperess

shaoanxli, 2022-04-28:
> Can you try on the latest main? The commit [900b672](https://github.com/zephyrproject-rtos/zephyr/commit/900b6723e76897841888a18d20d710d7dfdbee7c) has been reverted: [462b13e](https://github.com/zephyrproject-rtos/zephyr/commit/462b13e16e013a7c8e9c7187ca6a01b99c19494f)
> 
> cc @yperess

I tried on the latest main. The test case ran pass.

stephanosio, 2022-04-28:
Thanks for confirming.

Closing this since the issue has been fixed by reverting the offending commit.

closed 2022-04-28 by stephanosio