zephyrproject-rtos/zephyr #45211: settings - erase all feature

volodyaleo, 2022-04-28:
**Describe the solution you'd like**
I suggest to introduce a function to erase all settings.

There are two issue examples:
- device factory reset
- failure repair

In both cases easier way is to erase whole storage and reboot.

```c
void do_factory_reset(void)
{
       settings_erase_all();
       sys_reboot(SYS_REBOOT_COLD);
}
```

or

```c
ret = settings_load();
if (ret < 0) {
     // probably hardware is broken or software (fs) error
     // if software error, try to erase storage. It may help to fix
     settings_erase_all();
     sys_reboot(SYS_REBOOT_COLD)
}

ret = validate_app_settings();
if (ret < 0) {
     // app settings was corrupted (unexpected error)
     // probably device firmware was upgraded into incompatible version
     // reset settings to defaults
     settings_erase_all();
     sys_reboot(SYS_REBOOT_COLD)
}
```

The settings_erase_all may use functions bellow to make reliable full erase:
- nvs_clear
- fcb_clear
- fs_unlink

Laczen, 2022-04-28:
@volodyaleo, thanks for this enhancement proposal. 

I see one possible problem, what if your factory settings are stored in the settings backend?

volodyaleo, 2022-04-28:
@Laczen , thanks for quick answer.

The erase_all function pointer could be placed to custom backend structure. If somebody store factory settings in the backend he could also implement own erase_all function.

I had no issues with `settings_load()`. It was just an example. But I had issues with another sdk (not zephyr) key-value storage.

Such feature may be useful to keep device operable in unpredictable cases like:
- bugs
- flash map changes (some key-value storages may loose swap page and ability to operate)
- firmware versions incompatibility (new firmware replaced by old, etc)
- the data was created by previous firmware version, current firmware has no memory to save own data and don't know the old data name

The general idea is to have simple and reliable way for erase.

The one more use example - if user sold device, need to have method to empty flash. Sometimes devices have no flash readback protection, more reliable is to erase storage flash sectors. It is applicable for nvs.

volodyaleo, 2022-04-30:
@Laczen So, probably your suggestion is useful ( settings_delete_subtree(char *name) ).

If somebody store every setting with his manufacturer specific prefix (for example, /<company>/<module>) it may be useful to have ability to erase all his own settings. If user requested factory reset or settings validation failed this function may be used instead of one by one parameter deletion.

rettichschnidi, 2022-05-16:
I intend to come up with a solution related to this issue, willing to upstream if deemed a good idea:

- Create a "factory data" module, which is backed by FCB and can only be written to using a shell during manufacturing. Data can bot be changed - if something goes wrong during manufacturing, the factory data FCB partition needs to be completely wiped and started over. Having a separate partition is needed so that the settings partition can be securely wiped without putting the factory data at risk
- Optionally: Hook up the factory data FCB partition as source backend (read only) into the settings module. Would allow to access factory and user data seamlessly.

Laczen, 2022-05-16:
> I intend to come up with a solution related to this issue, willing to upstream if deemed a good idea:
>
> * Create a "factory data" module, which is backed by FCB and can only be written to using a [...]

This seems like a good idea, why not a TLV system direct on flash or eeprom? It that case the shell requirement could be avoided.

rettichschnidi, 2022-05-16:
> This seems like a good idea, why not a TLV system direct on flash or eeprom? It that case the shell requirement could be avoided.

I'd rather do not want to create something custom when we have a solution which is almost ready-made (FCB). Maybe I miss something?

Laczen, 2022-05-16:
> > This seems like a good idea, why not a TLV system direct on flash or eeprom? It that case the shell requirement could be avoided.
> 
> I'd rather do not want to create something custom when we have a solution which is almost ready-made (FCB). Maybe I miss something?

If you are already using fcb, this makes sense, otherwise it is another "file system" that is needed. Also it doesn't support eeprom (but then again, why would you use a eeprom for fixed data).

> 
> Using the shell is just one way of getting data on to the factory data partition. Could also be pre-flashed (FCB seems simple enough?) or created using the (new factory data modules) APIs. For my needs, a shell would be the preferred solution so this will be what I'll implement first.

To be honest, this should already be supported. Only writing to one backend is allowed, multiple reads is supported from the start. For the factory data write it would be needed to register the factory data as write and other backends as read. For normal operation it would be needed to register the other backend write and the factory data as read.

rettichschnidi, 2022-05-16:
> If you are already using fcb, this makes sense, otherwise it is another "file system" that is needed. Also it doesn't support eeprom (but then again, why would you use a eeprom for fixed data).

Have not yet used it, but would go for it because (seems like) low effort and also, has a CRC.

> To be honest, this should already be supported. Only writing to one backend is allowed, multiple reads is supported from the start. 

Will try to go this way. One issue however will probably need some extra code to avoid: Writing a value under the same name as the factory data will end up in the write-enabled backend of the settings subsystem. Once this happens, both sources will be iterated over on `settings_load()`.

> For the factory data write it would be needed to register the factory data as write and other backends as read. For normal operation it would be needed to register the other backend write and the factory data as read.

Would rather not hot-swap the write backend of the settings subsystem because random setting writes from any part of the system could taint the factory data.

owain-incus, 2022-08-22:
settings_erase_all() or settings_load_defaults() should be part of the setttings API whatever backend is in use.

I was using FAT as my settings storage; but even in this case I had to resort to overwriting MBR/filesystem headers with a low level flash API to force system to do a mkfs.

Should be able to do all from settings API without such nastiness.

Now I have just started using settings NVS; but am probably gong to have to sneak under settings to do something like nvs_clear().

Laczen, 2022-08-22:
> I was using FAT as my settings storage; but even in this case I had to resort to overwriting MBR/filesystem headers with a low level flash API to force system to do a mkfs.
>
Have you tried just deleting the settings file ? The filename is a config variable so deleting it should be straightforward. Even copying a file with default settings would give you a load defaults solution.

owain-incus, 2022-08-22:
Unfortunately must remove all user data. On other products we had a system
and user partition, so ability to either remove all user data or user data
and system settings. Which was done by rebuilding appropriate filesystems.
The subsystems were then responsible for defaults.