zephyrproject-rtos/zephyr #45218: rddrone_fmuk66: I2C

landonh12, 2022-04-28:
@dleach02 @danieldegrasse I am submitting this issue for my own tracking, I do not expect you two to put much effort towards this at the moment. :)

**Describe the bug**
I2C bus on rddrone_fmuk66 is broken

I have a branch on a fork to NXPHoverGames/zephyr where I am trying to add support for on-board sensors in the rddrone_fmuk66 board.

When testing samples/bmm150 or samples/bme280, the initial probing of the sensor fails. If you probe the I2C1_SDA_INTERNAL and I2C1_SCL_INTERNAL test points (TP27 and TP28, respectively) on the board and watch the I2C waveform, you will notice that the data line floats at 0.7v when the sensor tries to ACK. It seems that the sensor can not effectively drive to 0V. Image below:

[image]

Note that this is without the drive-open-drain parameter in the i2c1 device in the dtsi for the board. The reference manual states that all devices connected to the i2c bus must have open drain or open collector outputs. If drive-open-drain is configured as a parameter for the i2c1 device in the dtsi, the bus stays driven to 0V by the MCU. If bias-pull-up and drive-open-drain are enabled, then the bus stays at 320mV. By "bus stays at", I mean both the SCL and SDA lines are driven to that voltage.

Note the waveform seen if we try to run the samples/bme280 program with drive-open-drain enabled and bias-pull-up disabled:

[image]

**To Reproduce**
Steps to reproduce the behavior:
1. Build and flash the samples/bmm150 or samples/bme280 programs to the rddrone_fmuk66 board with the fork at https://github.com/nxphovergames/zephyr/tree/rddrone_fmuk66-i2c

**Expected behavior**
The sensors configured in the device tree should communicate with the FMUK66.

**Impact**
Showstopper - we are trying to enable the sensors on-board so we can use the device as a controller for Buggy3.

**Logs and console output**
For BMM150:
```
*** Booting Zephyr OS build zephyr-v3.0.0-3202-g69b28bfd076c  ***
BMM150 Geomagnetic sensor Application
There is no available Geomagnetic device.
[00:00:00.000,000] <err> BMM150: failed to bring up device from normal mode
[00:00:00.000,000] <err> BMM150: failed to initialize chip
```

For BME280:
```
[00:00:00.000,000] <dbg> BME280: bme280_chip_init: ID read failed: -5
*** Booting Zephyr OS build zephyr-v3.0.0-3203-gb31e736fcc86  ***

Error: Device "bme280" is not ready; check the driver initialization logs for erro.
```

**Environment (please complete the following information):**
Ubuntu 20.04
Tree: https://github.com/nxphovergames/zephyr/tree/rddrone_fmuk66-i2c

dleach02, 2022-05-03:
@landonh12, I'm going to assign this one to you for the moment

landonh12, 2022-07-03:
Please remove the stale label, this is actively being worked on internally. Will provide details when back in office Tuesday.

landonh12, 2022-07-04:
Some findings (for bmm150 sample):

1. drive-open-drain should be enabled for the data line, but not the clock. I have created two separate groups in the dtsi for i2c pinmux. 
2. The 3V3_S rail does not come on because regulator drivers are not enabled in these samples by default. If we enable them through menuconfig, the 3V3_S rail comes on. This is an important step.
3. The I2C bus init seems to be in a race condition with the application. If I boot with a normal build, the bmm150 application states that it can not bring up the device from normal mode. If we hook up a scope, we can't see the i2c lines switch at all. After putting a k_msleep(1000) in the bmm150 driver, I observed that the i2c bus does indeed switch, but the driver still fails to probe a device.

dleach02, 2022-11-03:
@landonh12, are you still looking at this issue?

danieldegrasse, 2022-11-03:
@dleach02, I believe that https://github.com/zephyrproject-rtos/zephyr/pull/51218 may address this. @landonh12, can you confirm?

landonh12, 2022-11-03:
Yes, this issue is the root of the sensor work being done in #51218.

danieldegrasse, 2023-01-03:
@landonh12, since #51218 is merged, I'm going to close this issue. Please reopen it if the problem still exists.

closed 2023-01-03 by danieldegrasse