zephyrproject-rtos/zephyr #45230: Bluetooth: Host: Read maximum advertising data length

Thalley, 2022-04-29:
**Is your enhancement proposal related to a problem? Please describe.**
The host is now capable of segmenting and sending up to 1650 octets for extended advertising, but the controller may not support such large values (based on `CONFIG_BT_CTLR_ADV_DATA_LEN_MAX`). 

The host should read the maximum adverting data length and reject requests for values larger than the response when the application requests it. 

**Describe the solution you'd like**
On host init do a call to send the `bt_hci_rp_le_read_max_adv_data_len` request to the controller and store the r`esult, as we do for e.g. ACL and ISO buffer sizes, and rejects requests to `bt_le_ext_adv_set_data` and `bt_le_per_adv_set_data` with too large advertising data. 

**Describe alternatives you've considered**
N/A

**Additional context**
N/A

Thalley, 2022-04-29:
ping @cvinayak @rugeGerritsen

rugeGerritsen, 2022-04-29:
I guess we can set a low priority on this? We can rely on the controller returning an error code when setting too much data

Thalley, 2022-04-29:
> I guess we can set a low priority on this? We can rely on the controller returning an error code when setting too much data

Aren't priorities for bugs only? 

Should we expect the controller to do the checks if `CONFIG_BT_CTLR_PARAM_CHECK=n`?

rugeGerritsen, 2022-04-29:
> 

Good point. I was not aware of this configuration, and I see it is specific to BT_LL_SW_SPLIT. Disabling this configuration puts the controller in a non-qualifiable state. In either case it may be a good idea to add this functionality to the host as it improves the debugging experience

closed 2024-04-29 by carlescufi