zephyrproject-rtos/zephyr #45234: STM32: Allow multiple GPIOs to trigger an interrupt

lsangild, 2022-04-29:
**Is your enhancement proposal related to a problem? Please describe.**
Due to the by-pin-number setup of the STM32 EXTI controller, it is currently only possible to have one interrupt per pin number, e.g. only PB10, not PB10 and PC10.

**Describe the solution you'd like**
I can handle one interrupt from multiple port, as shown in the irq_handler `gpio10_cb` above, but I am not allowed to setup the IRQ due to https://github.com/zephyrproject-rtos/zephyr/blob/44f3f2e49bccc889413c5b940f97a64d2aac2aa7/drivers/interrupt_controller/intc_exti_stm32.c#L437
Could this check be changed to allow the setup if the isr is either not set or is the same as the previously set one?

**Describe alternatives you've considered**
Currently I have to prioritize which GPIO states must be interrupts and which has to be polled.

**Additional context**
Example code tested on NucleoH743ZI
```c
#include <device.h>
#include <drivers/gpio.h>
#include <zephyr.h>

void gpio10_cb(
    const struct device * port,
    struct gpio_callback * cb,
    gpio_port_pins_t pins
) {
    if( port == DEVICE_DT_GET( DT_NODELABEL( gpiob ) ) ) {
        // Do one thing
    }
    else if ( port == DEVICE_DT_GET( DT_NODELABEL( gpioc ) ) ) {
        // Do something else
    }
}

void main(void)
{
    // Setup IRQ on PB10
    int rc = gpio_pin_interrupt_configure(
        DEVICE_DT_GET( DT_NODELABEL( gpiob ) ),
        10U,
        GPIO_INT_EDGE_BOTH
    );
    static struct gpio_callback gpiob10_cb;
    gpio_init_callback(
        &gpiob10_cb,
        &gpio10_cb,
        BIT( 10 )
    );
    rc = gpio_add_callback(
        DEVICE_DT_GET( DT_NODELABEL( gpiob ) ),
        &gpiob10_cb
    );

    // Setup IRQ on PC10
    rc = gpio_pin_interrupt_configure(
        DEVICE_DT_GET( DT_NODELABEL( gpioc ) ),
        10U,
        GPIO_INT_EDGE_BOTH
    );
    static struct gpio_callback gpioc10_cb;
    gpio_init_callback(
        &gpioc10_cb,
        &gpio10_cb,
        BIT( 10 )
    );
    rc = gpio_add_callback(
        DEVICE_DT_GET( DT_NODELABEL( gpioc ) ),
        &gpioc10_cb
    );
}
```

lsangild, 2022-05-02:
Hmmm, I see.
`

> > [...] if the isr is either not set or is the same as the previously set one?
> 
> Can you clarify the request ? Do you mean a callback is configured ?
> 
> The problem is that because of the hw limitation (One IRQ line mapped to a pin number), we're not able to discriminate the port. Interrupt raised on line 10 could be either for port B or C, there's no way say more.
> 
> Did I misunderstood the request ?

Hmm I see.

Yes, if the callback is the same as the one already configured, could it be made so that both e.g. PB10 and PC10 triggers callbacks, and then you discriminate on the triggering pin by reading the value of the pins? The change in Zephyr would here be that the callback configuration is allowed, but it is up to the programmer to discriminate which pin triggered the ISR.
I know this is not the optimal solution, but not allowing me to use the same callback function for both, takes away a lot of versatility, and forces me use processing time on polling otherwise non-interesting signals, which, in theory, should never change state.

erwango, 2022-05-02:
> Hmm I see.
>
> Yes, if the callback is the same as the one already configured, could it be made so that both e.g. PB10 and PC10 triggers callbacks, and then you discriminate on the triggering pin by [...]

Well, I guess we can implement this as a Kconfig option, deactivated by default.

FRASTM, 2022-06-30:
@lsangild could you please check if PR https://github.com/zephyrproject-rtos/zephyr/pull/47108 answers your pb.
--> add the  a prj.conf to build your application
```
CONFIG_EXTI_STM32=y
CONFIG_EXTI_STM32_ONE_IRQ=y
```

lsangild, 2022-07-05:
@FRASTM I have tried to use the fix you provided, but can't seem to make it work.

I dug a little deeper in the Reference Manual [RM0433 Rev 7](https://www.st.com/resource/en/reference_manual/rm0433-stm32h742-stm32h743753-and-stm32h750-value-line-advanced-armbased-32bit-mcus-stmicroelectronics.pdf), and on pages 548-551 it seems that the SYSCFG registers SYSCFG_EXTICRx does not allow for multiple ports to trigger an IRQ. You can only select one port for each GPIO number. So e.g. SYSCFG_EXTICR4[EXTI13] can only be enabled for one port at a time.
This means my request cannot be solved, at least for the STM32H743 processor.

FRASTM, 2022-07-05:
This is the case for stm32 devices : one EXIT line can input only one among 4-bit max pins 
With stm32H7, the irq line 23 is common to EXTI9_5. So, if it helps to your application, you can still consider that one irq line corresponds to several EXTI lines.

FRASTM, 2022-07-07:
> This means my request cannot be solved, at least for the STM32H743 processor.

@lsangild  should we keep this issue ?

lsangild, 2022-07-15:
I don't think so. I have been given the answers I need, unfortunately my solution is not possible.
I guess I'll have to stick to the polling, and use interrupts for the most critical signals.

closed 2022-07-15 by lsangild