zephyrproject-rtos/zephyr #45241: (Probably) unnecessary branches in several modules

mbolivar-nordic, 2022-04-29:
**Describe the bug**

Several modules in the upstream west.yml file point to forks that have what look like unnecessary branches.

Each unnecessary branch:

1. creates wasted network traffic, since `west update` will fetch all remote branches by default (i.e. unless `west update --narrow` is used)
2. clutters the `west update` output for users, who see a line printed for each unnecessary branch:

```
=== updating some_module (modules/...):
--- some_module: fetching, need revision ########################################
From https://github.com/zephyrproject-rtos/some_module
 * [new branch]            asdfsdfsdfsdfsdf                  -> refs/west/asdfsdfsdfsdfsdf
 * [new branch]            asdfsdfsdfsdfsdf                  -> refs/west/asdfsdfsdfsdfsdf
 * [new branch]            asdfsdfsdfsdfsdf                  -> refs/west/asdfsdfsdfsdfsdf
[...]
 * [new branch]            zephyr                            -> refs/west/zephyr
```

**To Reproduce**

1. west init zephyrprpoject
2. cd zephyrpoject
3. west update
5. See error

**Expected behavior**

No branches should be present in the zephyrproject-rtos GitHub organization's fork of a module repository unless the branch contains commits that are actually needed by some west.yml file in zephyr's history. (So keeping 'zephyr' and 'v1.14-branch' branches is OK, but topic branches are not OK.)

**Impact**

Annoyance

**Logs and console output**

Here is a list of modules and branches that seem probably unnecessary to me.

civetweb:

-  wip-ssl-error-queue

hal_espressif:

- bugfix/espc32c3_eco0
- bugfix/warning_redefinition_esp32
- feature/clock_esp32c3
- feature/esp32c3_ble
- feature/esp32c3_boot_zephyr_sdk
- feature/esp32c3_intc
- feature/esp32c3_support
- feature/esp32c3_wifi
- feature/gpio_esp32c3
- feature/intc_bt_wifi
- feature/serial_driver_esp32c3
- feature/spi_driver_esp32c3
- feature/timer_driver_esp32c3
- hal_v4.2
- hal_v4.3
- zephyr_v4.3

hal_nordic:

- topic-nrfx-2.0.0

hal_nxp:

- agansari-patch-1

hal_rpi_pico:

- bugfix
- comments_for_register_arrays
- develop
- doc-build
- fix-uart-lcr-write
- host_packed
- irq_clear_comment
- mutex_speed
- pwm-comment-enable-disable
- track_upstream_tinyusb

hal_xtensa:

- RF

liblc3codec:

- sneak_peek

lz4:

- Makefile
- Makefile_export
- coverity_scan
- dev
- gh-pages
- install_ln
- lz4io
- release
- vs2005

mbedtls:

- mbedtls_move_zephyr_files
- tls-2.26-branch

mcuboot:

- add-license
- coverity_scan
- inclusive_lang
- revert-368-s1000_support
- sim
- suit
- v1.3-branch
- v1.3.1-branch

nanopb:

- cifuzz_test4
- dev_dynamic_nanopb_pb2
- dev_markdown_docs
- maintenance_0.1
- maintenance_0.2
- maintenance_0.3

tinycbor:

- dev
- gh-pages

TraceRecorderSource:

- up2

trusted-firmware-m:

- legacy

psa-arch-tests:

- ew_beta0
- key_derivation
- release
- release-attestation-1.0-beta0
- release-crypto-mbedcrypto-1.0.0
- v0.9_staging

zscilib:
- gh-pages

Casper-Bonde-Bose, 2022-04-29:
liblc3codec is clean

dcpleung, 2022-04-29:
`hal_xtensa` is scrubbed.

Navin-Sankar, 2022-05-02:
**lz4** is clean

microbuilder, 2022-05-02:
@mbolivar-nordic The `gh-pages` branch in zscilib is used to store the following documentation and host it via Github: https://zephyrproject-rtos.github.io/zscilib/ ... seems like a valid use case to have a (smaller) branch???

tleksell-pe, 2022-05-02:
@nashif the up2 branch in the TraceRecorderSource repo was created by you, does it fill any purpose or can it be deleted?

stephanosio, 2022-05-02:
> The `gh-pages` branch in zscilib is used to store the following documentation and host it via Github:

@microbuilder Is there a separate "upstream" repository for the `zscilib`? or is https://github.com/zephyrproject-rtos/zscilib in itself supposed to be the upstream?

If the latter, I suppose we have no option but to leave the branch as-is since it is a special case (we usually have separate upstream repos).

microbuilder, 2022-05-02:
@stephanosio This is the 'home' repo for zsclib.

dleach02, 2022-05-04:
hal_nxp cleaned up

mbolivar-nordic, 2022-05-13:
Thanks to everyone who's updated so far!

Pinging the maintainers for modules which are still TODO:

- hal_nordic: @anangl
- mbedtls: @d3zd3z @ceolin
- mcuboot: @d3zd3z
- TraceRecorderSource: @tleksell-pe
- trusted-firmware-m: @microbuilder
- psa-arch-tests: @microbuilder

mbolivar-nordic, 2022-05-13:
> TraceRecorderSource: @tleksell-pe

Sorry, this should have been a ping to @nashif given this comment:

https://github.com/zephyrproject-rtos/zephyr/issues/45241#issuecomment-1114663654

d3zd3z, 2022-05-25:
I've cleaned up the branches in mcuboot.

mbolivar-nordic, 2022-07-25:
Pinging again:

- hal_nordic: @anangl
- mbedtls: @d3zd3z @ceolin
- TraceRecorderSource: @nashif

anangl, 2022-07-26:
> Pinging again:
> 
> * hal_nordic: @anangl

?
`hal_nordic` has already been cleaned up. There is only `v1.14-branch` there, what is okay according to this issue description. Am I missing something?

closed 2023-08-13 by github-actions[bot]