zephyrproject-rtos/zephyr #45248: Please do not redefine 32bit integer types like __UINT32_TYPE__

rweickelt, 2022-04-29:
**Describe the bug**

Zephyr [redefines built-in 32 bit integer types](https://github.com/zephyrproject-rtos/zephyr/pull/16645). While the arm-none-eabi-gcc package shipped by ARM uses "long int", Zephyr uses just "int". The end result may look like the same in C, but this creates problems when linking external C++ libraries. C++ mangles types into the symbol name. "long int" and "int" result in different mangled symbol names. When linking a Zephyr application to an externally built C++ library, symbols will not be found.

**Expected behavior**

I expect Zephyr to keep all types as defined by the toolchain.

**Impact**

Linker errors when linking external C++ libraries built with the same toolchain.

stephanosio, 2022-04-29:
This was a deliberate decision as described in the issue you linked (https://github.com/zephyrproject-rtos/zephyr/pull/16645).

For now, the expectation for the third-party libraries to be used with Zephyr is that they should include the Zephyr `zephyr_stdint.h` globally so that they use the same type system as the upstream Zephyr code.

The plan is to eventually get rid of this "hack":
https://github.com/zephyrproject-rtos/zephyr/issues/37718#issuecomment-917435777

stephanosio, 2022-04-29:
Please add something like the following when you are building third-party libraries:
https://github.com/zephyrproject-rtos/zephyr/blob/bf3cd116e9f553c33b7d2ef483a3ea00132a5bf1/CMakeLists.txt#L288

rweickelt, 2022-05-02:
Thank you for the suggested workaround. I will instead define a replacement type for uint32_t in the public API of our library.

I would be glad if this hack is removed.

stephanosio, 2022-05-26:
Closing in favour of https://github.com/zephyrproject-rtos/zephyr/issues/46032, which provides the implementation plan for this.

closed 2022-05-26 by stephanosio