zephyrproject-rtos/zephyr #45261: mcumgr: conversion of version to string fails (snprintf format issue)

LeBlue, 2022-04-30:
**Describe the bug**
The version information is not correctly converted to a string when querying image slots.

**To Reproduce**
Steps to reproduce the behavior:
run `mcumgr image list` command with suitable firmware.

The version string received is: `hu.hu.`

**Expected behavior**
Version should be converted correctly

**Impact**
showstopper

**Logs and console output**

hexdump of response:
<pre>
    0000  01 00 00 87 00 01 00 00  bf 66 69 6d 61 67 65 73  |.........fimages|
    0016  9f bf 64 73 6c 6f 74 00  67 76 65 72 73 69 6f 6e  |..dslot.gversion|
    0032  66 68 75 2e 68 75 2e 64  68 61 73 68 58 20 c5 ef  |f<b>hu.hu.</b>dhashX ..|
    0048  01 3a 69 e8 2e f7 f8 24  e9 83 86 46 a0 58 48 3b  |.:i....$...F.XH;|
    0064  49 27 a3 64 0d f0 37 0d  01 56 e2 a8 e9 16 68 62  |I'.d..7..V....hb|
    0080  6f 6f 74 61 62 6c 65 f5  67 70 65 6e 64 69 6e 67  |ootable.gpending|
    0096  f4 69 63 6f 6e 66 69 72  6d 65 64 f5 66 61 63 74  |.iconfirmed.fact|
    0112  69 76 65 f5 69 70 65 72  6d 61 6e 65 6e 74 f4 ff  |ive.ipermanent..|
    0128  ff 6b 73 70 6c 69 74 53  74 61 74 75 73 00 ff     |.ksplitStatus.. |
</pre>

gdb log

<pre>
Breakpoint 2, img_mgmt_ver_str (ver=ver@entry=0x2000272c <sys_work_q_stack+1492>, dst=dst@entry=0x20002754 <sys_work_q_stack+1532> "hu.hu.")
    at $ZEPHYR_BASE/subsys/mgmt/mcumgr/lib/cmd/img_mgmt/src/img_mgmt_util.c:15
15	{
(gdb) p ver
$11 = (const struct image_version *) 0x2000272c <sys_work_q_stack+1492>
(gdb) p *ver
$12 = <b>{iv_major = 0 '\000', iv_minor = 0 '\000', iv_revision = 1, iv_build_num = 0}</b>
(gdb) l
10	#include "img_mgmt/image.h"
11	#include "img_mgmt/img_mgmt.h"
12	
13	int
14	img_mgmt_ver_str(const struct image_version *ver, char *dst)
15	{
16		int rc = 0;
17		int rc1 = 0;
18	
19		rc = snprintf(dst, IMG_MGMT_VER_MAX_STR_LEN, <b>"%hhu.%hhu.%.hu"</b>,
(gdb) n
19		rc = snprintf(dst, IMG_MGMT_VER_MAX_STR_LEN, "%hhu.%hhu.%.hu",
(gdb) p rc
$14 = 6
(gdb) p dst
$15 = 0x20002754 <sys_work_q_stack+1532> <b>"hu.hu."</b>
...
img_mgmt_state_read (ctxt=<optimized out>) at $ZEPHYR_BASE/subsys/mgmt/mcumgr/lib/cmd/img_mgmt/src/img_mgmt_state.c:235
235					vers_str[sizeof(vers_str) - 1] = '\0';
(gdb) p vers_str 
$17 = "hu.hu.\000\252\315\025\001\020\000\000\000\000\223+\000 \247'\000", <incomplete sequence \315>
</pre>

**Environment (please complete the following information):**
 - Commit SHA 3.0.99: main@83c79d10510e28e8706581168d257c1b556732d2
 
**Additional context**
Probably relevant config of libc

```
Config lib c

#
# C Library
#
CONFIG_SUPPORT_MINIMAL_LIBC=y
# CONFIG_MINIMAL_LIBC is not set
CONFIG_NEWLIB_LIBC=y
# CONFIG_EXTERNAL_LIBC is not set
CONFIG_HAS_NEWLIB_LIBC_NANO=y
CONFIG_NEWLIB_LIBC_NANO=y
CONFIG_NEWLIB_LIBC_MIN_REQUIRED_HEAP_SIZE=2048
# CONFIG_NEWLIB_LIBC_FLOAT_PRINTF is not set
# CONFIG_NEWLIB_LIBC_FLOAT_SCANF is not set
# CONFIG_NEWLIB_LIBC_HEAP_LISTENER is not set
```

**Tests/Fixes**
Problem was found with a LPC55S28 SoC.

There are two problems here
1. `%hhu` format is not recognized with this libc configuration. Might also be an integer width problem?
2. The version number is not correctly formated, if build_num != 0 and patch == 0: `255.255..4294967295`

Note: the following format string works for this configuration: `"%hu.%hu.%hu"`, also for the maximum possible version+build number and also problem 2.

LeBlue, 2022-04-30:
1. Problem might be related to https://github.com/zephyrproject-rtos/zephyr/issues/45209

de-nordic, 2022-05-04:
Provided fix here, https://github.com/zephyrproject-rtos/zephyr/pull/45329 , nevertheless I find fixing the %hhu to be actually workaround for New C lib not correctly processing it.

stephanosio, 2022-05-04:
See https://github.com/zephyrproject-rtos/zephyr/pull/45329#discussion_r864711556

PR #45329 (2022-05-15): Fix version string formatting
Fixes astray dot (my bad) and a problem with newlib not recognizing %hhu formatting ~~(their bad)~~ when `CONFIG_NEWLIB_LIBC_NANO=y`

Fixes #45261
commit 578cb132 (2022-05-15): mgmt/mcumgr/lib: Make version string work with CONFIG_NEWLIB_LIBC

By default the newlib does not compile with %hhu processing,
to avoid turning on additional options for newlib compilation
this commit changes formatting string to use %hu instead.
The commit fixes problem with malformed version string being
returned, by smp_svr, when CONFIG_NEWLIB_LIBC=y is set together
with CONFIG_NEWLIB_LIBC_NANO=y.

Fixes #45261
closed 2022-05-15 by carlescufi