zephyrproject-rtos/zephyr #45279: Update to CMSIS 5.9.0

stephanosio, 2022-05-02:
**Describe the solution you'd like**
[CMSIS 5.9.0 has been released.](https://github.com/ARM-software/CMSIS_5/releases/tag/5.9.0)

Update the following CMSIS components in the [Zephyr CMSIS fork](https://github.com/zephyrproject-rtos/cmsis) and integrate them to the Zephyr:

* CMSIS-Core
* CMSIS-DSP
* CMSIS-NN

stephanosio, 2022-05-02:
cc @JordanYates @microbuilder @povergoing

stephanosio, 2022-05-02:
Since the feature freeze for 3.1 is in two weeks (2022/05/13), we should target 3.2 for this feature.

stephanosio, 2022-08-30:
This will not make into 3.2. Targeting a future release.

kartben, 2023-12-14:
Pretty sure this is done now, letting @stephanosio confirm

nashif, 2023-12-15:
yes, it is done, last commit:

```
commit 4b96cbb174678dcd3ca86e11e1f24bc5f8726da0 (HEAD, manifest-rev)
Author: Marcin Niestroj <m.niestroj@emb.dev>
Date:   Mon Nov 6 12:29:38 2023 +0100

    doc: Update documentations for CMSIS 5.9.0

    This commit updates the documentations for CMSIS 5.9.0.

    Signed-off-by: Marcin Niestroj <m.niestroj@emb.dev>
```

closed 2023-12-15 by nashif