zephyrproject-rtos/zephyr #45307: Improve documentation on backport / release process

cfriedt, 2022-05-03:
**Is your enhancement proposal related to a problem? Please describe.**
While it is OK to submit PRs without an associated issue to `main`, all backport PRs require an associated issue. Even those automatically created by `zephyrbot`.

If the original PR to `main` does have a linked issue, or if the backport is part of a larger qualification effort with an issue, or if the backport is part of a CVE with an issue, then a new issue does not need to be created, and the existing associated issue can simply be logged in the description of the backport PR.

Otherwise, if no associated issue exists, then a new issue should be created and linked to the backport PR.

**Describe the solution you'd like**
@dkalowsk pointed out that MCUBoot has a good document w.r.t. backporting, which is below

https://docs.zephyrproject.org/3.0.0/guides/device_mgmt/mcumgr_backporting.html

Perhaps it makes sense to generalize this document and include it in the release process documentation.

**Describe alternatives you've considered**
Currently, the release team needs to remind PR authors to create an associated issue, and it's a very common occurrence. 

**Additional context**

keith-zephyr, 2023-09-13:
@carlescufi or @cfriedt, is this something either one of you can take on?

carlescufi, 2023-10-20:
> @carlescufi or @cfriedt, is this something either one of you can take on?

Sure, I will try to send something.

cfriedt, 2023-10-20:
Thanks - I'm a bit behind on docs ATM.

fabiobaltieri, 2023-10-20:
@cfriedt what if instead of running the "Backport Issue Check" we change it to run on the original PR instead, and fail if there's any backport tag but there's no issue associated with that one? Then as long as the tags are added before merge we'd always have an issue on the original PR, and then that's much easier to handle during backport triaging.

cfriedt, 2023-10-20:
> @cfriedt what if instead of running the "Backport Issue Check" we change it to run on the original PR instead, and fail if there's any backport tag but there's no issue associated with that one? [...]

Yeah, sounds good. Also could automatically insert the existing linked issue "fixes.." in the new backport pr description

keith-zephyr, 2024-06-26:
Process WG: this item also needs to be moved to the Release WG.

keith-zephyr, 2026-10-19:
This should be owned by the Release WG.

dkalowsk, 2026-10-19:
I had forgotten about this one.  @d3zd3z note the security details fall in line more with what was discussed at the ZMF in Prague.