zephyrproject-rtos/zephyr #45319: Add MPS2 AN386 board for emulated Cortex-M4 test coverage

stephanosio, 2022-05-04:
**Is your enhancement proposal related to a problem? Please describe.**

As noted in #22870, we do not currently have Cortex-M4 test coverage in the CI because Zephyr does not support any emulation-capable boards with a Cortex-M4 core.

**Describe the solution you'd like**

Add `mps2_an386` board support (mainly for emulation).

**Describe alternatives you've considered**

There is #22870, which tried to introduce Cortex-M4 emulation support through the `netduinoplus2` machine type, which is based on a STM32 SoC with a Cortex-M4 core; but, it is better to target an ARM reference platform.

**Additional context**

[MPS2 AN386 documentation](https://developer.arm.com/documentation/dai0386/latest/)
[QEMU MPS2 AN386 implementation](https://github.com/qemu/qemu/commit/897d27260a7eaccbf4ff01e49021205c5616c8ef)

stephanosio, 2022-05-04:
cc @microbuilder @povergoing

microbuilder, 2022-11-08:
I see QEMU was also recently updated with the mps2-an500, which is an M7 and may include support for double-precision floating point operations: https://github.com/qemu/qemu/commit/6d4811c4b688de368748c8095250ba12c905a21c

stephanosio, 2022-11-08:
> I see QEMU was also recently updated with the mps2-an500, which is an M7 and may include support for double-precision floating point operations: [qemu/qemu@6d4811c](https://github.com/qemu/qemu/commit/6d4811c4b688de368748c8095250ba12c905a21c)

Yes, https://github.com/zephyrproject-rtos/zephyr/issues/22870#issuecomment-1116967262. Though, adding support for it will likely involve a bit more work than this.

JPHutchins, 2023-05-01:
If I understand correctly, the current solution to test emulated M4 / M33 is to use `mps2_an521` like this?

```yaml
tests:
  sample.testing.ztest:
    platform_allow: mps2_an521
    tags: testing
```