zephyrproject-rtos/zephyr #45323: Implement handling of delayed notifications

erbr-ot, 2022-05-04:
Figure out how to and implement handling delay in sending notification to host.

kruithofa, 2022-09-13:
Which procedures are involved in this behaviour:
- Instant based procedures
- Procedures where you would send a notification after responding

In legacy llcp we piggy back on the rx-node
Possible solution: piggy back on the rx-node
- Does not work for instant based procedures, and other procedures that do not get a response
- Create a queue for delayed notifications, this queue contains the connection-handle and the procedure

Action items:	
- Create overview of affected procedures for both central and peripheral
- Evaluate whether we use the ‘delayed queue’ always, also when resources are available, or only when no resources are available. Current preference is to use the ‘delayed queue’ always.
- Create new github issue for Implementing piggy backing on the rx-node

erbr-ot, 2023-01-10:
Notes re. DELAYED NOTIFICATION HANDLING

* Instant based procedures cannot piggy-back on RX node for NTF as RX node is released pre instant,
  thus alloc can fail @ instant
* Delay in handling notification leads to air-complete procedure, and possibly non complete proc
* Spec indicates that NTF should not be cached/over-written, ie every individual NTF should be sent up the 
  chain (e.g. p2257 re. DLE)
* Not viable to nACK remote request when in 'air-complete pending NTF completion' state
* Possible to stall remote procedure (procedure timeout) to allow NTF to complete
     - accept remote req, hold off sending response until NTF wait done
* Local initiated procs are implicitly queued up, as they do not complete-complete until NTF is handled
* If local/remote initiated procedure is NTF wait, reset compatibility flag on instant, 
  so collision check on remote req is valid
* Previously propose solution: Queue up remote request: allocate new CTX, decode PDU content and push into pending queue, requires 
  splitting proc prepare and rx - NOTE this will 'destroy' piggy-back'ing
     - remote reqs resulting in NTF: DLE, CU, PU, ENC (one little piggy!), CC (another little piggy!)
* Alternative1: Complete proc @ instant and store NTF data, defer handling NTF wait state to 'global' context,
               introduce wait state before proc tx'ing RSP/REQ to stall remote/local procedure 
               in case of pending NTF
* Alternative2: keep original RX node (until instant!) and re-use for NTF (pre allocate extra for PU/DLE),
               ie use piggy-backing on all NTF, serialize PU/DLE - hold off on procedure TX depending on BOTH TX node avail and NTF node (ie RX node) avail
* Alternative3: as Alternative2 but use alternate pool of smaller nodes for NTF, pre-alloc node for NTF
               on first TX in procedure, hold-off first TX in case NTF not avail, serialize PU/DLE NTF
               

                       Device 1 (P)                    Device 2 (C)
                            |                              |
                            |<------ REQ ----------------- |
                            |                     ack ---->|
                            | ------ RSP ----------------->|
                            |<---- ack                     |

                                         or
                            |                              |
                            | ------ REQ ----------------->|
                            |<---- ack                     |

                                        ...

                            |<------ IND ----------------- |
                            |                     ack ---->|
                            |                              |
                            |                              |
                            |                              |
                            |                              |
                          ------------- INSTANT -------------
                            |                          -- NTF --
                            |                              | 
                            |<------ REQ ----------------- |   Note: Here a nACK would push back explicitly
                            |                     ack ---->|         Since we cannot, a hold-off on TX of RSP/REQ
                            |                              |         will use procedure timeout as "push back"
                            |                              |
                            |                              |
                            |                              |
                            |                              |
                        -- NTF! --                         |
                            | ------ RSP/REQ-------------->|
                            |<---- ack                     |
                            |<------ IND ----------------- |
                            |                     ack ---->|
                            |                              |
                            |                              |

As decided in meeting 10/01-23, new preferred solution is Alternative2 with a follow-up of Alternative3 under KConfig, as Alt3 will result in increase memory usage

kruithofa, 2023-03-07:
fixed by #54669

PR #55921 (2023-05-12): Bluetooth: Controller: Rx node piggy v2

Fixes #45323.

When possible re-use the already allocated RX node for notifications.
Store (retain) RX node and Link element on RX if NTF could occur.
Pass link element to LLCP (ull_cp_rx()) together with RX node.
New RX node type RETAIN introduced to signal retention

When no RX node is available allocate one and hold off TX on procedures
until such time that a node is available for NTF.

In case waiting for NTF buffer avail is needed, allocate and store TX
node to use for TX once NTF becomes available.

On termination of connection, pass list of retai [... cut]
closed 2023-03-07 by kruithofa