zephyrproject-rtos/zephyr #45324: NET_TCP_BACKLOG_SIZE is unused, it has to be either implemented or deleted

AndreyDodonov-EH, 2022-05-04:
### Discussed in https://github.com/zephyrproject-rtos/zephyr/discussions/44973

<div type='discussions-op-text'>

<sup>Originally posted by **AndreyDodonov-EH** April 20, 2022</sup>
According to the quick grep, it's not used anymore, but still present in KConfig:
https://github.com/zephyrproject-rtos/zephyr/blob/main/subsys/net/ip/Kconfig#L366

Is it not needed anymore or am I missing something?</div>

mbolivar-nordic, 2022-05-10:
@AndreyDodonov-EH could you please send a PR removing the unused option as suggested by @jukkar ?

AndreyDodonov-EH, 2022-05-11:
@mbolivar-nordic Yeah, will do!
But we need to keep this (or similar) issue open to make sure that we know that there is implementation missing.

AndreyDodonov-EH, 2022-05-13:
@mbolivar-nordic Here you go: https://github.com/zephyrproject-rtos/zephyr/pull/45614

closed 2022-05-13 by carlescufi

mbolivar-nordic, 2022-05-13:
> But we need to keep this (or similar) issue open to make sure that we know that there is implementation missing.

OK, please open a new issue and mark it as an "enhancement" describing the missing implementation

AndreyDodonov-EH, 2022-05-14:
@mbolivar-nordic 
I think I was incorrect, I spent some time trying to understand the current implementation, and now I think that this functionality is not missing, because we can have multiple contexts (i.e. sockets) independent of each other, so the number of simultaneous connection attempts can be arbitrary.

@rlubos  Can you confirm that this is indeed so and it's not necessary to open a new issue?

As reference:
https://docs.zephyrproject.org/3.0.0/reference/kconfig/CONFIG_NET_TCP_BACKLOG_SIZE.html

rlubos, 2022-05-16:
> @mbolivar-nordic I think I was incorrect, I spent some time trying to understand the current implementation, and now I think that this functionality is not missing, because we can have multiple [...]

I think you're right, we allocate a new TCP context for each initiated connection so that config is no longer applicable.