zephyrproject-rtos/zephyr #45337: Undefined reference to timing functions

torres98, 2022-05-04:
During the compilation of a C++ project i get an undefined reference error regarding the timing functions declared in zephyr/timing/timing.h. Even though the compilation flag 'CONFIG_TIMING_FUNCTIONS' is set and the include doesn't raise any error, the linker doesn't seem to add the function code in the binaries...

Here is the complete output of the build command:

```
west build -p -b cc1352r1_launchxl samples/matrix_test/

-- west build: making build dir /home/torres/zephyrproject/zephyr/build pristine
-- west build: generating a build system
Loading Zephyr default modules (Zephyr base).
-- Application: /home/torres/zephyrproject/zephyr/samples/matrix_test
-- Found Python3: /usr/bin/python3.8 (found suitable exact version "3.8.10") found components: Interpreter 
-- Cache files will be written to: /home/torres/.cache/zephyr
-- Zephyr version: 3.0.99 (/home/torres/zephyrproject/zephyr)
-- Found west (found suitable version "0.13.1", minimum required is "0.7.1")
-- Board: cc1352r1_launchxl
-- ZEPHYR_TOOLCHAIN_VARIANT not set, trying to locate Zephyr SDK
-- Found host-tools: zephyr 0.14.1 (/home/torres/zephyr-sdk-0.14.1)
-- Found dtc: /home/torres/zephyr-sdk-0.14.1/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.6.0", minimum required is "1.4.6")
-- Found toolchain: zephyr 0.14.1 (/home/torres/zephyr-sdk-0.14.1)
-- Found BOARD.dts: /home/torres/zephyrproject/zephyr/boards/arm/cc1352r1_launchxl/cc1352r1_launchxl.dts
-- Generated zephyr.dts: /home/torres/zephyrproject/zephyr/build/zephyr/zephyr.dts
-- Generated devicetree_unfixed.h: /home/torres/zephyrproject/zephyr/build/zephyr/include/generated/devicetree_unfixed.h
-- Generated device_extern.h: /home/torres/zephyrproject/zephyr/build/zephyr/include/generated/device_extern.h
-- Including generated dts.cmake file: /home/torres/zephyrproject/zephyr/build/zephyr/dts.cmake
Parsing /home/torres/zephyrproject/zephyr/Kconfig
Loaded configuration '/home/torres/zephyrproject/zephyr/boards/arm/cc1352r1_launchxl/cc1352r1_launchxl_defconfig'
Merged configuration '/home/torres/zephyrproject/zephyr/samples/matrix_test/prj.conf'
Configuration saved to '/home/torres/zephyrproject/zephyr/build/zephyr/.config'
Kconfig header saved to '/home/torres/zephyrproject/zephyr/build/zephyr/include/generated/autoconf.h'
-- The C compiler identification is GNU 10.3.0
-- The CXX compiler identification is GNU 10.3.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/torres/zephyr-sdk-0.14.1/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc
-- Configuring done
-- Generating done
-- Build files have been written to: /home/torres/zephyrproject/zephyr/build
-- west build: building application
[1/155] Preparing syscall dependency handling

[2/155] Generating include/generated/version.h
-- Zephyr version: 3.0.99 (/home/torres/zephyrproject/zephyr), build: zephyr-v3.0.0-3059-ga7969279613d
[145/155] Linking CXX executable zephyr/zephyr_pre0.elf
FAILED: zephyr/zephyr_pre0.elf zephyr/zephyr_pre0.map /home/torres/zephyrproject/zephyr/build/zephyr/zephyr_pre0.map 
: && ccache /home/torres/zephyr-sdk-0.14.1/arm-zephyr-eabi/bin/arm-zephyr-eabi-g++   /home/torres/zephyr-sdk-0.14.1/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/10.3.0/thumb/v7e-m/nofp/crtbegin.o zephyr/CMakeFiles/zephyr_pre0.dir/misc/empty_file.c.obj -o zephyr/zephyr_pre0.elf  zephyr/CMakeFiles/offsets.dir/./arch/arm/core/offsets/offsets.c.obj  -fuse-ld=bfd  -Wl,-T  zephyr/linker_zephyr_pre0.cmd  -Wl,-Map=/home/torres/zephyrproject/zephyr/build/zephyr/zephyr_pre0.map  -Wl,--whole-archive  app/libapp.a  zephyr/libzephyr.a  zephyr/arch/common/libarch__common.a  zephyr/arch/arch/arm/core/aarch32/libarch__arm__core__aarch32.a  zephyr/arch/arch/arm/core/aarch32/cortex_m/libarch__arm__core__aarch32__cortex_m.a  zephyr/lib/libc/newlib/liblib__libc__newlib.a  zephyr/lib/posix/liblib__posix.a  zephyr/subsys/random/libsubsys__random.a  zephyr/subsys/timing/libsubsys__timing.a  zephyr/drivers/console/libdrivers__console.a  zephyr/drivers/gpio/libdrivers__gpio.a  zephyr/drivers/serial/libdrivers__serial.a  zephyr/drivers/entropy/libdrivers__entropy.a  zephyr/drivers/timer/libdrivers__timer.a  zephyr/drivers/pinctrl/libdrivers__pinctrl.a  modules/ti/simplelink/source/ti/devices/cc13x2_cc26x2/lib..__modules__hal__ti__simplelink__source__ti__devices__cc13x2_cc26x2.a  modules/ti/simplelink/lib..__modules__hal__ti__simplelink.a  -Wl,--no-whole-archive  zephyr/kernel/libkernel.a  -L"/home/torres/zephyr-sdk-0.14.1/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/10.3.0/thumb/v7e-m/nofp"  -L/home/torres/zephyrproject/zephyr/build/zephyr  -lgcc  zephyr/arch/common/libisr_tables.a  -no-pie  -mcpu=cortex-m4  -mthumb  -mabi=aapcs  -mfp16-format=ieee  -Wl,--gc-sections  -Wl,--build-id=none  -Wl,--sort-common=descending  -Wl,--sort-section=alignment  -Wl,-u,_OffsetAbsSyms  -Wl,-u,_ConfigAbsSyms  -nostdlib  -static  -Wl,-X  -Wl,-N  -Wl,--orphan-handling=warn  -lstdc++  -lm  -Wl,-lc  -L"/home/torres/zephyr-sdk-0.14.1/arm-zephyr-eabi/arm-zephyr-eabi"/lib/thumb/v7e-m/nofp  -Wl,-lgcc  -lc /home/torres/zephyr-sdk-0.14.1/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/10.3.0/thumb/v7e-m/nofp/crtend.o && cd /home/torres/zephyrproject/zephyr/build/zephyr && /usr/local/bin/cmake -E echo
/home/torres/zephyr-sdk-0.14.1/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/10.3.0/../../../../arm-zephyr-eabi/bin/ld.bfd: app/libapp.a(main.cpp.obj): in function `main':
/home/torres/zephyrproject/zephyr/samples/matrix_test/src/main.cpp:18: undefined reference to `timing_init()'
/home/torres/zephyr-sdk-0.14.1/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/10.3.0/../../../../arm-zephyr-eabi/bin/ld.bfd: /home/torres/zephyrproject/zephyr/samples/matrix_test/src/main.cpp:19: undefined reference to `timing_start()'
/home/torres/zephyr-sdk-0.14.1/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/10.3.0/../../../../arm-zephyr-eabi/bin/ld.bfd: /home/torres/zephyrproject/zephyr/samples/matrix_test/src/main.cpp:75: undefined reference to `timing_stop()'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
```

This is the main.cpp file:

```
#include <iostream>

#include <kernel.h>
#include <zephyr/timing/timing.h>
#include <timing/timing.h>

int main() {

    timing_t start_time, end_time;
    uint64_t total_cycles;
    uint64_t total_ns;

    timing_init();
    timing_start();

    start_time = timing_counter_get();
    end_time = timing_counter_get();
    total_cycles = timing_cycles_get(&start_time, &end_time);
    total_ns = timing_cycles_to_ns(total_cycles);

    timing_stop();

    std::cout << "Time required for matrix_test: " << end_time << std::endl;
    return 0;
}
```

And here the prj.conf file:

```
CONFIG_CPLUSPLUS=y

#include std c/c++
CONFIG_NEWLIB_LIBC=y
CONFIG_LIB_CPLUSPLUS=y

#for exceptions
CONFIG_NEWLIB_LIBC_NANO=n
CONFIG_EXCEPTIONS=y

#for timing
CONFIG_TIMING_FUNCTIONS=y

#for randomness
CONFIG_ENTROPY_GENERATOR=y

#54000 bytes seems to be roughly the max heap size that works... 
CONFIG_HEAP_MEM_POOL_SIZE=54000
```

This issue is with the latest stable version of zephyr (3.0.99) and with the LaunchXL-cc1352r1 board.

mbolivar-nordic, 2022-05-10:
@torres98 can you please post a reproducer? matrix_test is not an upstream application, so we cannot diagnose this issue without more information

stephanosio, 2022-05-10:
This is because the `timing.h` is missing the global `extern "C"`:

https://github.com/zephyrproject-rtos/zephyr/blob/83c7baf34a0de457dcd6b1e178bb898645a8cc01/include/zephyr/kernel.h#L24-L26

torres98, 2022-05-11:
@stephanosio Thank you for the answer, i tried adding that option to the `timing.h` file and it compiles correctly. Is this change going to be added to the next release?

stephanosio, 2022-05-11:
> Is this change going to be added to the next release?

Please feel free to create a PR if you would like; if not, I will do that to get this change in for 3.1.0.

torres98, 2022-05-11:
I'll surely do that. Thanks again!

stephanosio, 2022-05-11:
> I'll surely do that.

That's great, thanks.

I will keep this issue open so that we do not forget to fix this for 3.1.0. When you create a PR, you can specify `Fixes #45337` to link this issue.

PR #46047 (2022-05-26): timing: Fix symbol extern for C++
This commit adds `extern "C"` to `timing.h` so that C function names do
not get mangled when including this header file from a C++ source file.

Fixes #45337
closed 2022-05-26 by mbolivar-nordic