zephyrproject-rtos/zephyr #45349: ESP32: fails to chain-load sample/board/esp32/wifi_station from MCUboot

mniestroj, 2022-05-04:
**Describe the bug**
Complex application, such as `samples/boards/esp32/wifi_station`, fails to boot when chain-loaded from MCUboot (`Fatal exception (0): IllegalInstruction`). Simpler application, such as `samples/boards/esp32/wifi_station`, works without issues.

Issue seems to be related to some low-level details like code alignment on flash or RAM usage. When reproducing with downstream application, then simply disabling/enabling `CONFIG_BOOT_BANNER=y` results in succeded/failed boot.

**To Reproduce**
Steps to reproduce the behavior:
1. Build and flash bootloader using https://github.com/mcu-tools/mcuboot/blob/main/docs/readme-espressif.md
2. Build and flash `samples/boards/esp32/wifi_station` with MCUboot bootloader support:
```
west build -p -b esp32 zephyr/samples/boards/esp32/wifi_station -t flash -- -DCONFIG_BOOTLOADER_MCUBOOT=y
```
3. See error: `Fatal exception (0): IllegalInstruction`

**Expected behavior**
`wifi_station` application should be successfully chain-loaded and executed without errors

**Impact**
Showstopper. This prevents using complex applications on ESP32 platform with DFU capabilities.

**Logs and console output**
This is `samples/boards/esp32/wifi_station` (fails to boot):
```
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff8598,len:4932
load:0x40093000,len:13804
load:0x4009b800,len:5976
entry 0x40093448
[esp32] [INF] Enabling RNG early entropy source...
[esp32] [INF] *** Booting MCUboot build v1.8.0-1724-ge86f575 ***
[esp32] [INF] Primary image: magic=unset, swap_type=0x1, copy_done=0x3, image_ok=0x3
[esp32] [INF] Scratch: magic=unset, swap_type=0x1, copy_done=0x3, image_ok=0x3
[esp32] [INF] Boot source: primary slot
[esp32] [INF] Swap type: none
[esp32] [INF] Disabling RNG early entropy source...
[esp32] [INF] br_image_off = 0x10000
[esp32] [INF] ih_hdr_size = 0x20
[esp32] [INF] Loading image 0 - slot 0 from flash, area id: 1
[esp32] [INF] DRAM segment: start=0x12c16, size=0x1c0c, vaddr=0x3ffb0000
[esp32] [INF] IRAM segment: start=0x14824, size=0x14744, vaddr=0x40080000
[esp32] [INF] start=0x4008cae4
Fatal exception (0): IllegalInstruction
epc1=0x4009343d, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000
```

This is `samples/hello_world` (succeeds to boot):
```
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff8598,len:4932
load:0x40093000,len:13804
load:0x4009b800,len:5976
entry 0x40093448
[esp32] [INF] Enabling RNG early entropy source...
[esp32] [INF] *** Booting MCUboot build v1.8.0-1724-ge86f575 ***
[esp32] [INF] Primary image: magic=unset, swap_type=0x1, copy_done=0x3, image_ok=0x3
[esp32] [INF] Scratch: magic=unset, swap_type=0x1, copy_done=0x3, image_ok=0x3
[esp32] [INF] Boot source: primary slot
[esp32] [INF] Swap type: none
[esp32] [INF] Disabling RNG early entropy source...
[esp32] [INF] br_image_off = 0x10000
[esp32] [INF] ih_hdr_size = 0x20
[esp32] [INF] Loading image 0 - slot 0 from flash, area id: 1
[esp32] [INF] DRAM segment: start=0xf46, size=0x2a8, vaddr=0x3ffb0000
[esp32] [INF] IRAM segment: start=0x11f0, size=0x3278, vaddr=0x40080000
[esp32] [INF] start=0x400830d8
�*** Booting Zephyr OS build zephyr-v3.0.0-3349-gd5e7732e0d0f  ***
Hello World! esp32
```

**Environment (please complete the following information):**
 - OS: Linux
 - Zephyr SDK 0.14.1
 - `d5e7732e0d0f66bc9a3684f9597d9fd6a2d351b2` (tip of `main`) and `v3.0.0`

sylvioalves, 2022-05-06:
@mniestroj Thanks. I found the issue. It will be submitted upstream. It might take a bit to be synced in Zephyr. Let me know if you need the changes asap.

sylvioalves, 2022-05-13:
The PR that fixes this issue was merged upstream: https://github.com/mcu-tools/mcuboot/commit/c4b305863a5d508c3223361d4a20c089a803879f

mniestroj, 2022-05-13:
> The PR that fixes this issue was merged upstream: [mcu-tools/mcuboot@c4b3058](https://github.com/mcu-tools/mcuboot/commit/c4b305863a5d508c3223361d4a20c089a803879f)

Was looking at it just few minutes ago, as a "candidate" that could solve this issue. Would be nice if such PRs would include "fix" in the commit message somewhere, so it would be obvious :)

Thanks for fast feedback!

sylvioalves, 2022-05-13:
Thanks for the note. I'll redirect to the team.

PR #46106 (2022-06-05): manifest: MCUboot update
bring fix from mcu-tools/mcuboot:

- espressif:esp32: Move app entry point call back to
  iram_loader_seg region

fixes #45349
fixes #46093
commit 0b9f0a45 (2022-06-05): manifest: MCUboot update

bring fix from mcu-tools/mcuboot:

- espressif:esp32: Move app entry point call back to
  iram_loader_seg region

fixes #45349
fixes #46093
closed 2022-06-05 by carlescufi