zephyrproject-rtos/zephyr #45361: samples/bluetooth/hci_usb doesn't build for nucleo_wb55rg

karlp, 2022-05-04:
The Nucleo WB55RG board supports both USB and Bluetooth, so, by the definitions of the requirements: https://docs.zephyrproject.org/latest/samples/bluetooth/hci_usb/README.html it should be supported.

https://docs.zephyrproject.org/latest/boards/arm/nucleo_wb55rg/doc/nucleo_wb55rg.html doesn't explicitly mention that USB is supported, but it was added in https://github.com/zephyrproject-rtos/zephyr/commit/91d61393388759a8ab3218f1c6e311aa59aa58f6 back in the v2.3 era, so... it _should_ work.

It fails to link however:
```
(.env3) karlp@beros:~/src/zephyr-k2022/zephyr (main)$ west build -p auto -b nucleo_wb55rg samples/bluetooth/hci_usb
[172/182] Linking C executable zephyr/zephyr_pre0.elf
FAILED: zephyr/zephyr_pre0.elf zephyr/zephyr_pre0.map 
: && /home/karlp/tools/zephyr-sdk-0.14.1/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc   zephyr/CMakeFiles/zephyr_pre0.dir/misc/empty_file.c.obj -o zephyr/zephyr_pre0.elf  -fuse-ld=bfd  -Wl,-T  zephyr/linker_zephyr_pre0.cmd  -Wl,-Map=/home/karlp/src/zephyr-k2022/zephyr/build/zephyr/zephyr_pre0.map  -Wl,--whole-archive  app/libapp.a  zephyr/libzephyr.a  zephyr/arch/common/libarch__common.a  zephyr/arch/arch/arm/core/aarch32/libarch__arm__core__aarch32.a  zephyr/arch/arch/arm/core/aarch32/cortex_m/libarch__arm__core__aarch32__cortex_m.a  zephyr/arch/arch/arm/core/aarch32/mpu/libarch__arm__core__aarch32__mpu.a  zephyr/lib/libc/minimal/liblib__libc__minimal.a  zephyr/lib/posix/liblib__posix.a  zephyr/soc/arm/common/cortex_m/libsoc__arm__common__cortex_m.a  zephyr/subsys/bluetooth/common/libsubsys__bluetooth__common.a  zephyr/subsys/bluetooth/host/libsubsys__bluetooth__host.a  zephyr/subsys/net/libsubsys__net.a  zephyr/drivers/interrupt_controller/libdrivers__interrupt_controller.a  zephyr/drivers/usb/device/libdrivers__usb__device.a  zephyr/drivers/clock_control/libdrivers__clock_control.a  zephyr/drivers/console/libdrivers__console.a  zephyr/drivers/gpio/libdrivers__gpio.a  zephyr/drivers/hwinfo/libdrivers__hwinfo.a  zephyr/drivers/serial/libdrivers__serial.a  zephyr/drivers/bluetooth/libdrivers__bluetooth.a  zephyr/drivers/timer/libdrivers__timer.a  zephyr/drivers/pinctrl/libdrivers__pinctrl.a  modules/stm32/stm32cube/lib..__modules__hal__stm32__stm32cube.a  -Wl,--no-whole-archive  zephyr/kernel/libkernel.a  zephyr/CMakeFiles/offsets.dir/./arch/arm/core/offsets/offsets.c.obj  -L"/home/karlp/tools/zephyr-sdk-0.14.1/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/10.3.0/thumb/v7e-m/nofp"  -L/home/karlp/src/zephyr-k2022/zephyr/build/zephyr  -lgcc  zephyr/arch/common/libisr_tables.a  -no-pie  -mcpu=cortex-m4  -mthumb  -mabi=aapcs  -mfp16-format=ieee  -Wl,--gc-sections  -Wl,--build-id=none  -Wl,--sort-common=descending  -Wl,--sort-section=alignment  -Wl,-u,_OffsetAbsSyms  -Wl,-u,_ConfigAbsSyms  -nostdlib  -static  -Wl,-X  -Wl,-N  -Wl,--orphan-handling=warn && cd /home/karlp/src/zephyr-k2022/zephyr/build/zephyr && /usr/bin/cmake -E echo
/home/karlp/tools/zephyr-sdk-0.14.1/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/10.3.0/../../../../arm-zephyr-eabi/bin/ld.bfd: zephyr/drivers/bluetooth/libdrivers__bluetooth.a(ipm_stm32wb.c.obj): in function `bt_ipm_close':
/home/karlp/src/zephyr-k2022/zephyr/drivers/bluetooth/hci/ipm_stm32wb.c:594: undefined reference to `bt_hci_cmd_send_sync'
/home/karlp/tools/zephyr-sdk-0.14.1/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/10.3.0/../../../../arm-zephyr-eabi/bin/ld.bfd: zephyr/drivers/bluetooth/libdrivers__bluetooth.a(ipm_stm32wb.c.obj): in function `bt_ipm_ble_init':
/home/karlp/src/zephyr-k2022/zephyr/drivers/bluetooth/hci/ipm_stm32wb.c:505: undefined reference to `bt_hci_cmd_send_sync'
/home/karlp/tools/zephyr-sdk-0.14.1/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/10.3.0/../../../../arm-zephyr-eabi/bin/ld.bfd: zephyr/drivers/bluetooth/libdrivers__bluetooth.a(ipm_stm32wb.c.obj): in function `bt_ipm_set_addr':
/home/karlp/src/zephyr-k2022/zephyr/drivers/bluetooth/hci/ipm_stm32wb.c:474: undefined reference to `bt_hci_cmd_create'
/home/karlp/tools/zephyr-sdk-0.14.1/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/10.3.0/../../../../arm-zephyr-eabi/bin/ld.bfd: /home/karlp/src/zephyr-k2022/zephyr/drivers/bluetooth/hci/ipm_stm32wb.c:490: undefined reference to `bt_hci_cmd_send_sync'
/home/karlp/tools/zephyr-sdk-0.14.1/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/10.3.0/../../../../arm-zephyr-eabi/bin/ld.bfd: zephyr/drivers/bluetooth/libdrivers__bluetooth.a(ipm_stm32wb.c.obj): in function `bt_ipm_ble_init':
/home/karlp/src/zephyr-k2022/zephyr/drivers/bluetooth/hci/ipm_stm32wb.c:516: undefined reference to `bt_hci_cmd_create'
/home/karlp/tools/zephyr-sdk-0.14.1/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/10.3.0/../../../../arm-zephyr-eabi/bin/ld.bfd: /home/karlp/src/zephyr-k2022/zephyr/drivers/bluetooth/hci/ipm_stm32wb.c:525: undefined reference to `bt_hci_cmd_send_sync'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: /usr/bin/cmake --build /home/karlp/src/zephyr-k2022/zephyr/build
(.env3) karlp@beros:~/src/zephyr-k2022/zephyr (main)$
```

I believe this _should_ be a supported configuration, and that this is therefore a bug.

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain Zephyr SDK 0.14.1
 - zephyr tree: zephyr-v3.0.0-3371-gaceefa90ec

erwango, 2022-05-05:
@karlp `USB_DEVICE_BLUETOOTH` implies support of BT_CHI_RAW protocol, which is not supported on nucleo_wb55rg.

```
menuconfig USB_DEVICE_BLUETOOTH
	bool "USB Bluetooth Device Class support"
	select BT
	select BT_HCI_RAW
```

This explains why `sample/hci_usb` won't build.
Hence this is not a bug, but a new feature to be supported, and then I'm converting this issue as an `feature request`.
Please close if this is not something that you don't actively request (eg only testing this 'out of curiosity').

karlp, 2022-05-05:
Given that it didn't work, I did find an alternative board, yes.  I also understand that you don't want to keep feature requests open as issues permanently, so close it if you feel you must.

closed 2022-05-05 by erwango